MichaIng/DietPi #1506: -bash: warning: setlocale: LC_ALL: cannot change locale (0)

Akito13, 2018-02-13:
## Creating a bug report/issue:
Yes.

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
6
1
- SBC Device (EG: RPi 3)?
Raspberry Pi 3 Model B
- Power supply used (EG: 5v 1A RAVpower)?
5V DC 1,5A
- SD card used (EG: Sandisk ultra)?
Transcend Premium
- Distro (EG: Jessie) | ```uname -a```?
Linux hostname 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Software title?
locales; locale-gen; localedef; 
- Can this issue be replicated on a fresh installation of DietPi?
Yes, dietpi is installed on 3 machines, all the same problem.

#### Expected behaviour:

Locale should be able to be changed by user and stay like that.

#### Actual behaviour:

Locale can be changed by user but after every re-login to shell, in what way ever, the locale are partially set automatically to "0". No matter how extremely the user changes every environment variable, etc. the locale is still set to 0 on every login or reboot or access to shell.

#### Steps to reproduce:

1. Install dietpi with with standard locale chosen by system, login to shell.
2. Change every locale setting, export every important locale environment variable and see that nothing changed, after re-login.

#### Did you submit a ```dietpi-bugreport```?

No.

#### Extra details:

```
$ cat /etc/environment
LANG=en_GB.UTF-8

$ cat /etc/bash.bashrc
...
export LC_ALL=en_GB.UTF-8
...

$ locale
LANG=0                          #is ALWAYS set by system on re-boot/re-login, no matter what the user chose
LANGUAGE=0:0              #is ALWAYS set by system on re-boot/re-login, no matter what the user chose
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC="en_GB.UTF-8"
LC_TIME="en_GB.UTF-8"
LC_COLLATE="en_GB.UTF-8"
LC_MONETARY="en_GB.UTF-8"
LC_MESSAGES="en_GB.UTF-8"
LC_PAPER="en_GB.UTF-8"
LC_NAME="en_GB.UTF-8"
LC_ADDRESS="en_GB.UTF-8"
LC_TELEPHONE="en_GB.UTF-8"
LC_MEASUREMENT="en_GB.UTF-8"
LC_IDENTIFICATION="en_GB.UTF-8"
LC_ALL=en_GB.UTF-8

$ locale -a
en_GB.utf8

$ cat ~/.bashrc
...
export LC_ALL=en_GB.UTF-8
...
```

NOTE! issue #1430 is something similar, but not exactly the same. Therefore, this is not a duplicate.

MichaIng, 2018-02-13:
@Akito13 
I am pretty sure it is the same issue 😄. Please follow the steps in the other topic and also try to remove `LANG=en_GB.UTF-8` and `export LC_ALL=en_GB.UTF-8` from environment/bash.bashrc.
These two files should be able to stay untouched. `/etc/default/locale` will be adjusted by locales packages configuration at that is enough for all needed. We touched these files ourself too much in the past and it often lead to errors here and there 🙂.

Only if you want to force a locale on Dropbear SSH panel, then you need to export `LC_ALL` somewhere, as it ignores `/etc/default/locale` and just uses POSIX as default. Best for my impression is `/etc/profile.d/<own_file>` to avoid adjusting system files, or your own `~/.bashrc`, if you don't want to force it for other users.

Fourdee, 2018-02-13:
@Akito13 

All locales will be resolved during v6.2 patch. ETA release this week. Please wait until then, to resolve the issue.

I'll mark this as closed (https://github.com/Fourdee/DietPi/issues/1430#issuecomment-365364406)

closed 2018-02-13 by Fourdee