MichaIng/DietPi #1508: Permission denied after flashing image to SD card

dirtyharrywk, 2018-02-13:
## Creating a bug report/issue:
ssh root@192.168.2.110
root@192.168.2.110's password: 
Permission denied, please try again.

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?  DietPi_v6.0_OdroidXU4-ARMv7-Jessie
- SBC Device (EG: RPi 3)?  Odroid HC1
- Power supply used (EG: 5v 1A RAVpower)?  5V/4A POWER SUPPLY US PLUG
- SD card used (EG: Sandisk ultra)?  Sandisk ultra
- Distro (EG: Jessie) | ```uname -a```?  Jessie

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:
To be able to login via ssh

#### Actual behaviour:
permission denied using 'root' as the user and 'dietpi' as the password.

#### Steps to reproduce:
Open terminal (I'm using a MacBook Air)
enter 'ssh root@192.168.2.110' press enter
terminal displays 'root@192.168.2.110's password:'
enter 'dietpi'

#### Did you submit a ```dietpi-bugreport```?
No

#### Extra details:

This is a recently flashed of the latest image onto a Sandisk Ultra microSD card.  I'm connect via wifi.

Fourdee, 2018-02-14:
@dirtyharrywk 

Thanks for the report 👍 

I'am unable to replicate using Putty/Kitty SSH clients on my HC1. Login is successful.

This may be an issue with the Mac SSH client (which I both lack any knowledge in). However, its most likely a bad image write.

Please try the following:
- Full format the SD card 
- Write the DietPi image again
- Plug into HC1 and retry.

If problems persist, the next step would be to locally login to system, but as we know HC1 lacks HDMI. Therefore, the only option after that, would be to try 
- Copy/paste password into SSH window
- A different SSH client
- Verify PSU is stable (eg: official 5v/4a)
- Try a fresh image, without any 2.5inch HDD attached.

dirtyharrywk, 2018-02-14:
Fourdee,

Thanks for the quick response.  I was using a USB adapter to flash the image to the microSD.  I switched to an SD card adapter and refreshed the dietpi image.  I inserted the card into my Odroid and was able to login via SSH with root/diepi.

Thanks again!!

closed 2018-02-14 by dirtyharrywk