MichaIng/DietPi #1511: Rock64 eMMC image

elRadix, 2018-02-14:
@Fourdee 

How to flash debian strech v160 on the eMMC ?  Mine is 64gb.
Will you have to create special eMMC image?

I already have a USB eMMC reader bought together with the Rock64.

Thank you

Fourdee, 2018-02-14:
@elRadix 

I have no idea, lack the EMMC for device to test ☹️ 

I'll flag this for research, leave it with me.

elRadix, 2018-02-14:
@Fourdee 

In the meantime I tried to experiment. Flashed the latest image through etcher on the eMMC card.
Booting no issues at all but I got an issue with RootFS

[image]

[image]

Current partition table
[image]

Is there a solution to resize the partition correctly?

TheExpertNoob, 2018-03-10:
Sorry I'm late to this party, but I used Windows' WinDiskImager to image the eMMC with the adapter just as I would a micro SD card with no issues.

DmitryMyadzelets, 2018-04-04:
Run `resize_rootfs.sh` from the console. For some reason it fails to do automatically, for now.

Fourdee, 2018-04-05:
EMMC works fine in local testing.

Marking closed in favor of : https://github.com/Fourdee/DietPi/issues/1452.

closed 2018-04-05 by Fourdee