MichaIng/DietPi #1513: DietPi_OPiLite-ARMv7-Stretch

peon83, 2018-02-14:
## Creating a bug report/issue:
hello, i have a problem when install dietpi.
i download dietpi image write in the sd image and not install.
the SD write in two computer and not possibility install.

help my please.
#### Required Information:
- DietPi Version 6.0?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Kingston)?
- Orange pi Lite

Fourdee, 2018-02-15:
@peon83 

We'll need more info to assist.

Please can you provide following info:
- Any output/text on the HDMI display?
- Does the device boot up? 
- Have you tried re-writing the image using a different SD card and PSU?

The more info + additional info you can provide, the more we can assist.

peon83, 2018-02-16:
We'll need more info to assist.

Please can you provide following info:

Any output/text on the HDMI display?
nothin, more color line, not text

Does the device boot up?
no, device not show nothin text for output hdmi

Have you tried re-writing the image using a different SD card and PSU?
the problem is when write the .img  dietpi.

the version 147  not problem, the version 160 not install or the output hdmi no is correct.

in moment the run .img armbian (ubuntu) the funcion is ok.

Fourdee, 2018-02-16:
@peon83 

Thanks, we'll try to replicate:
- Download http://dietpi.com/downloads/images/DietPi_OPiLite-ARMv7-Stretch.7z
- Write to SD
- Power on
- HDMI output seems fine, text onscreen, however, there are some line artifacts during reboots (possible driver/kernel issue outside of our control)
- After 2 reboots (FS expand), able to login and run DietPi 1st run setup
- Setup WiFi
- Connect via SSH

🈯️ Unable to replicate.
[image]

@peon83 

Please try a different SD card, verify PSU is stable.

As I cannot replicate the issue you are experiencing, i'll mark as closed. However, please reopen if required.

closed 2018-02-16 by Fourdee