MichaIng/DietPi #1515: Can't install Retropie with latest Odroid image

LieDanG, 2018-02-15:
Retropie using latest 6.0 image for Odroid C2 fails to build, it seems that there are missing packages that can't be installed

I read that it should build with previous versions of Dietpi, here is the log

https://pastebin.com/dfg69hyR

Fourdee, 2018-02-15:
@LieDanG 

> -- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- broken
CMake Error at /usr/share/cmake-3.0/Modules/CMakeTestCCompiler.cmake:61 (message):
  The C compiler "/usr/bin/cc" is not able to compile a simple test program.

Please verify following is installed:
- ```apt-get install build-essential automake make cmake```

I cant see any mention of ```build-essential``` in the log, so the RetroPie install must assume is already installed.

Fourdee, 2018-02-15:
NB: Flagging as non-DietPi issue, however, we'll assist where we can.

LieDanG, 2018-02-16:
I previously installed build-essential, installed the other packages and tried again with same results

https://pastebin.com/58NKdeAt

Fourdee, 2018-04-01:
REF: https://github.com/Fourdee/DietPi/issues/1653

Fourdee, 2018-04-01:
Notes: Could require Debian Stretch for Odroid C2? Wiki recommends Ubuntu: https://github.com/RetroPie/RetroPie-Setup/wiki/Odroid-C1-C2#download-ubuntu-image-for-odroid

Fourdee, 2018-04-01:
Odroid C2 Stretch:
[image]
[image]
[image]
[image]

Fourdee, 2018-04-02:
```
wget http://deb.odroid.in/c2/pool/main/m/mali-fbdev/mali-fbdev_20161003-r6p1-ee42997-15_arm64.deb -O package.deb
dpkg -i package.deb
rm package.deb
```

Appears ```mali450-odroid``` is the same package:
```
root@DietPi:~# dpkg -i package.deb
Selecting previously unselected package mali-fbdev.
(Reading database ... 36819 files and directories currently installed.)
Preparing to unpack package.deb ...
Unpacking mali-fbdev (20161003-r6p1-ee42997-15) ...
dpkg: error processing archive package.deb (--install):
 trying to overwrite '/usr/include/KHR/khrplatform.h', which is also in package mali450-odroid 20161218-r6p1-1
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 package.deb
root@DietPi:~# rm package.deb^C
root@DietPi:~# apt-cache search ^C
root@DietPi:~# dpkg -l | grep mali450
ii  mali450-odroid              20161218-r6p1-1                  arm64        Mali binary blob and development headers (ODROID build)
```

Fourdee, 2018-04-02:
Tried following, failed lvl10 init SDL
```
git clone --depth=1 https://github.com/RetroPie/RetroPie-Setup.git
cd RetroPie-Setup

#Odroid C2: Replace mali-fbdev depends with mali450-odroid used in meveric's repo
#sed -i 's/mali-fbdev/mali450-odroid/g' scriptmodules/supplementary/sdl2.sh
#sed -i 's/mali-fbdev/mali450-odroid/g' scriptmodules/system.sh
#sed -i 's/mali-fbdev/mali450-odroid/g' scriptmodules/emulators/retroarch.sh

./retropie_setup.sh
```

Fourdee, 2018-04-05:
I'll mark this as closed due to the following:
- RetroPie Wiki specifies Ubuntu image must be used.
- Seems incompatible with Debian Stretch + Meveric's repo currently.
- Outside our support range.

bernhard263, 2020-06-28:
Even if you have mali-fbdev, the setup will later fail when it installs the libegl1-mesa and libgles2-mesa packages because they try to install the same files and create a conflict. It's bugged. My solution was to create and install an empty debian package "mali-fbdev" to satisfy the installer and if necessary, manually install the correct mali fbdev drivers myself later. There are probably cleaner solutions to this but it worked for me.

Here's how to create an empty package mali-fbdev: https://ubuntuforums.org/showthread.php?t=910717

And here's how to manually install the files that are contained in mali-fbdev:
`git clone https://github.com/bootlin/mali-blobs.git
cd mali-blobs
cp -a r6p2/your_architecture/fbdev/lib* /usr/lib/your_architecture`