MichaIng/DietPi #1516: DietPi not booting after flashing the image.

ronakmd, 2018-02-16:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?>>DietPi_v6.0_RPi-ARMv6-Stretch
- SBC Device (EG: RPi 3)?>>>>Rpi MOdel B (armv71) index=3
- Power supply used (EG: 5v 1A RAVpower)?>>>From Adapter 5v 1.5A
- SD card used (EG: Sandisk ultra)?>>>Samsung 16
[image]

- Distro (EG: Jessie) | ```uname -a```?>>G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title?
[image]

- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:
[image]

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2018-02-16:
@ronakmd 

Hi,

Thanks for the report 👍 

Appears to be a basic network setup issue. 
When the prompt appears (as per screenshot), press ENTER (OK) or ESC. When asked to configure network in DietPi-Config, select OK.

You can now enter the networking menu to check/configure networking as needed.

@ronakmd 
Ethernet or WiFi?

ronakmd, 2018-02-16:
Hi,
I want to use wi-fi.
Can you also let me know that how much time it shall take to boot onto
desktop?
Thankyou.
Regards,
Ronak Dedhia.

ronakmd, 2018-02-16:
> @ronakmd <https://github.com/ronakmd>
>
> For WiFi, you will need to enable it:
>
>    - When the error prompt appears (as per screenshot), press ENTER (OK)
>    or ESC.
>    - When asked to configure network in DietPi-Config, select OK.
>    - Select Networking Options: Adapters
>    - Select WiFi and confirm enable
>    - Select WiFi again and enter the WiFi details as needed.
>    - Save changes when your done, and exit dietpi-config
>    - Setup will now resume.
>
> DietPi does not come pre-installed with a desktop, you can install one as
> required:
> http://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p5

Hi,
Desktop as in,now I am using raspbian stretch and it takes 18 seconds to
boot.
I am connecting it to monitor for display.
So I wanted to try diet pi to reduce boot time.
Is this possible?
Thanks in advance.

Fourdee, 2018-02-16:
@ronakmd 

Yes, all possible. 

Please refer to the Getting Started Guide:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=9#p9

If you have any further questions and/or troubleshooting in regards to setting up DietPi, please post these on our forums section for further assistance:
http://dietpi.com/phpbb/viewforum.php?f=11

I'll mark this as closed, forward support to guides/forum.

closed 2018-02-16 by Fourdee

ronakmd, 2018-02-19:
Hi Dan,
I cannot login to my dietpi account.
Also i followed all the steps of enabling wifi.
Still i am getting the same error as in screenshot.
Please let me know the issue.

Thankyou.

-- 
Thank you.

Regards,
Ronak Dedhia.