MichaIng/DietPi #1517: Internet and Hotspot with 2 wifi adapters

symbios24, 2018-02-16:
Hello, is it possible to create a hotspot using two wifi adapters one internal and one external?
using the first wifi adapter to connect to another hotspot that has internet, and the second to make a hotspot and share that internet?? 

instead of using the ethernet cable to give internet to Rpi from the router and after create a hotspot with eg the interal wifi

Thanks

SuBLiNeR, 2018-02-18:
HI there, 
yes, that's possible, i'm running a RPi Zero W with a second WiFi-USB-Adapter. It is mine travel Hotspot at which i could use than wpa_cli to change the connection of my WLAN1 as mine WLAN0 builds up mine private network.

For more upon this, i could give you some links to read on how to set this all up with dnsmasq and not installing dhcpcd or any other dhcpc-service. 

-> Setup HOSTAPD & DNSMASQ:
https://frillip.com/using-your-raspberry-pi-3-as-a-wifi-access-point-with-hostapd/

-> Setup HOSTAPD & ISC-DHCP-Service:
https://learn.adafruit.com/setting-up-a-raspberry-pi-as-a-wifi-access-point?view=all

symbios24, 2018-02-23:
All this are to make a hotspot that already is done by dietpi.

SuBLiNeR, 2018-02-25:
@symbios24 hi there, what are you trying to tell me? Above mentioned stuff is not done by dietpi automatically. I'm currently running this above setup via dietpi and this works well for me.
At dietpi you have to have eht0 active to create this thingy. Mine setup is only with internal wlan0 and wlan1 (USB-Dongle). This runs for an TOR Hotspot on a ZeroW and an RPi3 for travel with integrated VPN.
All within console and not on a Desktop; whenever i've away from home, i plug mine rpi on power, connect via SSH to it and than change the WPA_supplicant of the wlan1. Happy surfing without being spyed on.

So what do you want to know?

Fourdee, 2018-07-09:
Marking as closed in favor of: https://github.com/Fourdee/DietPi/issues/1542

closed 2018-07-09 by Fourdee