MichaIng/DietPi #1518: Installing OpenCV 3.4 on DietPi v6.1. cvNamedWindow not implemented.

KenthJohan, 2018-02-16:
#### Required Information:
- DietPi Version | 6.1
- SBC Device RPi 3

#### Additional Information (if applicable):
- OpenCV 3.4
- OpenGL 1080p
- LXDE
- Can be replicated on a fresh installation of DietPi

#### Expected behaviour:

Display a window from OpenCV

#### Actual behaviour:

`OpenCV Error: Unspecified error (The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script) in cvNamedWindow, file /root/opencv-3.4.0/modules/highgui/src/window.cpp, line 593 terminate called after throwing an instance of 'cv::Exception' what(): /root/opencv-3.4.0/modules/highgui/src/window.cpp:593: error: (-2) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function cvNamedWindow `

#### Steps to reproduce:

By following this guide: [Install OpenCV 3.2 Python/C++ on Raspberry PI](http://pklab.net/?id=392&lang=EN)

#### Extra details:

OpenCV 3.2 is outputting the same error.

Fourdee, 2018-02-20:
- Guide looks outdated (Jessie targeted?)?
- Packages not automatically being pulled in, only with ```libgtk2.0-dev```
```
root@DietPi:~#  apt-get install \
>   libjpeg-dev \
>   libtiff5-dev \
>   libjasper-dev \
>   libpng12-dev \
>   libavcodec-dev \
>   libavformat-dev \
>   libswscale-dev \
>   libeigen3-dev \
>   libxvidcore-dev \
>   libx264-dev \
>   libgtk2.0-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgtk2.0-dev : Depends: libgdk-pixbuf2.0-dev (>= 2.21.0) but it is not going to be installed
                 Depends: libpango1.0-dev (>= 1.20) but it is not going to be installed
                 Depends: libcairo2-dev (>= 1.6.4-6.1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

```
 apt-get install \
  libjpeg-dev \
  libtiff5-dev \
  libjasper-dev \
  libpng12-dev \
  libavcodec-dev \
  libavformat-dev \
  libswscale-dev \
  libeigen3-dev \
  libxvidcore-dev \
  libx264-dev \
  #libgtk2.0-dev libgdk-pixbuf2.0-dev libpango1.0-dev libcairo2-dev
```

Fourdee, 2018-02-20:
@KenthJohan 

During the guide, replace ```libpng12-dev``` with ```libpng-dev```, (due to https://github.com/Fourdee/DietPi/issues/1518#issuecomment-367063704) eg:
```
apt-get install \
  libjpeg-dev \
  libtiff5-dev \
  libjasper-dev \
  libpng-dev \
  libavcodec-dev \
  libavformat-dev \
  libswscale-dev \
  libeigen3-dev \
  libxvidcore-dev \
  libx264-dev \
  libgtk2.0-dev
```

KenthJohan, 2018-02-21:
> During the guide, replace libpng12-dev with libpng-dev

That fixes the problem. Thank you!

To be able to open a window with `CV_WINDOW_OPENGL` you must have `libgtkglext1 libgtkglext1-dev` before building OpenCV.

I run this script to install it now:
```
sudo apt-get remove libopencv*
sudo apt-get autoremove
sudo apt-get update
sudo apt-get upgrade

sudo apt-get -y install build-essential cmake cmake-curses-gui pkg-config
sudo apt-get -y install \
  libjpeg-dev \
  libtiff5-dev \
  libjasper-dev \
  libpng-dev \
  libavcodec-dev \
  libavformat-dev \
  libswscale-dev \
  libeigen3-dev \
  libxvidcore-dev \
  libx264-dev \
  libgtk2.0-dev
sudo apt-get -y install libv4l-dev v4l-utils
sudo modprobe bcm2835-v4l2
sudo apt-get -y install libatlas-base-dev gfortran
sudo apt-get -y install libgtkglext1 libgtkglext1-dev

wget https://github.com/opencv/opencv/archive/3.4.0.zip -O opencv_source.zip
wget https://github.com/opencv/opencv_contrib/archive/3.4.0.zip -O opencv_contrib.zip
unzip opencv_source.zip
unzip opencv_contrib.zip
cd opencv-3.4.0
mkdir build
cd build
cmake -D CMAKE_BUILD_TYPE=RELEASE \
	-D CMAKE_INSTALL_PREFIX=/usr/local \
	-D BUILD_WITH_DEBUG_INFO=OFF \
	-D BUILD_DOCS=OFF \
	-D BUILD_EXAMPLES=OFF \
	-D BUILD_TESTS=OFF \
	-D BUILD_opencv_ts=OFF \
	-D BUILD_PERF_TESTS=OFF \
	-D INSTALL_C_EXAMPLES=ON \
	-D INSTALL_PYTHON_EXAMPLES=OFF \
	-D OPENCV_EXTRA_MODULES_PATH=../../opencv_contrib-3.4.0/modules \
	-D ENABLE_NEON=ON \
	-D WITH_LIBV4L=ON \
	-D WITH_OPENGL=ON \
../
make -j3
sudo make install
sudo ldconfig
```

Fourdee, 2018-02-21:
@KenthJohan 

> That fixes the problem. Thank you!
To be able to open a window with CV_WINDOW_OPENGL you must have libgtkglext1 libgtkglext1-dev before building OpenCV.

Excellent, great to hear and thanks for sharing the install script (nice one) 👍 

As the original issue is now resolved, I'll mark this as closed. Please reopen if required.

closed 2018-02-21 by Fourdee