MichaIng/DietPi #1519: Odroid N1

Fourdee, 2018-02-16:
[image]

N1:
- Case was fun, although, needs a step by step guide for all users
- Molex power fitting is extremely tight, may bend/break board for some users if done incorrectly.
- Onboard SATA slot lacks the clip system with SATA2.0+, means the cable is likely to come out unexpectedly.

---

```
libmali-rk-midgard-t86x-r14p0

root@odroid:~# dpkg -l | grep rock
ii  gstreamer1.0-rockchip1               20171218-1                        arm64        The Gstreamer plugins for Rockchip RK3xxx platform.
ii  gstreamer1.0-rockchip1-extra         20180115-1                        arm64        The Extra Gstreamer plugins for Rockchip platform.
ii  libdrm-rockchip1:arm64               2.4.74-2                          arm64        Userspace interface to msm/kgsl kernel DRM services -- runtime
ii  librockchip-mpp1                     1.4.0-2                           arm64        Media Process Platform
ii  librockchip-vpu0                     1.4.0-2                           arm64        Media Process Platform

root@odroid:~# dpkg --get-selections | grep ^linux
linux-base                                      install
linux-libc-dev:arm64                            install

Fourdee, 2018-02-16:
reserved.

k-plan, 2018-02-17:
@Fourdee 

nice to see, that you have get a "engineering samples" from hardkernel. 👍 

For all who don't know what is/will be a "Odroid N1":

[HardKernel Blog: ODROID-N1](http://com.odroid.com/sigong/blog/blog_list.php?bid=193)

[The Next ODROID!](https://forum.odroid.com/viewtopic.php?f=29&t=29932#p213505)

[Laser cut acrylic case samples](https://forum.odroid.com/viewtopic.php?f=29&t=29932#p213505)

Fourdee, 2018-02-24:
Image based on Meveric's, notes:
- ```boot.ini``` ```root=UUID=e139ce78-9841-40fe-8823-96a304a09859```

k-plan, 2018-02-24:
@Fourdee 

> Nothing over serial

Hmm . have you give GPIO  Pin: 8 - 9 - 10 a spin with a USB-to-UART adapter? 115200-8-N-1 ?

Fourdee, 2018-02-24:
@k-plan 

6-8-10

Will give 8-9-10 a shot 👍

k-plan, 2018-02-24:
@Fourdee 

What UART adapter do you own?

With 5V, 3.3V or 1,8V  interface?

http://www.hardkernel.com/main/products/prdt_info.php?g_code=G134111883934

Do you have a level shifter?

[image]

k-plan, 2018-02-24:
@Fourdee 

http://com.odroid.com/sigong/blog/blog_list.php?tag=ODROID-N1

> **Known issues in the engineering sample PCB:**
> ...
> - The UART port on the 40pin GPIO header doesn’t work due to a missing power line in the level-shifter circuit.  We will solder a wire on the PCB to solve this issue which will be corrected on the mass production version.

ups ... sorry.

Fourdee, 2018-02-24:
@k-plan 

> ups ... sorry.

Thanks 👍 No worries, my fault for not being able to keep up.

k-plan, 2018-02-24:
@Fourdee 

edit:

> The default baud rate for RK3399 is 1500000,Some USB to serial chip baud rate can not support 1500000，So before you buy it, make sure it supports 1500000.

> RK3399 use the following serial parameters:
> 
> - Baud rate: 1500000
> - Data bit: 8
> - Stop bit: 1
> - Parity check: none
> - Flow control: none

http://wiki.t-firefly.com/index.php/Firefly-RK3399/Serial_debug/en

http://wiki.t-firefly.com/index.php/Firefly-RK3399/UART/en

Fourdee, 2018-02-24:
@k-plan 

Thanks, i'll revisit this after: https://github.com/Fourdee/DietPi/issues/1555

k-plan, 2018-02-26:
**Info Odroid N1 engineering sample PCB Serial Debug (UART)** 

>  - Known issues in the engineering sample PCB:
> pin 38 on the gpio is 1.8 v out..so it seems to me that would be it..as it provides the missing power rail .for the serial uart

[image]

[image]

> - The RK3399 has 5 UARTs, but the N1 is using UART #0 for serial console

https://wiki.odroid.com/odroid-n1/hardware/expansion_connectors#expansion_connectors

> - What are the voltage specs on the N1 uart?
>  Will any of my Odroid UART boards work on it?
>  
>  You can use our USB-UART kit with a 4pin connector on the N1.
>  CP2104: USB to Serial UART bridge IC with 1.8~3.3V interface
>  
>  http://www.hardkernel.com/main/products/prdt_info.php?g_code=G134111883934

> - Ok I'm connected and have serial console access
>  `screen /dev/ttyUSB0 115200,n,8,1`
 
_____________________

Info Links:
 
 https://wiki.odroid.com/odroid-n1/hardware/expansion_connectors#expansion_connectors
 
 https://dn.odroid.com/RK3399/Schematic/N1_MAIN_REV0.1_20170929.pdf
 
 https://forum.odroid.com/viewtopic.php?f=153&t=30094#p214973
 
 https://forum.odroid.com/viewtopic.php?f=153&t=30130
 
 https://forum.odroid.com/viewtopic.php?f=153&t=30072
 
 TXB0102DCUT: http://www.ti.com/lit/ds/symlink/txb0102.pdf

Fourdee, 2018-03-23:
Alpha test image available: https://github.com/Fourdee/DietPi/issues/1519#issue-297886100

Fourdee, 2018-04-29:
Re-opened to remind me to check serial again:
https://github.com/Fourdee/DietPi/issues/1519#issuecomment-368595231

Fourdee, 2018-05-06:
Completed (for now), sent my dev sample to @midwan 😃 

Will order a new one and re-do image when mass production starts.

closed 2018-05-06 by Fourdee

MichaIng, 2019-03-08:
@Fourdee 
Was this ever really released with mass production?
Now superseded by Odroid N2: https://wiki.odroid.com/odroid-n2/odroid-n2
And we have zero Odroid N1 users: https://dietpi.com/survey/ 🤔

midwan, 2019-03-08:
@MichaIng 
It wasn't, it was aborted before it was mass produced due to high costs.
The N2 which just came out replaces it, like you said. :)

MichaIng, 2019-03-08:
@midway
Thanks for clarification.

So we can replace the download slot and hardware ID in code with N2, once we get one and are able to create an image for it.

Fourdee, 2019-03-08:
@MichaIng 

Yep 👍 
Odroid N2 already arrived (dev sample). Waiting on ARMbian/Meveric pre-images. 

I could simply use the kernel from current test image and switch to Debian, however, would rather wait for a pre-image we can roll out that supports future kernel updates.

MichaIng, 2019-03-08:
@Fourdee 
Thank is great. I opened a new issue to work on this: https://github.com/MichaIng/DietPi/issues/2623
Meveric image already exists, but still WIP and lacks graphics driver.