MichaIng/DietPi #1520: Certbot: error raised while starting configuration

types140, 2018-02-16:
#### Required Information:
- DietPi Version 6.1
- SBC Device RPi3
- Power supply used 5V 2.5A
- SD card used Kingston 16GB Class 10
- Distro Linux raspiweb 4.9.59+ #1047 Sun Oct 29 11:47:10 GMT 2017 armv6l GNU/Linux (Debian Stretch)

#### Additional Information (if applicable):
- Software title Certbot
- Can this issue be replicated on a fresh installation of DietPi?
Of course. I used the headless automatic installation with the following packages:
-- LEAP Webserver - Nginx / MariaDB / PHP (ID 79)
-- phpMyAdmin - Web interface SQL admin tool (ID 90)
-- Node-Red (ID 122)
-- Certbot (ID 92)

#### Expected behaviour:
Successful installation of the both keys and (hopefully?) complete configuration of Nginx to use SSL.

#### Actual behaviour:
Installation raises the following exception:

Saving debug log to /var/log/letsencrypt/letsencrypt.log
Obtaining a new certificate
Performing the following challenges:
Client with the currently selected authenticator does not support any combination of challenges that will satisfy the CA.

I had a look at this logfile - its empty.

#### Steps to reproduce:
Headless automatic installation with the installed packages written above. No later configuration.
After that start dietpi-launcher and use DietPi-LetsEncrypt with the following settings:
Redirect: Enabled
Auto Renew: Enabled
Key Size: 4096

#### Did you submit a ```dietpi-bugreport```?
8a14ebf8-4429-4acb-9f41-423a80d0dcbf-1

#### Extra details:
I changed the following settings in the dietpi.txt befor starting the raspberry:
AUTO_SETUP_NET_HOSTNAME=raspiweb
AUTO_SETUP_TIMEZONE=Europe/Berlin
AUTO_SETUP_LOCALE=de_DE.UTF-8
AUTO_SETUP_KEYBOARD_LAYOUT=de

I think the issue raised because of [this](https://community.letsencrypt.org/t/solution-client-with-the-currently-selected-authenticator-does-not-support-any-combination-of-challenges-that-will-satisfy-the-ca/49983) change by lets encrypt. Certbot needs an update by a minimum of up to 0.21.0.

MichaIng, 2018-02-16:
@types140
Jep this is known issue, as LetsEncrypt block one of their authentication methods, used by default by Nginx and Apache module. We fixed this with v6.2 by changing default authentication method: https://github.com/Fourdee/DietPi/pull/1459
Will be released soon, planned this week, so maybe Sunday. 

To fix now, do:
`certbot -a webroot -w /var/www/ -i nginx`

types140, 2018-02-17:
@MichaIng 
Whoops didn't see that already raised issue. Sorry.
Thank you for that workaround - it worked like a charm!

closed 2018-02-17 by types140