MichaIng/DietPi #1523: Error uninstalling VNC

aaro668, 2018-02-17:
#### Required Information:
- DietPi Version | v6.1
- SBC Device     | OrangePi Zero (armv7l) (index=32)
- Distro         | stretch (index=4)
- Command        | G_AGP: xcompmgr xterm xinit xauth xserver-xorg dbus-x11 xfonts-base x11-xserver-utils x11-common x11-utils
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libice6 : Depends: x11-common but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

Fourdee, 2018-02-17:
@aaro668 

Hi,

Thanks for the report 👍 

The error is for uninstall of Xserver. During the uninstall menu, which items did you select for uninstall?
eg:
- VNC4
- Xserver

aaro668, 2018-02-17:
VNC 4
LXDE
Xserver
ALSA

Aaron Martin
Northway Metals Inc.
aaron@northwaymetals.com<mailto:aaron@northwaymetals.com>
Sales: sales@northwaymetals.com<mailto:sales@northwaymetals.com>

aaro668, 2018-02-17:
I also tried uninstalling only VNC4 and got this one:
#### Required Information:
- DietPi Version | v6.1
- SBC Device     | OrangePi Zero (armv7l) (index=32)
- Distro         | stretch (index=4)
- Command        | G_AGP: realvnc-vnc-server
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package realvnc-vnc-server
```

Fourdee, 2018-02-17:
@aaro668 

Thanks 👍 

> E: Unable to locate package realvnc-vnc-server

This is resolved for v6.2 (release tomorrow). I'll try to replicate the other issue.

Fourdee, 2018-02-17:
🈴 Confirmed RPi uninstall of:
VNC 4
LXDE
Xserver
ALSA

aaro668, 2018-02-17:
Ok, so how do I un-install from my system?

Fourdee, 2018-02-17:
@aaro668 

Please wait.

Fourdee, 2018-02-17:
@aaro668 

Uninstall workaround:
```
dietpi-software uninstall 23 28 5 6
```

aaro668, 2018-02-17:
Ok, thanks, should I run cleanup after un-install?

Fourdee, 2018-02-17:
@aaro668 
No need, DietPi will run ```apt-get autoremove --purge``` automatically at the end.

aaro668, 2018-02-17:
Using workaround (dietpi-software uninstall 23 28 5 6)
I get this error yet:
#### Required Information:
- DietPi Version | v6.1
- SBC Device     | OrangePi Zero (armv7l) (index=32)
- Distro         | stretch (index=4)
- Command        | G_AGP: realvnc-vnc-server
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package realvnc-vnc-server
```

Fourdee, 2018-02-17:
@aaro668 

Ah yes, this requires v6.2, please update the global script temporarily for this uninstall:
```
mv /DietPi/dietpi/func/dietpi-globals /DietPi/dietpi/func/dietpi-globals.bak
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/func/dietpi-globals -O /DietPi/dietpi/func/dietpi-globals
chmod +x /DietPi/dietpi/func/dietpi-globals
dietpi-software uninstall 23 28 5 6
mv /DietPi/dietpi/func/dietpi-globals.bak /DietPi/dietpi/func/dietpi-globals
```

Fourdee, 2018-02-17:
Moved to 6.3:
- We require an ordering system for Uninstalls
- Installs have this currently via location in script
- Maybe an additional array for "install order index" assigned to each software title, we can use to order install/uninstall.

aaro668, 2018-02-17:
Got it un-installed with this:
mv /DietPi/dietpi/func/dietpi-globals /DietPi/dietpi/func/dietpi-globals.bak
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/func/dietpi-globals -O /DietPi/dietpi/func/dietpi-globals
chmod +x /DietPi/dietpi/func/dietpi-globals
dietpi-software uninstall 23 28 5 6
mv /DietPi/dietpi/func/dietpi-globals.bak /DietPi/dietpi/func/dietpi-globals

thanks!

aaro668, 2018-02-17:
Since doing the uninstall the blynkserver service isnt running successfully, any pointers why this is happening?

DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] cron   active (running) since Sat 2018-02-17 10:56:21 EST; 7min ago
[  OK  ] proftpd        active (running) since Sat 2018-02-17 10:56:21 EST; 7min ago
[FAILED] blynkserver    failed (Result: exit-code) since Sat 2018-02-17 10:56:21 EST; 7min ago
● blynkserver.service - Blynk Server
   Loaded: loaded (/etc/systemd/system/blynkserver.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2018-02-17 10:56:21 EST; 7min ago
  Process: 11405 ExecStart=/usr/bin/java -jar /etc/blynkserver/server.jar -serverConfig /mnt/dietpi_userdata/blynk/server.properties (code=exited, status=203/EXEC)
 Main PID: 11405 (code=exited, status=203/EXEC)

Feb 17 10:56:21 DietPi-Clare systemd[1]: Started Blynk Server.
Feb 17 10:56:21 DietPi-Clare systemd[1]: blynkserver.service: Main process exited, code=exited, status=203/EXEC
Feb 17 10:56:21 DietPi-Clare systemd[1]: blynkserver.service: Unit entered failed state.
Feb 17 10:56:21 DietPi-Clare systemd[1]: blynkserver.service: Failed with result 'exit-code'.

Fourdee, 2018-02-17:
@aaro668 

Could be a pre-req from the uninstalled items.

What happens when you run?:
```
/usr/bin/java -jar /etc/blynkserver/server.jar -serverConfig /mnt/dietpi_userdata/blynk/server.properties
```

aaro668, 2018-02-17:
root@DietPi-Clare:~# service blynkserver stop
root@DietPi-Clare:~# /usr/bin/java -jar /etc/blynkserver/server.jar -serverConfig /mnt/dietpi_userdata/blynk/server.properties
-bash: /usr/bin/java: No such file or directory
root@DietPi-Clare:~#

Fourdee, 2018-02-17:
@aaro668 

One of the uninstalled software items most likely removed Java, due to APT req's (out of our control).

Easiest solution, reinstall blynkserver (which automatically pulls in java):
```
dietpi-software reinstall 131
```

aaro668, 2018-02-17:
ok, but I have apps and settings already for Blynkserver, will those be overwritten? I would prefer not to have to set everything up again.....

Fourdee, 2018-02-17:
@aaro668 

The only thing I can see that will be overwritten during reinstall is ```"$G_FP_DIETPI_USERDATA"/blynk/server.properties``` file, however, to avoid loss, please create a full system backup before hand:
```
dietpi-backup
```

aaro668, 2018-02-17:
Thanks, it worked

MichaIng, 2018-02-25:
I will close in favour of: https://github.com/Fourdee/DietPi/issues/1462
Collect ideas there how to realize specific uninstall order.

Fourdee, 2018-03-01:
> I will close in favour of: #1462

😃

closed 2018-03-01 by Fourdee