MichaIng/DietPi #1524: DietPi-Software | PiHole: dnsmasq fails to run

Fourdee, 2018-02-17:
```
root@DietPi:~# systemctl status dnsmasq -l
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled)
  Drop-In: /run/systemd/generator/dnsmasq.service.d
           └─50-dnsmasq-$named.conf, 50-insserv.conf-$named.conf
   Active: failed (Result: exit-code) since Sat 2018-02-17 16:14:51 GMT; 6min ago
  Process: 486 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=1/FAILURE)

Feb 17 16:14:51 DietPi dnsmasq[486]: dnsmasq: extraneous parameter at line 37 of /etc/dnsmasq.d/01-pihole.conf
```

Fourdee, 2018-02-17:
offending line
```
log-queries=extra
```

Fourdee, 2018-02-17:
Cause, breaks Jessie support:
https://github.com/pi-hole/pi-hole/commit/5c2dcb5e74f464e2ef5ec1afd67b569ea180acde

Fourdee, 2018-02-17:
We'll disable Jessie installs for now (https://github.com/pi-hole/pi-hole/commit/5c2dcb5e74f464e2ef5ec1afd67b569ea180acde#commitcomment-27626937), pending release of FTL 4.0.

LieDanG, 2018-02-17:
I noticed this too, if you want to use pi-hole (who doesn't, it's awesome!) just comment that line, it will work but you won't be able to see stats

Fourdee, 2018-02-18:
Notes for workaround attempt:
- PiHole install script returns a exit status of >0 with Jessie (fail).
- Chance of ```sed -i '/^log-queries/c\log-queries' /etc/dnsmasq.d/01-pihole.conf``` change being overwritten during updates
- Stats not functional, expect bug reports for this.

I'll leave this as disabled for now, workaround is too messy and will cause more problems in the end for user.

breneser, 2018-02-18:
@Fourdee  Maybe you could checkout version v3.2.1 to install ?

As a related question, is there going to be a dist upgrade in the future ?

Fourdee, 2018-02-18:
> Maybe you could checkout version v3.2.1 to install ?

Not for v6.2 release (wrapping it up for today release). But if we get enough "complaints" regarding the install being disabled, i'll investigate.

> As a related question, is there going to be a dist upgrade in the future ?

Yes, but we need to wait for Meveric to update his repo to Stretch. Without which, Kodi and GPU packages are not functional in Stretch.

SuBLiNeR, 2018-02-18:
Answer and temporary solution for Debian (Jessie) is given here: 
https://discourse.pi-hole.net/t/pi-hole-v3-3-released-its-extra-special/7159

breneser, 2018-02-18:
Thanks, I had to disable logging. 

Installing dnsmasq 2.76 did not work for me as I am running it on Odroid C2. Tried to download arm64 version from Stretch repo but bumbped into dependency problems. Open to suggestions about how to update dnsmasq on DIetpi Odroid C2 version.

SuBLiNeR, 2018-02-18:
Do you tried this: https://discourse.pi-hole.net/t/dnsmasq-fails-on-pi-hole-3-3/7161/13?u=jacob.salmela ?

breneser, 2018-02-18:
Thanks,I tried this partly :) 

I did downgrade to v3.2.1, but did not downgrade FTL version. Will give it a go now. I am hopeful :)

breneser, 2018-02-18:
Checking out FTL v2.13.2 did the trick. All seem to be working now :) Much appreciated.

SuBLiNeR, 2018-02-18:
@breneser You're welcome! 👍

MichaIng, 2018-02-25:
For reference:
- It seems possible to build dnsmasq 2.76 from source on Jessie: https://wiki.fogproject.org/wiki/index.php?title=ProxyDHCP_with_dnsmasq#Compiling_dnsmasq_2.76_if_you_need_uefi_support
- However confirmed just on Raspbian Jessie and Ubuntu 16.04 LTS, where our Odroid devices are on Debian Jessie.
- This manual workarounds with disable logging/downgrading are no proper stable end user solution.
- There are no dnsmasq 2.76 backports for Debian and manual installing the Stretch version brings a pig tail of Stretch dependencies, too much to test/guarantee proper work on Jessie.
_____________________
What we can do:
- Test source build method for dnsmasq 2.76 (or even 2.78?) on Debian Jessie, preferably on Odroid device directly.
- Wait for Meverics Odroid Stretch image to support Kodi with GPU acceleration, which is on the way: https://github.com/Fourdee/DietPi/issues/1552 > https://forum.odroid.com/viewtopic.php?f=138&t=27449&start=100#p216453
_____________________
- But the latter is no solution for current Odroid Jessie image users. Distro upgrade is always a risk, dietpi-backup just helps, if the device still works/boots up after failure. We have to go though our code to assure, that our variables/methods are switched correctly to Stretch way. Not everybody has possibility do to whole SD image backup.
- Especially existing non-APT package (source build/manual package installations) software titles are not upgraded automatically, would need reinstall with Stretch version/rebuild, outcome unsure, when automated.
- DietPi-Prep would be a safer solution to upgrade, but purges much user data/settings.
- We can encourage users to flash new Stretch images, but shortly after v6.0 forcing flash new image, we should not force users to do that, thus keep support for Jessie.
- Finally I would love to step by step include software specific data+settings backups. Would need some additional backup&restore scripts for each software title, included into dietpi-backup (only show up for installed titles), allowing to execute together or separately. Prototype for ownCloud/Nextcloud: https://github.com/Fourdee/DietPi/blob/dietpi-cloud-migration/dietpi-cloud-migration, https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L12361-L12408, https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L3841-L3846, https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L3841-L3846

MichaIng, 2018-07-27:
I mark this as closed. We hope for soon Pi-hole FTLDNS release which will re-enable Jessie support, AFAWK.

Let's collect Pi-hole related discussion here: https://github.com/Fourdee/DietPi/issues/1696
We will retest and rework our installer after mentioned release.

closed 2018-07-27 by MichaIng