MichaIng/DietPi #1526: OpenVPN: No such file or directory

types140, 2018-02-17:
**Required Information:**

    DietPi Version 6.1
    SBC Device RPi3
    Power supply used 5V 2.5A
    SD card used Kingston 16GB Class 10
    Distro Linux raspiweb 4.9.59+ #1047 Sun Oct 29 11:47:10 GMT 2017 armv6l GNU/Linux (Debian Stretch)

**Additional Information (if applicable):**

    Software title OpenVPN
    Can this issue be replicated on a fresh installation of DietPi?
    Of course. I used the headless automatic installation with the following packages:
    -- LEAP Webserver - Nginx / MariaDB / PHP (ID 79)
    -- phpMyAdmin - Web interface SQL admin tool (ID 90)
    -- Node-Red (ID 122)
    -- Certbot (ID 92) used with fix #1459 

**Expected behaviour:**

Successful installation.

**Actual behaviour:**

Installation raises the following exception:

  No /etc/openvpn/easy-rsa/openssl.cnf file could be found
  Further invocations will fail
**************************************************************
NOTE: If you run ./clean-all, I will be doing a rm -rf on /etc/openvpn/easy-rsa/keys
Using CA Common Name: DietPi_OpenVPN_Server
grep: /etc/openvpn/easy-rsa/openssl.cnf: Datei oder Verzeichnis nicht gefunden
pkitool: KEY_CONFIG (set by the ./vars script) is pointing to the wrong version of openssl.cnf: /etc/openvpn/easy-rsa/openssl.cnf
The correct version should have a comment that says: easy-rsa version 2.x
grep: /etc/openvpn/easy-rsa/openssl.cnf: Datei oder Verzeichnis nicht gefunden
pkitool: KEY_CONFIG (set by the ./vars script) is pointing to the wrong version of openssl.cnf: /etc/openvpn/easy-rsa/openssl.cnf
The correct version should have a comment that says: easy-rsa version 2.x
cp: der Aufruf von stat für '/etc/openvpn/easy-rsa/keys/DietPi_OpenVPN_Server.crt' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
cp: der Aufruf von stat für '/etc/openvpn/easy-rsa/keys/DietPi_OpenVPN_Server.key' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
cp: der Aufruf von stat für '/etc/openvpn/easy-rsa/keys/ca.crt' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
grep: /etc/openvpn/easy-rsa/openssl.cnf: Datei oder Verzeichnis nicht gefunden
pkitool: KEY_CONFIG (set by the ./vars script) is pointing to the wrong version of openssl.cnf: /etc/openvpn/easy-rsa/openssl.cnf
The correct version should have a comment that says: easy-rsa version 2.x
cat: /etc/openvpn/ca.crt: Datei oder Verzeichnis nicht gefunden
cat: /etc/openvpn/easy-rsa/keys/DietPi_OpenVPN_Client.crt: Datei oder Verzeichnis nicht gefunden
cat: /etc/openvpn/easy-rsa/keys/DietPi_OpenVPN_Client.key: Datei oder Verzeichnis nicht gefunden

> "Datei oder Verzeichnis nicht gefunden" is german and means "No such file or directory".

**Steps to reproduce:**

Headless automatic installation with the installed packages written above.
Only after-configuration was Certbot to create the certificates. Nothing more in any other packages.
Installation of OpenVPN (ID 97) manually after headless installation.

**Did you submit a dietpi-bugreport?**

8a14ebf8-4429-4acb-9f41-423a80d0dcbf-2
**Extra details:**

k-plan, 2018-02-18:
@types140 

> `DietPi Version 6.1`
> `SBC Device RPi3`
> `Debian Stretch`
> `OpenVPN`
> `/etc/openvpn/easy-rsa/openssl.cnf` file could be found

Please have a look here: https://github.com/Fourdee/DietPi/issues/1450

Should be solved with `DietPi v6.2`

Please wait for  version release or you can switch to testing branch, if you need it immediately: [OpenVPN setup seems broken in DietPi 6](http://dietpi.com/phpbb/viewtopic.php?f=11&t=2768#p10616)

Fourdee, 2018-02-18:
@k-plan 

Yep, this is resolved for v6.2, which will be released today. I'll mark this as closed. Please reopen if required.

closed 2018-02-18 by Fourdee

k-plan, 2018-02-18:
@Fourdee 

> @k-plan  
> 
> Yep, this is resolved for v6.2, which will be released today. I'll mark this as closed. Please reopen if required.

Do you really want to addess me? 😃  I don't think so.

Belief it was vectored to @types140