MichaIng/DietPi #1527: Aria2 webgui blank

kodirfan, 2018-02-18:
## issue:
Cant access aria2 webgui. Aria2 is running. Tried accessing http://localhost/aria2 but no luck. Any help?

Here is the log file aria2 running.

root@DietPi:~# systemctl status aria2 -l
● aria2.service - DietPi Aria2
   Loaded: loaded (/etc/systemd/system/aria2.service; disabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-02-18 04:01:49 EST; 5min ago
 Main PID: 312 (aria2c)
   CGroup: /system.slice/aria2.service
           └─312 /usr/bin/aria2c --enable-rpc --rpc-listen-all --rpc-secret=****** --dir=/mnt/dietpi_userdata/downloads

Feb 18 04:01:49 DietPi systemd[1]: Started DietPi Aria2.
Feb 18 04:01:49 DietPi aria2c[312]: 02/18 04:01:49 [NOTICE] IPv4 RPC: listening on TCP port 6800
Feb 18 04:01:49 DietPi aria2c[312]: 02/18 04:01:49 [NOTICE] IPv6 RPC: listening on TCP port 6800

#### Required Information:
- DietPi Version 6.1
- SBC Device (RPi 3)
- Power supply used 5v 2A 
- SD card used ( Sandisk ultra)

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

MichaIng, 2018-02-18:
@kodirfan 
Many thanks for your report. We will investigate and try to help you. Some infos:
- All non essential (network related etc.) services are by default disabled for DietPi to take control. Enabled means: `systemd`, Debian init system, enables it during boot based on its criteria. However we want to make sure, that we start services after our core components are loaded, thus control it ourself, currently via `/etc/rc.local` (after after v6.3 via new own service).
- For this reason you see `service; disabled` as output, but `Active: active (running)` is the important one.
- As far as I know, the web ui show-up shouldn't even depend on running Aria2 server, thus I guess there seems to be some issue with Lighttpd, perhaps in combination with PiHole. Lighttpd shows some warnings, but that should not break things.

To help you more effectively could you give some more information:
- Could you paste content of your: `/var/log/lighttpd/error.log`
- as well as: `/etc/lighttpd/lighttpd.conf`

MichaIng, 2018-02-18:
@kodirfan @Fourdee 
Could verify, also found blank Aria2 web ui on Stretch in combination with PiHole. Couldn't find any error, so logs are not necessary any more. However browser console could give some hint:
```
angular.min.js:6 Uncaught Error: [$injector:modulerr] http://errors.angularjs.org/1.6.0/$injector/modulerr?p0=webui&p1=TypeError%3A%20Cannot%20read%20property%20'Search'%20of%20undefined%0A%20%20%20%20at%20mergeTranslation%20(http%3A%2F%2F192.168.178.31%2Faria2%2Fjs%2Finit.js%3A22%3A19)%0A%20%20%20%20at%20http%3A%2F%2F192.168.178.31%2Faria2%2Fjs%2Finit.js%3A46%3A30%0A%20%20%20%20at%20Object.invoke%20(http%3A%2F%2F192.168.178.31%2Faria2%2Fjs%2Flibs%2Fangular.min.js%3A43%3A175)%0A%20%20%20%20at%20d%20(http%3A%2F%2F192.168.178.31%2Faria2%2Fjs%2Flibs%2Fangular.min.js%3A41%3A64)%0A%20%20%20%20at%20http%3A%2F%2F192.168.178.31%2Faria2%2Fjs%2Flibs%2Fangular.min.js%3A41%3A188%0A%20%20%20%20at%20q%20(http%3A%2F%2F192.168.178.31%2Faria2%2Fjs%2Flibs%2Fangular.min.js%3A7%3A351)%0A%20%20%20%20at%20g%20(http%3A%2F%2F192.168.178.31%2Faria2%2Fjs%2Flibs%2Fangular.min.js%3A40%3A488)%0A%20%20%20%20at%20eb%20(http%3A%2F%2F192.168.178.31%2Faria2%2Fjs%2Flibs%2Fangular.min.js%3A45%3A56)%0A%20%20%20%20at%20c%20(http%3A%2F%2F192.168.178.31%2Faria2%2Fjs%2Flibs%2Fangular.min.js%3A21%3A19)%0A%20%20%20%20at%20Object.Lc%20%5Bas%20bootstrap%5D%20(http%3A%2F%2F192.168.178.31%2Faria2%2Fjs%2Flibs%2Fangular.min.js%3A21%3A332)
    at angular.min.js:6
    at angular.min.js:41
    at q (angular.min.js:7)
    at g (angular.min.js:40)
    at eb (angular.min.js:45)
    at c (angular.min.js:21)
    at Object.Lc [as bootstrap] (angular.min.js:21)
    at HTMLDocument.<anonymous> (init.js:66)
    at i (jquery-2.2.4.min.js:2)
    at Object.fireWith [as resolveWith] (jquery-2.2.4.min.js:2)
```
- ownCloud, Nextcloud and Minio web uis work fine btw., so it is not related to a general webserver issue.

Will test after uninstalling PiHole...

kodirfan, 2018-02-18:
Hi MichaIng, thanks so much for looking into it. I uninstalled lighttpd and installed nginx as lighttpd was showing some warnings ( I am newbie, do not know much if the error is significant) but it is still the same way. I will look forward to a solution.

Also, I noticed pihole has an update for it webgui but I can not update either by command line or from dietpi software launcher. 

Best,

kanzetu, 2018-02-19:
This is the bug of webui-aria2
see this https://github.com/ziahamza/webui-aria2/pull/436
Just fix the file in /var/www/aria2/js/translate/id_ID.js

correct this line that on the first few line
translations.en_US = {  // replace en_US to ll_CC, examples: zh_CN, de_AT.
to
translations.id_ID = {  // replace en_US to ll_CC, examples: zh_CN, de_AT.

Fourdee, 2018-02-20:
@kodirfan @MichaIng 

PR fix from Aira2 is now in their master. 👍 

Quick test of RPi3 install:
- 🈯️ 

Solution for those affected, reinstall aria2:
```
dietpi-software reinstall 132
```

Marking as closed.

closed 2018-02-20 by Fourdee