MichaIng/DietPi #1529: ProFtp login

symbios24, 2018-02-18:
Hello, ProFtp Fileserver is not working with  ftp://root:dietpi@192.168.0.10 it does not accept the password, alternative is working with ftp://dietpi@192.168.0.100 and password dietpi

Can you fix this ?

Thanks

symbios24, 2018-02-18:
i forgot im using a raspberry pi with latest DietPi Stretch.

Fourdee, 2018-02-18:
@symbios24 

Hi,

Please confirm ftp://root:dietpi@192.168.0.100 was tested (you mentioned a different IP of .10).
ftp://root:dietpi@192.168.0.100 won't work with all programs (eg: chrome doesnt accept it and still prompts for user/pw).

In which case, simply try:
ftp://192.168.0.100

I'am unable to replicate any issues here (using the IP address of .105 to match my system):
ftp://root:dietpi@192.168.0.105

[image]

symbios24, 2018-02-18:
sorry i meaN  ftp://dietpi@192.168.0.100 it was a typo, i  tried only with chrome, i did not know it does not accept it.so in firefox the ftp://root:dietpi@192.168.0.100 will work correct?

thanks

Fourdee, 2018-02-18:
@symbios24 

> so in firefox the ftp://root:dietpi@192.168.0.100 will work correct?

Unsure. If in doubt, simply use the address without user/pw, then enter them manually in when prompted.

symbios24, 2018-02-18:
if i do this when i try to copy files in a folder i get an access denied error, i dont get any prompt for password.
alternative  ftp://dietpi@192.168.0.100 works with files copy

symbios24, 2018-02-19:
I did all this from a linux machine from the file manager,from the browser it opens fine with the ftp://root:dietpi@192.168.0.100 but i cannot add files like this only view the contends of the folder,you need to do this from a file-manager/windows explorer
 
So when i do this from a linux file manager in ubuntu mate (ftp://192.168.0.100) when i try to copy files in a folder i get an access denied error, i dont get any prompt for password.

Alternative i tried ftp://dietpi@192.168.0.100 works fine with files copy. it prompts for password(dietpi)

ftp://root:dietpi@192.168.0.100 is not working it asks for password i put dietpi it does not accept it and still prompts for the pass.

k-plan, 2018-02-22:
@symbios24 

Question is answered? Can we close this issue?

@Fourdee 

Should we better  set `RootLogin off` in `/etc/proftpd/proftpd.conf` as default?
https://github.com/Fourdee/DietPi/blob/master/dietpi/conf/proftpd.conf#L37

symbios24, 2018-02-22:
Yes thank you

Fourdee, 2018-02-22:
@k-plan 

> Should we better set RootLogin off in /etc/proftpd/proftpd.conf as default?

Yep, we can change to ```dietpi``` user as ```/mnt/dietpi_userdata``` has those permissions.

Fourdee, 2018-02-22:
🈯️ Tested via Windows explorer

Completed.

closed 2018-02-22 by Fourdee