MichaIng/DietPi #1530: Dietpi For raspberrypi 3B，it’s  64-bit？

dyxang, 2018-02-18:
Raspberry pie uses 64 bit processors，I just want to know
My mother tongue is not English, and the above is **copied from Google translation**. I'm sorry。

k-plan, 2018-02-18:
@dyxang 

> Dietpi For raspberrypi 3B，it’s 64-bit？

In short: No

Why? Because it's only one RPi image (based on Rasbian) which support all RPi hardware revisions since 2012

dyxang, 2018-02-19:
@k-plan  thanks

MichaIng, 2018-02-24:
@dyxang 
Check our the discussion about this on Raspbian forum: https://www.raspberrypi.org/forums/viewtopic.php?t=192321

We will follow them. As fast as they release a x64 repository for RPi3 (or 4 then, who knows), we will do as well. But indeed the benefits seem to be marginal and do now weight out the effort for now.

dyxang, 2018-02-24:
@MichaIng Ok,now I know.Thanks.
==
My parent languageis not English, and the above is **copied** **from** **Google** **translation.** I'm sorry for that。

MichaIng, 2018-02-24:
@dyxang 
No worries about language here, many of us are learning by doing 😃.

dyxang, 2018-03-02:
@k-plan @MichaIng Thank you-all very much.A week later, I found another 64-bit system project——[ubuntu64-rpi](https://github.com/chainsx/ubuntu64-rpi)(It is **Chinese** project,but it included 64-bit **Centos**)。

- The root directory built directly based on ubuntu-Base-16.04-arm64
- About ubuntu,/etc/apt/sources.list have been modified. 
eg. **archive.raspberrypi.org** be modified into **mirrors.tuna.tsinghua.edu.cn**(Chinese Web site)
- About ubuntu,
username = ubuntu
password = ubuntu

--
My parent languageis not English, and the above is copied from Google translation. I'm sorry for that。

MichaIng, 2018-03-02:
@dyxang 
Thanks for sharing.

I guess since all the documentation is in Chinese language as well, and as DietPi-Prep does not official support Ubuntu base images, this is no option. Anyway I see no reason to move to a 3rd party project, that might bring instabilities and other risks and most importantly does not have a large user/support space as Raspbian has.

Although, with some adjustments, DietPi-Prep installs fine on Ubuntu: https://github.com/Fourdee/DietPi/issues/1557
But this is more an "if nothing else as Ubuntu base image is available" worst case solution 😉.

dyxang, 2018-03-02:
@MichaIng  Well,English readme.md is ubuntu64-rpi/README-EN.md.
Just mention it.Thank you for your reply

closed 2018-03-02 by dyxang

MichaIng, 2018-03-02:
@dyxang 
Ah apologize, seems I had a too rough look at this. Still, not an option for DietPi from my point of view, but interesting to know. Maybe an option for others and great to have people offering x64 RPi images generally to bring forward this for hopefully soon all devices.