MichaIng/DietPi #1533: DietPi not booting.

ronakmd, 2018-02-19:
## Creating a bug report/issue:

DietPi Version | cat /DietPi/dietpi/.version?>>DietPi_v6.0_RPi-ARMv6-Stretch

SBC Device (EG: RPi 3)?>>>>Rpi MOdel B (armv71) index=3

Power supply used (EG: 5v 1A RAVpower)?>>>From Adapter 5v 1.5A

SD card used (EG: Sandisk ultra)?>>>Samsung 16

#### Expected behaviour:

[image]

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

As mentioned to me earlier i tried enabling wifi and configuring again.
These were the steps.
For WiFi, you will need to enable it:

When the error prompt appears (as per screenshot), press ENTER (OK) or ESC.
When asked to configure network in DietPi-Config, select OK.
Select Networking Options: Adapters
Select WiFi and confirm enable
Select WiFi again and enter the WiFi details as needed.
Save changes when your done, and exit dietpi-config
Setup will now resume.

But still i am getting the same error.
Can you also suggest DietPi image that comes pre-installed with desktop.

Thankyou.

ronakmd, 2018-02-20:
> @ronakmd <https://github.com/ronakmd>
>
> Please verify the WiFi key entered is a match.
>
> Does the WiFi key contain any "strange" characters (eg: $#/)?

Hi,
Password is simple with 5chars and 3 numbers and no special characters.
Yes I tried with http:/www.google.com but I get failed as status in WiFi
test.
But wifi shows connected and enabled.
Please let me know.
Thankyou.

Do you receive the same error in the connection test? (dietpi-config >
Networking options: Adapters > test

Please also try with http://google.com

DietPi is a netinstall image, we do not offer any pre-installed images with
desktop at this time.

<https://github.com/Fourdee/DietPi/issues/1533#issuecomment-366825365>, or mute
the thread

<http://www.somaiya.edu>             <http://www.somaiya.edu/kjsmc>  
<http://www.nareshwadi.org> <http://www.somaiya.edu>  
<http://www.helpachild.in> <http://nareshwadi.org>

ronakmd, 2018-02-20:
Sorry its http:/google.com
Mistakenly written www
Thankyou

On 20 Feb 2018 7:02 a.m., "RONAK DEDHIA" <ronak.md@somaiya.edu> wrote:

Hi,
Password is simple with 5chars and 3 numbers and no special characters.
Yes I tried with http:/www.google.com but I get failed as status in WiFi
test.
But wifi shows connected and enabled.
Please let me know.
Thankyou.

<http://www.somaiya.edu>             <http://www.somaiya.edu/kjsmc>  
<http://www.nareshwadi.org> <http://www.somaiya.edu>  
<http://www.helpachild.in> <http://nareshwadi.org>

Fourdee, 2018-02-20:
@ronakmd 

Ok, google.com is failing aswell.

It could be your router is WEP only (which we do not support officially), DietPi is default to WPA. But even then, you shouldn't be able to obtain an IP address.

Some things to try/check:
- Try moving the device closer to the router and reconnecting the WiFi (range)
- Reboot the router, then the RPi
- Verify WPA is used and enabled in your router's web interface

k-plan, 2018-02-20:
After this please:

- ping ip address of your router:  `root@RPi:~# ping "ip-address-of-your-router"`

- look at you RPi ip address:  `root@RPi:~# ip address`

- look at your RPi Kernel IP routing table: `root@RPi:~# route`

- look at your RPi DNS server:  `root@RPi:~# cat /etc/resolv.conf`

ronakmd, 2018-02-21:
Hi Dan,
I am using my mobile as a router.So the hotspot is Wpa2 security and
strength of signal is good enough.
It shows connected and enabled.I also get an ip address but after saving
changes and exiting to resume changes i get the error i mailed earlier.
Please let me know.
Thankyou.

-- 
Thank you.

Regards,
Ronak Dedhia.

Fourdee, 2018-02-21:
@ronakmd 

If the IP addresses are obtained. We can assume the connection to the phone is established. This may simply be a DNS issue and/or configuration issue with your phone.

When the error appears, go into ```dietpi-config``` > Networking (as previously). 
In the WiFi settings
- change mode to STATIC
- select "Copy Current address to Static",
- select "Static DNS"
- select "Google"
- select "Save Changes (Restart Networking)"

Once done, exit dietpi-config and resume setup.

failing above, please try on a non-phone/hotspot router.

k-plan, 2018-02-27:
No reply so far,  i'll mark as closed. 
However, please reopen if required.
Thanks.

closed 2018-02-27 by k-plan