MichaIng/DietPi #1534: Cannot detect external hard disk

kanzetu, 2018-02-19:
#### Issue: Cannot detect external hard disk

#### Required Information:
- DietPi Version | ```6.2 ```
- SBC Device :RPi 3 Model B (armv7l)

####  Information:

I am trying to mount my external hard disk via DietPi-Drive_Manager, but my HD can't be found

 Result of fdisk -l

>Disk /dev/mmcblk0: 14.7 GiB, 15720251392 bytes, 30703616 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x6f0c83bc

>Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       8192    93236    85045 41.5M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      94208 30703615 30609408 14.6G 83 Linux

 Result of mount

> /dev/mmcblk0p2 on / type ext4 (rw,noatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=495548k,nr_inodes=123887,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=27,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
tmpfs on /DietPi type tmpfs (rw,nosuid,nodev,noatime,size=10240k)
tmpfs on /var/log type tmpfs (rw,nosuid,nodev,noatime,size=20480k)
/dev/mmcblk0p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
nfsd on /proc/fs/nfsd type nfsd (rw,relatime)

/etc/fstab

> proc            /proc           proc    defaults											0 0
tmpfs 			/tmp  			tmpfs 	defaults,noatime,nodev,nosuid,mode=1777				0 0
tmpfs 			/var/log 		tmpfs 	defaults,size=20m,noatime,nodev,nosuid,mode=1777	0 0
tmpfs 			/DietPi 		tmpfs 	defaults,size=10m,noatime,nodev,nosuid,mode=1777	0 0

>#Internal Drives---------------------------------------------------
PARTUUID=6f0c83bc-01    /boot    auto    defaults,noatime    0 0
PARTUUID=6f0c83bc-02    /    auto    defaults,noatime    0 0

Hard disk information
diskutil info on my mac

>    Device Identifier:        disk6
   Device Node:              /dev/disk6
   Whole:                    Yes
   Part of Whole:            disk6
   Device / Media Name:      External

>   Volume Name:              Not applicable (no file system)
   Mounted:                  Not applicable (no file system)
   File System:              None

>   Content (IOContent):      FDisk_partition_scheme
   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 USB
   SMART Status:             Not Supported

>   Disk Size:                500.1 GB (500107861504 Bytes) (exactly 976773167 512-Byte-Units)
   Device Block Size:        512 Bytes

>   Read-Only Media:          No
   Read-Only Volume:         Not applicable (no file system)

>   Device Location:          External
   Removable Media:          Fixed

>   Virtual:                  No
   OS 9 Drivers:             No
   Low Level Format:         Not supported

and also

>   Device Identifier:        disk6s1
   Device Node:              /dev/disk6s1
   Whole:                    No
   Part of Whole:            disk6

>   Volume Name:              exfat
   Mounted:                  Yes
   Mount Point:              /Volumes/exfat

>   Partition Type:           Windows_NTFS
   File System Personality:  ExFAT
   Type (Bundle):            exfat
   Name (User Visible):      ExFAT

>   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 USB
   SMART Status:             Not Supported
   Volume UUID:              CEAAB188-189F-38A5-A60E-D627D04DF6D6

>   Disk Size:                500.1 GB (500107860480 Bytes) (exactly 976773165 512-Byte-Units)
   Device Block Size:        512 Bytes

>   Volume Total Space:       500.1 GB (500091060224 Bytes) (exactly 976740352 512-Byte-Units)
   Volume Used Space:        9.0 MB (9043968 Bytes) (exactly 17664 512-Byte-Units) (0.0%)
   Volume Available Space:   500.1 GB (500082016256 Bytes) (exactly 976722688 512-Byte-Units) (100.0%)
   Allocation Block Size:    131072 Bytes

>   Read-Only Media:          No
   Read-Only Volume:         No

>   Device Location:          External
   Removable Media:          Fixed

I try to format to ext4, exFAT or NTFS, both cannot be detected
And I also try to connect other harddisks but also cannot detected

k-plan, 2018-02-19:
Hi,

can you please connect your usb drive to RPi and provide:

`root@RPi-Zero:~# lsusb`

`root@RPi-Zero:~# lsblk -f`

`root@RPi-Zero:~# dmesg`

kanzetu, 2018-02-19:
Thank you for reply!

lsusb

> Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lsblk -f

> NAME        FSTYPE LABEL  UUID                                 MOUNTPOINT
mmcblk0
|-mmcblk0p1 vfat   boot   CDD4-B453                            /boot
`-mmcblk0p2 ext4   rootfs 72bfc10d-73ec-4d9e-a54a-1cc507ee7ed2 /

dmesg

> [    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.59-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1047 SMP Sun Oct 29 12:19:23 GMT 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3dc00000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 255488
[    0.000000] free_area_init_node: node 0, pgdat 80c6f380, node_mem_map bd336000
[    0.000000]   Normal zone: 2246 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 255488 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @be5b0000 s25600 r8192 d23552 u57344
[    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 253242
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=720 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 console=ttyS0,115200 root=PARTUUID=6f0c83bc-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 991096K/1021952K available (7168K kernel code, 486K rwdata, 2012K rodata, 1024K init, 770K bss, 22664K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
                   lowmem  : 0x80000000 - 0xbe600000   ( 998 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c79834   ( 487 kB)
                      .bss : 0x80c7b000 - 0x80d3b964   ( 771 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000023] Switching to timer-based delay loop, resolution 52ns
[    0.000302] Console: colour dummy device 80x30
[    0.001213] console [tty1] enabled
[    0.001257] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001325] pid_max: default: 32768 minimum: 301
[    0.001655] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001698] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002731] Disabling memory control group subsystem
[    0.002833] CPU: Testing write buffer coherency: ok
[    0.002896] ftrace: allocating 22392 entries in 66 pages
[    0.050035] CPU0: update cpu_capacity 1024
[    0.050087] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.050149] Setting up static identity map for 0x100000 - 0x100034
[    0.052042] CPU1: update cpu_capacity 1024
[... 189 lines cut ...]
[    8.756197] NET: Registered protocol family 31
[    8.756203] Bluetooth: HCI device and connection manager initialized
[    8.757795] Bluetooth: HCI socket layer initialized
[    8.757815] Bluetooth: L2CAP socket layer initialized
[    8.757861] Bluetooth: SCO socket layer initialized
[    8.805623] Bluetooth: HCI UART driver ver 2.3
[    8.805631] Bluetooth: HCI UART protocol H4 registered
[    8.805635] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    8.805796] Bluetooth: HCI UART protocol Broadcom registered
[    9.037736] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    9.037746] Bluetooth: BNEP filters: protocol multicast
[    9.037764] Bluetooth: BNEP socket layer initialized
[   10.073524] PPP generic driver version 2.4.2
[   10.076651] NET: Registered protocol family 24
[   10.109689] l2tp_core: L2TP core driver, V2.0
[   10.113645] l2tp_netlink: L2TP netlink interface
[   10.118133] l2tp_ppp: PPPoL2TP kernel driver, V2.0
[... cut ...] .swb.de).
[   21.256427] NFSD: starting 90-second grace period (net 80c65340)
[   25.041079] brcmfmac: power management disabled
[   34.853765] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   47.042747] usb 1-1.4: new high-speed USB device number 4 using dwc_otg
[   73.671408] usb 1-1.4: new high-speed USB device number 5 using dwc_otg
[  100.291350] usb 1-1.4: new high-speed USB device number 6 using dwc_otg
[  126.921355] usb 1-1.4: new high-speed USB device number 7 using dwc_otg

k-plan, 2018-02-19:
Hmmm ...

- wrong PSU (min 2.5A / 5.1V) to power RPi ?

- faulty USB cable to power RPi ?

- faulty USB cable to power your USB HDD ?

- our USB HDD  need a externel PSU to power up or a Y-USB cable ?

closed 2018-02-22 by Fourdee