MichaIng/DietPi #1535: DietPi 6.2 | Update available

k-plan, 2018-02-19:
## Creating a bug report/issue:

```
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
[  OK  ] Root access verified.
cat: /tmp/find_network_index: No such file or directory
cat: /tmp/find_network_index: No such file or directory
cat: /tmp/find_network_index: No such file or directory
cat: /tmp/find_network_index: No such file or directory
cat: /tmp/find_network_index: No such file or directory
cat: /tmp/find_network_index: No such file or directory
cat: /tmp/find_network_index: No such file or directory
cat: /tmp/find_network_index: No such file or directory
 ───────────────────────────────────────
 DietPi     | Update available
 ───────────────────────────────────────
 v6.2       | RPi Zero W (armv6l)
 ───────────────────────────────────────
 IP eth0    | 192.168.0.250
 ───────────────────────────────────────

 Created by : Daniel Knight
 Web        : http://DietPi.com
 Twitter    : http://twitter.com/dietpi_
 Donate     : http://goo.gl/pzISt9
 DietPi's web hosting is powered by: MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 dietpi-update    = Run now to update DietPi (from v6.2 to v6.1).
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.
```

```
root@RPi-Zero:~# dietpi-update 2
[  OK  ] Root access verified.
[ INFO ] DietPi-Drive_Manager | Checking RootFS R/W access. Please wait...
[  OK  ] DietPi-Drive_Manager | RootFS R/W access.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking Mirror : http://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100     4  100     4    0     0     43      0 --:--:-- --:--:-- --:--:--    44
[  OK  ] DietPi-Update | Using update server: http://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6
```

```
root@RPi-Zero:~# dietpi-update 1
[  OK  ] Root access verified.
[ INFO ] DietPi-Drive_Manager | Checking RootFS R/W access. Please wait...
[  OK  ] DietPi-Drive_Manager | RootFS R/W access.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking Mirror : http://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100     4  100     4    0     0     44      0 --:--:-- --:--:-- --:--:--    44
[  OK  ] DietPi-Update | Using update server: http://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6

[  OK  ] DietPi-Update | No updates required, your DietPi installation is up to date.

[ INFO ] DietPi-Update | Current Version : 6.2
[ INFO ] DietPi-Update | Server Version  : 6.2
```

```
root@RPi-Zero:~# cat /DietPi/dietpi/.version
6
2
```

```
root@RPi-Zero:~# cat /DietPi/dietpi/.update_stage
0
```

```
root@RPi-Zero:~# cat /DietPi/dietpi/server_version-6
6
2
```

`root@RPi-Zero:~# exit`

.... and re-login:

```
───────────────────────────────────────
 DietPi     | 10:46 | Mon 19/02/18
 ───────────────────────────────────────
 v6.2       | RPi Zero W (armv6l)
 ───────────────────────────────────────
 IP eth0    | 192.168.0.250
 ───────────────────────────────────────

 Created by : Daniel Knight
 Web        : http://DietPi.com
 Twitter    : http://twitter.com/dietpi_
 Donate     : http://goo.gl/pzISt9
 DietPi's web hosting is powered by: MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

root@RPi-Zero:~# 
```

Fourdee, 2018-02-19:
@k-plan 

👍 

Thanks, looks like ```/DietPi/dietpi/.update_available``` wasn't cleared after the update. Only when you ran ```dietpi-update 2```. I'll take a look.

```cat: /tmp/find_network_index: No such file or directory``` Interesting, i'll see if we can track this one down in the code.

k-plan, 2018-02-19:
@Fourdee 

> ... wasn't cleared after the update. 

This will happen to me:

- after first boot up after `dietpi-update` v6.1 ==> v6.2

- and first boot up after auto-reboot by install some software for testing (eg. ProFTP, pihole, let’s encrypt) on fresh updated v6.2 image via `dietpi-software`

k-plan, 2018-02-20:
@Fourdee 

it's happen again, after switching `de_DE.UTF-8` => `en_GB.UTF-8` => `reboot`

> Thanks, looks like /DietPi/dietpi/.update_available wasn't cleared after the update. 

 ```
───────────────────────────────────────
 DietPi     | Update available
 ───────────────────────────────────────
 v6.2       | RPi Zero W (armv6l)
 ───────────────────────────────────────
 IP eth0    | 192.168.0.250
 ───────────────────────────────────────

 Created by : Daniel Knight
 Web        : http://DietPi.com
 Twitter    : http://twitter.com/dietpi_
 Donate     : http://goo.gl/pzISt9
 DietPi's web hosting is powered by: MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 dietpi-update    = Run now to update DietPi (from v6.2 to v6.1).
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

root@RPi-Zero:~# cat /DietPi/dietpi/.
./                       .dietpi-led_control      .install_stage           .update_stage
../                      .dietpi-process_tool     .installed               .version
.dietpi-autostart_index  .dietpi-survey           .network
.dietpi-backup_settings  .hw_model                .uninstalled

root@RPi-Zero:~# cat /DietPi/dietpi/.version
6
2

root@RPi-Zero:~# cat /DietPi/dietpi/.update_stage
0

root@RPi-Zero:~# cat /DietPi/dietpi/server_version-6
6
2

root@RPi-Zero:~# cat /DietPi/dietpi/.update_available
cat: /DietPi/dietpi/.update_available: No such file or directory

root@RPi-Zero:~# cat /boot/dietpi/
.dietpi-autostart_index  .update_stage            dietpi-cpu_set           dietpi-ramlog
.dietpi-backup_settings  .version                 dietpi-cpuinfo           dietpi-services
.dietpi-led_control      boot                     dietpi-cron              dietpi-software
.dietpi-process_tool     conf/                    dietpi-drive_manager     dietpi-survey
.dietpi-survey           dietpi-autostart         dietpi-launcher          dietpi-sync
.hw_model                dietpi-backup            dietpi-letsencrypt       dietpi-update
.install_stage           dietpi-banner            dietpi-logclear          func/
.installed               dietpi-bugreport         dietpi-morsecode         login
.network                 dietpi-cleaner           dietpi-obtain_hw_model   misc/
.uninstalled             dietpi-cloudshell        dietpi-process_tool      patch_file
.update_available        dietpi-config            dietpi-ramdisk           server_version-6

root@RPi-Zero:~# cat /boot/dietpi/.update_available
6.1

```

:cry:  Why?

` root@RPi-Zero:~# cat /boot/dietpi/.update_available`

Fourdee, 2018-02-20:
@k-plan 

Thanks 👍 

Code checks out, cant see why this is reporting update available when file does not exist:
```
	UPDATE_AVAILABLE=0
	UPDATE_AVAILABLE_VERSION='' #-1 = image required, other value=latest version

	Obtain_Update_Available(){

		#Update Available
		if [ -f /DietPi/dietpi/.update_available ]; then

			UPDATE_AVAILABLE=1
			UPDATE_AVAILABLE_VERSION=$(cat /DietPi/dietpi/.update_available)

		fi

	}
```

mmm

k-plan, 2018-02-20:
Hmm ... for my the question is, why is it in  `/boot/dietpi/.update_available` and will not be deleted, if no update exist?

Fourdee, 2018-02-20:
@k-plan 

Ramdisk simply copies all back to ```/boot``` during shutdown, so we need to manually remove that file during the process.

k-plan, 2018-02-20:
> so we need to manually remove that file during the process.

:wink: 

Yes, correct. If you move it out of the way it works.
```

root@RPi-Zero:~# mv /boot/dietpi/.update_available /boot/dietpi/.update_available--
root@RPi-Zero:~# reboot

...

 ───────────────────────────────────────
 DietPi     | 20:17 | Tue 20/02/18
 ───────────────────────────────────────
 v6.2       | RPi Zero W (armv6l)
 ───────────────────────────────────────

```

Fourdee, 2018-02-20:
@k-plan 

Excellent 👍 Thank you.

I'll mark this as completed.

We are on fire 🔥 😃 (bug fixing)

closed 2018-02-20 by Fourdee