MichaIng/DietPi #1536: Where can we download older versions of DietPi?

KenthJohan, 2018-02-19:
I need DietPi v159 image for Raspberry Pi 3.

Where can we download older versions of DietPi?

k-plan, 2018-02-19:
@KenthJohan 

> DietPi v159 image for Raspberry

Why please, if you don't mind my asking?
Do you want/need/search a Debian based Jessie DietPi image for RPi?
Something like: `DietPi_v6.0_RPi-ARMv6-Jessie.img `

KenthJohan, 2018-02-19:
> Why please, if you don't mind my asking?

I want to run OpenCV 3.2 on v159 because it has no problem.
LXDE filemanager has no missing icons and more functions on v159.

> Do you want/need/search a Debian based Jessie DietPi image for RPi?

Yes. I downloded a older version of v160 at [http://dietpi.com/#download](http://dietpi.com/#download) before v160 was available.

k-plan, 2018-02-19:
Hmm, don't know, what what OpenCV is, but anyway.

Last pre-build image form DietPi before version change for RPi is` DietPi_v150_RPi-armv6-(Jessie).img`.

But please read this:  https://github.com/Fourdee/DietPi/issues/1355#issuecomment-361309473

- out of our support
- no DietPi version updates for this image and it will stay forever on v149 (mastertree is down)
- you will get some error messages during inital setup (auto-update to v150 will fail)
- software list and version will not be maintained any more
 and @Fourdee  will slay me, if I give you the download link

Can you live with this?

KenthJohan, 2018-02-19:
No need to give me the link I already have v150, it has other issues also. The sweet spot is v159.
So now I know that I have to take more backups. I took a backup of 32gb SD Card v159 but that got corrupted. I have your answer. I can live without v159.

OpenCV issue regarding v6.2 [https://github.com/Fourdee/DietPi/issues/1518](https://github.com/Fourdee/DietPi/issues/1518)

Fourdee, 2018-02-19:
@KenthJohan 

Does the issue with OpenCV occur on the v150 image aswell?

KenthJohan, 2018-02-19:
> Does the issue with OpenCV occur on the v150 image aswell?

I have not tried OpenCV on v150 the only thing I tried was OpenGL 1080p resolution which gives a blank black screen on boot up on RPi3.

Fourdee, 2018-02-19:
@KenthJohan 

> OpenGL 1080p resolution which gives a blank black screen on boot up on RPi3.

Yep, there was a kernel specific issue with the OpenGL in the past. This is now resolved by using the latest RPi APT kernel.

Another option is ```rpi-update```, to try bleeding edge.

k-plan, 2018-02-20:
@KenthJohan 

> No need to give me the link I already have v150, it has other issues also. The sweet spot is v159.
> So now I know that I have to take more backups. I took a backup of 32gb SD Card v159 but that got corrupted. I have your answer.

Hmm ... I can feel with you. This is annoying.
Sorry, `DietPi_v150_RPi-armv6-(Jessie).img` is the last build from Fourdee.

My ".img shrink system" is running on a RPi 3. Can't remember exactly, but I think it's running DietPi v158 with lastest by Fourdee patched RPi Kernel 4.9. If you have some time, I can make a backup,  tear all stuff down and build a unsupported `DietPi_v158_RPi-armv6-(Jessie).img`. But be warned, I'm not the great "image builder" like Fourdee. :smiley: 

Can this be a temporary solution  for you, until @Fourdee  can look or work at your v6.x issue ticket?
Will it be okay @Fourdee ?

Fourdee, 2018-02-20:
@k-plan Your call. Solution may be as per: https://github.com/Fourdee/DietPi/issues/1518#issuecomment-367065474

k-plan, 2018-02-20:
@Fourdee 

> Solution may be as per: # 1518 (comment)

Wow, so quick!  Very nice.  :+1:   :smiley: 
And for "unsupported" software.
Had the feeling, you have played around with this stuff before.

@KenthJohan can give it a try.

Let's hear, what he say. 
`DietPi_v158_RPi-armv6-(Jessie).img` will not run away any more.  :smile:

Fourdee, 2018-02-21:
OpenCV issue resolved: https://github.com/Fourdee/DietPi/issues/1518#issuecomment-367293397

I'll mark this ticket as closed due to the following:
- We only officially offer/release/support current DietPi images in the v6.x revision.

closed 2018-02-21 by Fourdee