MichaIng/DietPi #1537: DietPi Images | Restructure of images and supported devices

Fourdee, 2018-02-19:
In the interest of both parties involved:
- I have taken the decision to remove all ARMbian based images from our lineup.

We will asses which are the most popular devices, and attempt to offer a non-ARMbian based image for those device.

Other devices which fail to reach an acceptable level of end user installations (survey), will be removed. In favor of allowing us to focus on the popular devices, reducing the stretch and strain on our support infrastructure.

---

@k-plan @MichaIng 

Once i've calculated the installation counts via survey, i'll post here and we can further discuss.

Fourdee, 2018-02-19:
#### Current ARMbian based images:
- ```$G_HW_MODEL``` | Image name
- 50 DietPi_v6.0_BPiM2Plus-ARMv7-Stretch.img
- 51 DietPi_v6.0_BPiPro-ARMv7-Stretch.img
- 64 DietPi_v6.0_NanoPiNEOAir-ARMv7-Stretch.img
- 60 DietPi_v6.0_NanoPiNEO-ARMv7-Stretch.img
- 33 DietPi_v6.0_OPiLite-ARMv7-Stretch.img
- 31 DietPi_v6.0_OPiOne-ARMv7-Stretch.img
- 38 DietPi_v6.0_OPiPC2-ARMv8-Stretch.img
- 30 DietPi_v6.0_OPiPC-ARMv7-Stretch.img
- 34 DietPi_v6.0_OPiPlus-ARMv7-Stretch.img
- 37 DietPi_v6.0_OPiPrime-ARMv8-Stretch.img
- 35 DietPi_v6.0_OPiZero2H3-ARMv7-Stretch.img
- 35 DietPi_v6.0_OPiZero2H5-ARMv8-Stretch.img
- 32 DietPi_v6.0_OPiZeroH2-ARMv7-Stretch.img

---

### Total unique installs, all devices (as of v6.0 release)
- 10734

Fourdee, 2018-02-19:
@k-plan 

> remove all from lineup (incl. all remaining NanoPi's, because I'm unsure on which base FrindlyArm has build their images)

Would this alot easier 👍 

I actually like the NanoPi boards, great build quality and I would prefer to keep them at least. As far as i'am aware, their kernel is maintained by them for some of the boards: https://github.com/friendlyarm/linux-3.4.y, but i'll check them individually.

So yes, lets drop all OrangePi and BananaPi devices? Lets try to keep the NanoPi images (which are either not ARMbian based, and/or, have a non-ARMbian based image available)?

Fourdee, 2018-02-19:
@k-plan 
Ok, I can find no evidence in the image, or friendly ARM kernel, that ARMbian was used in any form, for the following images:
- DietPi_v6.0_NanoPiM1-ARMv7-Stretch.img
- DietPi_v6.0_NanoPiM1Plus-ARMv7-Stretch.img
- DietPi_v6.0_NanoPiM2-ARMv7-Stretch.img
- DietPi_v6.0_NanoPiM3-ARMv7-Stretch.img
- DietPi_v6.0_NanoPiNEO2-ARMv8-Stretch.img

I honestly believe the above images are officially made and maintained solely by Friendly ARM. 

Therefore @k-plan @MichaIng I believe the solution here is:
- Drop all OrangePi images
- Drop all BananaPi images
- Drop DietPi_v6.0_NanoPiNEO-ARMv7-Stretch.img
- DietPi_v6.0_NanoPiNEOAir-ARMv7-Stretch.img

We can still allow user to run PREP on those devices, but we will not recommend or endorse any specific pre-image, that is down to the user and out of our control.
Whats your thoughts?

k-plan, 2018-02-19:
@Fourdee 

> Would make this alot easier +1

Yes, will be the most simple and clean solution.

All RPi's , Odroid's and some VM's.  Finish.

> So yes, lets drop all OrangePi and BananaPi devices?

 I'm afraid, yes.

> Lets try to keep the NanoPi images (which are either not ARMbian based, and/or, have a non-ARMbian based image available)?

Hmm ... to get it clear and structured - NanoPi - all or nothing.
If all, we have please to freeze the list of supported FriendlyARM devices now and forever.
Cancel the hole NanoPi stuff, if we get hassle to find base images, can sill be done  later.

But @MichaIng  will not like this at all.

Fourdee, 2018-02-19:
@MichaIng Would love to get your views on this aswell if possible 👍 

Heres my original idea, to keep NanoPi devices that are not ARMbian:
http://dietpi.com/index_new.html

k-plan, 2018-02-19:
@MichaIng 
> Would love to get your views on this aswell if possible +1

Me too! :+1: 

@Fourdee 

> Heres my original idea, to keep NanoPi devices that are not ARMbian:

Nice one! :+1: 
More then enough supported devices. :+1: 

May it's allowed to tell ...

> _No worries, DietPi can be easily installed on any device!_
> _Simply run the DietPi PREP script in the above link and ..._

...   I can hear or feel trolls in a distance with  a smell of red herring ... :wink:

Fourdee, 2018-02-19:
@k-plan 

👍 

Ok, i've updated the initial ticket (we'll link this to explain the changes to our users):
https://github.com/Fourdee/DietPi/issues/1537#issue-298337026

@MichaIng Apologies you may have been unable to reply (which I respect and would love to know your views on these changes?). But, I need to move this forward. I'll set the new download page live and remove the OPi/BPi images from the server.

@MichaIng

Fourdee, 2018-02-19:
@k-plan @MichaIng 

> I need to move this forward. I'll set the new download page live and remove the OPi/BPi images from the server.

Done

http://dietpi.com/download

Fourdee, 2018-02-19:
@k-plan @MichaIng 

I'll mark this as closed for now. But continued discussions are welcomed 👍

Fourdee, 2018-02-19:
Actually, we still need to do the following:
- Remove 'ARMbian' image credits from ```obtain_hw_model```

MichaIng, 2018-02-20:
@Fourdee @k-plan 
You know my opinion about this generally 😉. I am unhappy with the fact, that the affected users are the ones, that finally pay, just because of this unnecessary dev conflict. Also I love the fact, that we were able to support that much different devices, even that some of them are not used much and others are bad firmware/software wise maintained by producers. Some people do just not know when buying BananaPi or OrangePi, others want to experiment, and the fact that DietPi runs well on all of them, makes surely some peoples lives easier 😄. Also from what I could read here on github and forum, OrangePis and BananaPis didn't seem to produce much issues or work for us, I mostly see Odroid issues actually. I guess this is due to the special/additional APT sources and yeah surely for the fact, that Ordoids are the only boards, still on Jessie (which gets support drop for more and more software). This on the other hand might also show, that ARMbian made actually a good work to provide stable working images, behaving for our needs close enough to raw Debian.
But it could be also just related to the lower usage of BananaPi/OrangePi => less errors/issues reported 😉.

But following the discussion I can totally understand the step. This ongoing conflict must be stressful and frustrating and for sure what can be read on twitter & co. is damaging DietPi and ARMbian as well. At least I couldn't read any comment, supporting the negative words about DietPi, but defending and respecting us/both projects. I think most people see how childish and nonsense these attacks are.
It is hard enough to keep such a project alive and successful and motivation killers like this should be avoided where possible!

@Fourdee 
Maybe you can add some explanation sentence onto device download button, to people are not just faced by 404?
_____________________________________
Cut... restart...:
Might be we were wrong about OPi/BPi image offers, or things have changed, or I missunderstood?
http://www.orangepi.org/downloadresources/
http://www.banana-pi.org/downloadall.html

They have a quite rich offer, mostly even Raspbian besides Debian 😃. I mean yeah @k-plan you vote for dropping this longer, but as long as it's about ARMbian, it looks like we have alternatives. Of course would need test, if DietPi-Prep works as well on these.

Fourdee, 2018-02-20:
@MichaIng 

💘 👍 

Yep, your spot on, the user is the one that suffers. I wish there was another way, but, even with us trying to resolve, there simply isn't a solution.

> Sorry for late reply, I was really busy today with other stuff...

No need to apologise. I believe your opinion matters greatly to our project and I wanted it quicker than was possible 😃 

> This ongoing conflict must be stressful and frustrating

Nailed it in one. I cannot allow the situation to continue, its beyond silly and continued for far too long. Its affecting me and my passion (and clearly Igor), I felt this complete separation had to be done, in the interest of everyone affected.

> Maybe you can add some explanation sentence onto device download button, to people are not just faced by 404?

The updated download page should be live on http://dietpi.com/download, might need a CTRL+F5 to force a refresh.
Or, in reference to direct image download links in the wild across forums etc?

> Might be we were wrong about OPi/BPi image offers, or things have changed, or I missunderstood?

There are very messy images (OPi), last time I checked. Quickly hashed together using multiple different sources.
We'd need to verify everything in the image (kernel etc) is non-ARMbian. I'll keep it in mind, but lets see how our users feel about the image changes. 

Either way, I personally dislike the build quality of OPi, they really are the terrible. Only their boards have failed on me that I can remember (OPi PC HDMI dead, OPi Zero dead). And in our efforts of removing them, hopefully we can prevent our users from having to experience the same thing?
I dunno, just thinking aloud from past experience :)

k-plan, 2018-02-21:
> Actually, we still need to do the following:
> - Remove `Orange Pi` / 'ARMbian'  reference, credits and links from

- [ ]  README.md

- [ ] v6.0 | Recreate all images

- [ ] ... _work in process_

k-plan, 2018-02-21:
> Actually, we still need to do the following:
> - Remove `Orange Pi` / `ARMbian`  reference, credits and links from

- [x]  :yellow_heart: :  README.md   (`master` only -` testing `can't be done by me @Fourdee )

- [x] :green_heart: :   v6.0 | Recreate all images

- [x] :green_heart: :  checked DietPi v6.0 | FAQ (i.O.)

Fourdee, 2018-02-21:
@k-plan 

Thanking you, legend 👍 

Image credits and README.md (testing branch) all updated.

I'll mark this as closed. Will try and do a mini release for v6.3 to roll this out for users.

closed 2018-02-21 by Fourdee