MichaIng/DietPi #1538: dietpi-backup wont start up

sintacs, 2018-02-19:
## Creating a bug report/issue:

running dietpi-backup, it stops after:
[Info] Checking RootFS R/W access. Please wait...
[Ok] RootFS R/W access
[3;J

#### Required Information:
- DietPi Version
159

- SBC Device (EG: RPi 3)?
RPi 3

- Power supply used?
5v 1A RAVpower

- SD card used?
Sandisk ultra

- Distro (EG: Jessie) | ```uname -a```?
Linux DietPi 4.9.62-v7+ #2 SMP Fri Nov 17 23:52:26 GMT 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Software title?
dietpi-backup
- Can this issue be replicated on a fresh installation of DietPi?
not tried yet

### Did you submit a ```dietpi-bugreport```?
I tried, but it stops in the middle and does not process further

Fourdee, 2018-02-19:
@sintacs 

Hi,

Please note that v159 or lower, is no longer supported. v6.x is currently required: 
https://github.com/Fourdee/DietPi/issues/1355

> �[3;J

However, I believe this issue is due to broken locales. Please use ```dietpi-config``` > Regional, and ensure that ```en_GB.UTF-8``` is enabled on the 1st screen.

If problems persist, I would recommend updating to our current v6.x images.

NB: Unsure if you are doing this, however, restoring a backup on a v6.x image, from v159 or lower, is not compatible. It will essentially return the system to v159 and most likely in a dysfunctional state.

I'll mark this as closed. Please reopen if required.

closed 2018-02-19 by Fourdee

sintacs, 2018-02-21:
Hi,

thanks for your reply.

The locale is and was enabled all the time:

[image]

en_US.UTF-8 is enabled also.