MichaIng/DietPi #1539: [ODroid XU4] Your version of DietPi is now obsolete and cannot be updated.

bhaveshgohel, 2018-02-20:
Hello @Fourdee (Dan), 
The SBC is running DietPi v156. But when i try to update to latest version it is failing with error "Your version of DietPi is now obsolete and cannot be updated."

#### Required Information:
- DietPi Version - 156
- SBC Device -   oDroid XU3/4 (armv7l)
- Power supply used - 5V 3A
- SD card used - 32GB EMMC
- Distro - Linux brnsOdroid 3.10.106+ # 1 SMP PREEMPT Fri Nov 24 16:26:38 CET 2017 armv7l GNU/Linux

#### Did you submit a ```dietpi-bugreport```?
Yes! - Reference CODE: c67ca81e-3042-456c-af0a-127222a9df27-0 

#### Extra details:
[image]

Do let me know if you require more information.

MichaIng, 2018-02-20:
@bhaveshgohel 
Please read: https://github.com/Fourdee/DietPi/issues/1355#issue-285207849

And in case: http://dietpi.com/phpbb/viewtopic.php?f=9&t=2799

Fourdee, 2018-02-20:
I'll mark as closed. FAQ covers this.

closed 2018-02-20 by Fourdee

bhaveshgohel, 2018-02-21:
Ohh.. Thanks for Clarifying. i'll re-install DietPi. :smiley: 
Can we add FAQ link to README.md ? it will be helpful.