MichaIng/DietPi #1540: DietPi Software | Midnight Commander line characters

k-plan, 2018-02-20:
## Creating a bug report/issue:

http://dietpi.com/phpbb/viewtopic.php?f=11&t=2877

[image]

#### Required Information:
- DietPi Version |  v6.2
- SBC Device | all my devices
- Distro  | Stretch (Jessie? - untested)

#### Additional Information (if applicable):
- Software title? 

Midnight Commander

- Can this issue be replicated on a fresh installation of DietPi?

Yes

#### Steps to reproduce:

` dietpi-config` =>  Language and Regional Options  =>  Locale   Change Language and Regional Settings => `de_DE.UTF-8 `

[image]

`dietpi-software install id 3`

Fourdee, 2018-02-20:
@k-plan 

Thanks 👍 

Yep, we removed non UTF-8 options from locale generation. ``` ISO-8859-1``` required for MC.

k-plan, 2018-02-20:
@Fourdee 

> mm, fine under en_GB with kitty and OpenSSH. i'll test.

- `dietpi-config` => Language and Regional Options => Locale Change Language and Regional Settings => `en_GB.UTF-8
`

- `reboot`

- change` kitty ` to  `ISO-8859-1:1998 (Latin-1, West Europe)`

- SSH Server: Dropbear

- `mc` :

[image]

:+1: 

- `dietpi-software` : 

[image]

:-1: 

- login screen:

[image]

:-1:

Fourdee, 2018-02-20:
@k-plan 

👍 

🈯️ 
```
LANG='en_GB.UTF-8' mc
```

Options:
- Always export LC_ALL/LANG en_GB.UTF-8 when dropbear is installed. Our scripts are fine as we define it already in globals.
-- and/or export current locale as generated by locale
- Be aware this is a bug in dropbear, leave it as is (dont like)

Fourdee, 2018-02-20:
🈯️ Yep, fixes dropbear:
```
cat << _EOF_ > /etc/profile.d/99-dietpi-force-locale.sh
# To force server locales on SSH access, as dropbear does automatically overwrite them by client values:
export LANG="en_GB.UTF-8"
export LC_ALL="en_GB.UTF-8"
export LANGUAGE="en_GB:en"
_EOF_
```

However, it will always force en_GB.UTF-8, regardless of locale.

k-plan, 2018-02-20:
Hmm ... puh .... really difficult.

Can't say, what to do now.

Fourdee, 2018-02-20:
@k-plan 
Resolved for v6.3 👍 

Workaround for v6.2:
```
cat << _EOF_ > /etc/profile.d/99-dietpi-force-locale.sh
#!/bin/bash

#Workaround for setting locale to session when POSIX is detected (eg: dropbear): https://github.com/Fourdee/DietPi/issues/1540#issuecomment-367066178
if (( \$(locale | grep -ci -m1 'POSIX') )); then
CURRENT_LOCALE=\$(grep -m1 'AUTO_SETUP_LOCALE=' /DietPi/dietpi.txt | sed 's/.*=//')
export LANG="\$CURRENT_LOCALE"
export LC_ALL="\$CURRENT_LOCALE"
#export LANGUAGE="\$CURRENT_LOCALE"
fi
_EOF_
chmod +x  /etc/profile.d/99-dietpi-force-locale.sh
```

Fourdee, 2018-02-20:
NB: Left ```LANGUAGE``` out, requires different syntax and I believe its no longer used in Linux, LANG/LC_ALL should be sufficient.

closed 2018-02-20 by Fourdee

slim0287, 2018-05-05:
First, please pardon me if I am posting this in the wrong place.  When I ssh into my dietpi, I get the following message:
-sh: 4: /etc/profile.d/99-dietpi-posix-set-locale.sh: 0: not found

Comments in the 99-dietpi-posix-set-locale.sh script referred to issue 1540 which is why I am posting here.  I'm not running dropbear nor POSIX.  The error message appears to be with how the 'if' statement is written within the script.  I haven't changed it because it's a minor annoyance for a Pi that I log into infrequently, but I thought I would report it here.  And it it helps, this dietpi is running on a Raspberry Pi 2 that I use as a DNS server with Pi-hole and I log in with ssh keys, not a password.

Thanks.

MichaIng, 2018-05-05:
@slim0287 
Thanks for the report. No worries, first perfect here.

Hmm, the syntax should be fine, at least works well for me and is widely used within our scripts.
Please verify that `locale` and `grep` packages are installed, but should always be: `apt install locale grep`

Also try to run the command manually: `locale | grep -qi 'POSIX' || echo 'no POSIX locale'`

slim0287, 2018-05-05:
Both `locale` and `grep` are installed on my system.  When I run your suggested command manually, I get the following: `no POSIX locale`

When I run the `if` statement from the script on the command line, I get: `> > > > > -sh: 17: 0: not found`
Which is consistent with what I see on login.

Does that help the diagnosis?

Thanks

slim0287, 2018-05-05:
That solved it.  My script has a different `if` statement as well as other differences, as follows:
```
if (( $(locale | grep -ci -m1 'POSIX') )); then
	CURRENT_LOCALE=$(grep -m1 'AUTO_SETUP_LOCALE=' /DietPi/dietpi.txt | sed 's/.*=//')
	export LANG="$CURRENT_LOCALE"
	export LC_ALL="$CURRENT_LOCALE"
	#export LANGUAGE="$CURRENT_LOCALE"
fi
```
I never modified this script, so I have no idea how it was set this way.  At any rate, thanks for the help.  And I should add, I very much like DietPi!

Thanks again.