MichaIng/DietPi #1542: Wifi HotSpot Enchantment

symbios24, 2018-02-21:
Hello, is it possible to add this enchantment for the wifi/tor Hospot

If there is not an internal wifi eg: rpi2/rpi1 to use automatically the attached External Usb Wifi Adapter
If there is an Internal Wifi available  eg: rpiZeroW-Rpi3  but also an external Usb wifi adapter Attached 
to ask the user which wifi to use for the HotSpot
or if there is an Internal Wifi available  eg: rpiZeroW-Rpi3  but also an external Usb wifi adapter Attached to use by default the external USb Wifi

Thanks

Fourdee, 2018-02-21:
@symbios24 

Thanks for the request 👍 

Currently:
- If the device has onboard WiFi, that will be used by default
- If not, external USB adapter will be used (if detected)

This applies to hotspot and standard WiFi client connection.

Adding support for onboard/USB selection, will take some time to implement. Mainly due to the way ```/DietPi/dietpi/func/dietpi-set_hardware wifimodules enable``` is handled and hard coded for onboard WiFi adapters per device.

Leave it with us, we'll try to pick this up when possible.

Fourdee, 2018-02-21:
Flagged for low priority. If this ticket obtains more interest from our users, we'll change it accordingly.

SuBLiNeR, 2018-02-25:
@Fourdee @symbios24 there are more easier solutions for this to be handeld. Nothing to change on DietPi. It works oob if you're not afraid using the console and copy&paste some commands.

Main part is to have a compatible USB-WiFi-Dongle (Google is your friend!); easiest way to get stuff done is editing `/etc/interfaces/network` and be sure the wlan1 is found at start. 

btw. @symbios24 i just answered this in you last question upon handling two WiFis on the RPi.

reigelgallarde, 2018-03-07:
my way of doing this was with udev... haha I've swap wlan1 and wlan0... might not be a great solution but works for me.

thaihugo, 2018-09-04:
Sorry to add a comment on a closed item, but I’m using an USB adapter on my raspberry pi with DietPi. The setup is working but if by mistake I go to the WiFi parameter in DietPi-config, it will replace my wlan1 by wlan0 in /etc/interface/network. On my setup, it will exclude my access to the device (create_ap for an access point). It there a parameter or  a way to ask DietPi not to touch to my network file anymore and to say I’ll manage it myself instead of switching wlan0 and wlan1 in udev?

krishnasen2401, 2019-01-02:
it would be good as it will be easy to create wifi hotspot by bridging

FredericGuilbault, 2019-06-30:
On lysmarine I thied different ways to hotspot, looked at what RPI communuty do, and how openplotter scripted it. ButI endup selecting https://github.com/oblique/create_ap 
... If it can help

MichaIng, 2019-06-30:
@FredericGuilbault 
Thanks for sharing.

There is indeed no issue with the method to create the hotspot, that is quite clear. Currently there is simply no way within `dietpi-config` to choose which WiFi adapter to use. The first match will by used.

A topic on the list for v6.26 to make dietpi-config network adapter menu much more flexible by scanning for all available adapters and allowing to configure each of them individually, either as main/internet default route, for local network access only or as hotspot/NAS which would generally work for Ethernet adapters as well (basically internet sharing).

lwqcz, 2020-12-15:
> There is indeed no issue with the method to create the hotspot, that is quite clear. Currently there is simply no way within `dietpi-config` to choose which WiFi adapter to use. The first match will by used, which is the onboard adapter, if available and kernel module not blacklisted.

@MichaIng what about `udev` rules? I mean use them to rename whatever wifi device to `wlan0` which will be used as the primary device for everything specified in `dietpi-config` ?

MichaIng, 2020-12-15:
My plan is to allow setting up any network interface found, regardless of the name it has, for any role, via `/etc/network/interfaces.d/dietpi-<name>.conf`, for max transparency and making it easy to configure or remove or override each interface. No need then to change the names via udev rules, and it would even allow to keep the "new" "predictable" interface names like `wlp1s3`. I'd still keep the old names, since I think most users prefer, but I'll add a toggle to enable predictable names, in case users are used to them already, as those are default on Debian and many other distros.