MichaIng/DietPi #1546: G_WHIP improvements/additions

Fourdee, 2018-02-21:
> http://dietpi.com/phpbb/viewtopic.php?f=9&t=2882&p=11077#p11077

Fourdee, 2018-02-21:
- ```G_WHIP_MSG "info"``` | removed additional var
- ```G_WHIP_YESNO``` | added
- ```G_WHIP_VIEWLOG``` | added

MichaIng, 2018-02-22:
@Fourdee 
Great one, I already have so much ideas in how to use them 😄.

About this screen size recognition, I already thought that would make sense maybe for all our whiptail uses? I mean yeah, it is big enough, but on VirtualBox default 640x480 console they are too big, left and right are cut.
But I guess this would need vast rework of content to fit different screen sizes? I didn't look much into whiptail so no idea how much work it would be, just saw that some text is aligned manually.

Fourdee, 2018-02-22:
@MichaIng 

> I mean yeah, it is big enough, but on VirtualBox default 640x480 console they are too big, left and right are cut.

Yep, plan is to roll out ```G_WHIP``` to all our scripts, replacing the manual hard coded entries. I just need to add another 2 for array type selection (eg: dietpi-software and dietpi-config)

I'll check the screen size, we can always tweak it if needed based on ```tput``` values (eg: if > 100, then cap to 100 etc).

I don't have a VM setup, if you get a chance whats the values of:
```
tput lines
tput cols
```

MichaIng, 2018-02-23:
@Fourdee 
```
tput lines
25
tput cols
80
```
Default on VirtualBox images, but adjustable via grub defaults: https://github.com/Fourdee/DietPi/issues/1227

Fourdee, 2018-03-02:
To be added:
- ```G_WHIP_INPUTBOX```
- ```G_WHIP_MENU```
- ```G_WHIP_CHECKLIST```

Fourdee, 2018-03-02:
mmm, cant set result var, but echo $? works...
```
			export G_WHIP_RETURNED_VALUE=$(whiptail --title "$G_PROGRAM_NAME" --menu "$message" --default-item "$G_WHIP_DEFAULT_ITEM" --cancel-button "Cancel" --backtitle "$G_PROGRAM_NAME" $G_WHIP_SIZE_Y $G_WHIP_SIZE_X $G_WHIP_SIZE_Z "${G_WHIP_MENU_ARRAY[@]}" 3>&1 1>&2 2>&3; result=$?)
```

Fourdee, 2018-03-03:
Auto whiptail scaling based on screen dimensions and internal content:
test/notes:
```
. /DietPi/dietpi/func/dietpi-globals
G_WHIP_CHECKLIST_ARRAY=(

'option1' 'test' 'off'
'option2' 'test' 'on'
'option3' 'test' 'off'
'option4' 'test' 'off'
'option5' 'test' 'off'
'option6' 'test' 'off'
'option7' 'test' 'off'
'option8' 'test' 'off'
'option9' 'test' 'off'
'option10' 'test' 'off'
'option11' 'test' 'off'

)
G_WHIP_CHECKLIST 'Test selection\ntest\ntest\ntest'
```

Fourdee, 2018-03-03:
Automatic scaling of G_WHIP based on:
- Screen size
- Text message size (total lines and character count against screen X)
- Prints maximum Z options (eg: menu) available in the screen size total.

[image]

[image]

[image]

[image]

Fourdee, 2018-03-03:
Pretty much done, just needs rolling out across all DietPi scripts.

Fourdee, 2018-03-04:
Marking as completed, i'll roll ```G_WHIP``` out throughout all scripts gradually (will take some time).

closed 2018-03-04 by Fourdee