MichaIng/DietPi #1549: DietPi-Config | WiFi: add options for selection of WiFi proto

Fourdee, 2018-02-22:
WPA-PSK / WPA-EAP / NONE

> https://github.com/Fourdee/DietPi/pull/1547

And all additional vars linked to those options.

MichaIng, 2018-10-29:
Settings possible via `dietpi-wifi.db/txt` now, handled within new DietPi-WiFiDB script: https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-wifidb

But great would be the possibility to adjust EAP settings as well via GUI. Currently only WEP/PSK SSID+key possible.

MichaIng, 2019-03-26:
Could be added: https://github.com/MichaIng/DietPi/issues/2650
- Allow enabling/disabling hidden SSID support by adding/skipping these two settings. The extra options seem to increase connection/scan time and the vast majority of users will never need it.

MichaIng, 2019-04-23:
Related request: https://github.com/MichaIng/DietPi/issues/2732

MichaIng, 2019-05-11:
Marking this as closed in favour of the new request.

closed 2019-05-11 by MichaIng