MichaIng/DietPi #1550: Image | OPi PC

Fourdee, 2018-02-22:
Offer a trial/test image based on official OPi PC image.

> http://dietpi.com/phpbb/viewtopic.php?f=9&t=2883#p11055

Fourdee, 2018-02-23:
Odroid N1 to read SSD:
[image]

C2 Stretch fine, interesting.

k-plan, 2018-02-23:
@Fourdee 
> Image | OPi PC # 1550

Hmm ... what is it now? Odroid N1? C2 Stretch? Or OPi PC? 
Picture into the wrong issue? :smiley:

Fourdee, 2018-02-23:
@k-plan 

> wrong issue

Yep

k-plan, 2018-02-24:
_My little personal notes for Orange Pi (One):_

- Try out to find a image which is build by manufacturer ["OrangePi"](http://www.orangepi.org/downloadresources/) based on Debian

- Found [Raspbian_For_OrangePi_PC_v0_8_0.img](http://www.orangepi.org/downloadresources/orangepipc/oragepipc_e930546e866b23585721e5d2a6.html) and [debian_xfce_desktop_Orangepione_v0_9_1.img](debian_xfce_desktop_Orangepione_v0_9_1.img)

- Download `Raspbian_For_OrangePi_PC_v0_8_0.img` (google drive) work. 
Quick look inside, it's a `Wheezy` based image. without a FAT partition and many reference to "`BananaPi`" !? (what the heck...)

> ```
> ls -lah /Volumes/Untitled/home/
> total 32
> drwxr-xr-x  1 user  staff   252B  2 Mai  2015 .
> drwxrwxrwx  1 user  staff   1,5K 24 Feb 18:34 ..
> drwxr-xr-x  1 user  staff   567B  2 Mai  2015 bananapi
> drwxr-xr-x  1 user  staff   2,3K  3 Mai  2015 orangepi
> ```
Droped it into the bin ....

- Download `debian_xfce_desktop_Orangepione_v0_9_1.img` (Mega) didn't work for me ... (what the heck...)
Found a russian site, which have a working download link on Mail.ru ...
Quick look inside, it's a `Jessie` based image. with a FAT partition and  reference to "`BananaPi`" !?

> ```
> ls -lh /Volumes/linux/home/orangepi/
> total 7216
> drwxr-xr-x  1 user  staff   126B  1 Jan  1970 Desktop
> -rw-r--r--  1 user  staff   447K  5 Dez  2016 OrangePiBGP.jpg
> drwxr-xr-x  1 user  staff   126B  8 Dez  2016 Scratch
> drwxr-xr-x  1 user  staff   1,0K  8 Dez  2016 glmark2-2011.09
> drwxr-xr-x  1 user  staff   315B  8 Dez  2016 hw_accleration
> drwxr-xr-x  1 user  staff   882B  8 Okt  2015 orange_pi_WiringBP_guenter_geaendert_bananapro
> drwxr-xr-x  1 user  staff   693B 27 Okt  2015 orangepi_gpio_pyH3
> ```

It's a huge image (3.2 GB) without no auto resize file system, non working hdmi output (for me), but I decided, to give it a spin.

- After fs resize, you have to repair apt and `source.list` to have any chance, to get something up and running.
- After this and some fiddleing around, get `PREP_SYSTEM_FOR_DIETPI.sh` up and running.
Work very impressive, but it take some hours to run through. 👍 
- `reboot` but now nothing ... only red led lights up.
- With the help of a UART adapter, some bad messages, extracting some files for base image, I can get it up and running. 😄 
- editing some files and copy same files and dirs, delete and cleanup, I have get a working DietPi based on Debian Stretch.

**Conclusion:**

> Simply run the DietPi PREP script in the above link and follow the onscreen instructions.

As I have mention now several times, I am not willing to support this procedure for "normal end-user".
Sorry, simple no. 👎 

To get it clear for now, `PREP_SYSTEM_FOR_DIETPI.sh` is not the problem. 
@Fourdee  and @MichaIng  have done a great and really impressive work.
It's running like a charme. 💯  👍 

The problem is the base image, which is out of our control. To many thing are and can go wrong.
Nothing I can get supported, without seeing or testing by myself.
In this way, we only will get bad end-user experience any high support appearance.

k-plan, 2018-02-24:
Impessions:

```
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
[  OK  ] Root access verified.
 ───────────────────────────────────────
 DietPi     | 17:10 | Sat 24/02/18 
 ───────────────────────────────────────
 v6.2       | OrangePi One (armv7l)
 ───────────────────────────────────────
 IP eth0    | 192.168.0.100
 ───────────────────────────────────────
 Created by : Daniel Knight
 Web        : http://DietPi.com
 Twitter    : http://twitter.com/dietpi_
 Donate     : http://goo.gl/pzISt9

 Device image possible thanks to: ARMbian
 DietPi's web hosting is powered by: MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.
```

```
root@DietPi-OrangePi-One:~# cpu
[  OK  ] Root access verified.

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     armv7l
 Temp         |     44'c : 111'f | Optimal temperature.
 Governor     |     ondemand
 Throttle up  |     50% CPU usage

                 Current Freq    Min Freq   Max Freq
 CPU0         |      1200 MHz      480 MHz    1200 MHz
 CPU1         |      1200 MHz      480 MHz    1200 MHz
 CPU2         |      1200 MHz      480 MHz    1200 MHz
 CPU3         |      1200 MHz      480 MHz    1200 MHz

[ INFO ] DietPi-CPU_info | CPU current frequency, may be affected by this script, due to the processing required to run it.
```

```
root@DietPi-Orange-Pi-One:~# cat /etc/debian_version
9.3
```

```
root@DietPi-Orange-Pi-One:~# cat /proc/version
Linux version 3.4.112 (root@curry) (gcc version 4.6.3 20120201 (prerelease) (crosstool-NG linaro-1.13.1-2012.02-20120222 - Linaro GCC 2012.02) ) #1 SMP PREEMPT Thu Sep 22 15:08:09 CST 2016
```

```
root@DietPi-OrangePi-One:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
udev             10M     0   10M   0% /dev
tmpfs            99M  1.6M   98M   2% /run
/dev/mmcblk0p2  3.6G  617M  2.9G  18% /
tmpfs           248M     0  248M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           248M     0  248M   0% /sys/fs/cgroup
tmpfs            20M   44K   20M   1% /var/log
tmpfs            10M  1.4M  8.7M  14% /DietPi
/dev/mmcblk0p1   63M   35M   29M  56% /boot

```

I will do some sketchy testing, playing around, housekeeping and will upload it for dev testing and run-through.

As far as I can see, but NOT able to test, it's possible, it will run of several OrangePi devices.

```
root@DietPi-OrangePi-One:~# ls -lah /boot/
total 34M
drwxr-xr-x  3 root root 2.0K Jan  1  1970 .
drwxr-xr-x 21 root root 4.0K Feb 24 17:00 ..
-rwxr-xr-x  1 root root 6.0K Feb 24 16:56 README.md
-rwxr-xr-x  1 root root  32K Feb 24 15:56 boot0_OPI.fex
-rwxr-xr-x  1 root root 2.1K Feb 24 17:14 config.txt
drwxr-xr-x  5 root root 3.5K Feb 24 17:14 dietpi
-rwxr-xr-x  1 root root  11K Feb 24 17:14 dietpi.txt
-rwxr-xr-x  1 root root  36K Dec  9  2016 script.bin
-rwxr-xr-x  1 root root  35K Feb 24 15:56 script.bin.OPI-2_480p_hdmi
-rwxr-xr-x  1 root root  35K Feb 24 15:56 script.bin.OPI-2_720p60_hdmi
-rwxr-xr-x  1 root root  35K Feb 24 15:56 script.bin.OPI-LITE_720p_hdmi
-rwxr-xr-x  1 root root  36K Feb 24 15:56 script.bin.OPI-ONE_720p_hdmi
-rwxr-xr-x  1 root root  37K Feb 24 15:56 script.bin.OPI-PCPLUS_720p_hdmi
-rwxr-xr-x  1 root root  35K Feb 24 15:56 script.bin.OPI-PC_480p_hdmi
-rwxr-xr-x  1 root root  35K Feb 24 15:56 script.bin.OPI-PC_720p60_hdmi
-rwxr-xr-x  1 root root  38K Feb 24 15:56 script.bin.OPI-PLUS2E_720p_hdmi
-rwxr-xr-x  1 root root  36K Feb 24 15:56 script.bin.OPI-PLUS_480p_hdmi
-rwxr-xr-x  1 root root  36K Feb 24 15:56 script.bin.OPI-PLUS_720p60_hdmi
-rwxr-xr-x  1 root root 880K Feb 24 15:56 u-boot_OPI-emmc.fex
-rwxr-xr-x  1 root root 880K Feb 24 15:56 u-boot_OPI.fex
-rwxr-xr-x  1 root root  11M Feb 24 15:56 uImage
-rwxr-xr-x  1 root root  11M Feb 24 15:56 uImage_OPI-2
-rwxr-xr-x  1 root root  11M Feb 24 15:56 uImage_OPI-PLUS
```

MichaIng, 2018-02-24:
@k-plan 
Thanks for this testing, great work a bid hope for our OPi users.

But indeed you were right, this images from OPi are a mess. Messy confusing description, wrong and mixed icons, various different 3rd party file hosts, OS versions that do not fit to the image creation/upload date, missing details... but most importantly of course the content.

Seems they more less dropped support for the device, this image applies to. Found the github link from OPi One dead, instead repos for H2, H3, H5, A64, RDA are up with u-boot/firmware. Could be possible to use those to create an image one time from scratch and maintain/update it afterwards: https://github.com/orangepi-xunlong?tab=repositories
There seem to be anyway nothing done about it on github, looks like uploaded and left, so kernel version would be frozen.
But I never build an image from scratch, no idea how to start exactly and about the effort 😉.

k-plan, 2018-02-24:
@MichaIng 

H3 stuff will be the correct one for OPi - One, - PC, - Plus,  - 2, ...

http://linux-sunxi.org/H3

> But I never build an image from scratch, no idea how to start exactly and about the effort 😉.

For me as well. Never ever done this before and not willing or learn to do this.
I'm not the brighest light 💡  on chrismas tree 🎄  and my life time is running out.  👴 
🙈   😃 

So then, it's up to you. But I don't belive it's be worth the effort

MichaIng, 2018-02-24:
@k-plan 
> H3 stuff will be the correct one for OPi - One, - PC, - Plus, - 2, ...

Ah okay, good to know. Hmm yeah, there is other things to do and I don't want to start with developing on SBC images right now, focusing with limited time on topics that I can develop/test on VMs.

Especially with Fourdee time-out right now, I guess it would be fair enough to e.g. provide your image for testing/impressions to our OPi users. We can collect their feedback/issues and see if it has a chance.

Hmm on the other hand, I just reviewed the image creation steps. that can be done on VM. Looks not too hard and kernel 4.9 is possible: https://github.com/orangepi-xunlong/orangepi_h3_linux/blob/master/README.md
Maybe I could create a base image with this and someone with related OPi can try booting on device and run PREP. Maybe the outcome from these are cleaner then existing OPi images. I will decide tomorrow spontaneously, if I'm in mood 😛.

k-plan, 2018-02-24:
@MichaIng 

> But indeed you were right, this images from OPi are a mess.
> Seems they more less dropped support for the device, this image applies to.

Have said this  more then once before ...  but 🙉  .   😃  🙊 
If you pay peanuts, you will get monkeys.

> Looks not too hard and kernel 4.9 is possible

As far as I have notice and read, we should stay far, far away form Mainline Kernel (4.x) with Allwinner devices. 
Better to have a stable working Legacy Kerne (3.x).

The really bad problems have been started, since we have switch to 4.x with DietPi 6.x.

http://linux-sunxi.org/Category:H3_Boards

example:
http://linux-sunxi.org/Xunlong_Orange_Pi_One_%26_Lite

k-plan, 2018-02-24:
@MichaIng 

If you are interssed in UART debug output of running OPi-One build image:

```
DietPi-OrangePi-One login: [ 1818.381647] Restarting system.
[ 1818.388461] Restarting Linux version 3.4.112 (root@curry) (gcc version 4.6.3 20120201 (prerelease) (crosstool-NG linaro-1.13.1-2012.02-20120222 - Linaro GCC 2012.02) ) #1 SMP PREEMPT Thu Sep 22 15:08:09 CST 2016
[ 1818.388475] 
HELLO! BOOT0 is starting!
boot0 version : 4.0.0
fel_flag = 0x00000000
rtc[0] value = 0x00000000
rtc[1] value = 0x00000000
rtc[2] value = 0x00000000
rtc[3] value = 0x00000000
rtc[4] value = 0x00000000
rtc[5] value = 0x00000000
rtc[6] value = 0x00000000
rtc[7] value = 0x00000000
DRAM DRIVE INFO: V0.8
DRAM Type = 3 (2:DDR2,3:DDR3,6:LPDDR2,7:LPDDR3)
DRAM CLK = 672 MHz
DRAM zq value: 003b3bfb
READ DQS LCDL = 001f1e1f
DRAM SIZE =512 M
odt delay 
dram size =512
card boot number = 0
card no is 0
sdcard 0 line count 4
[mmc]: mmc driver ver 2014-12-10 21:20:39
[mmc]: ***Try SD card 0***
[mmc]: SD/MMC Card: 4bit, capacity: 3781MB
[mmc]: vendor: Man 00035344 Snr 70d3fa56
[mmc]: product: SU04G
[mmc]: revision: 8.0
[mmc]: ***SD/MMC 0 init OK!!!***
sdcard 0 init ok
The size of uboot is 000dc000.
sum=ede66ee7
src_sum=ede66ee7
Succeed in loading uboot from sdmmc flash.
Ready to disable icache.
Jump to secend Boot.
SUNXI_NORMAL_MODE   
[      0.351]e mode

U-Boot 2011.09-rc1 (Jun 21 2015 - 19:33:17) Allwinner Technology 

[      0.358]version: 1.1.0
normal mode
[      0.366]pmbus:   ready
not set main pmu id
axp_probe error
sunxi_set_cpux_voltage_by_i2c error 
[      0.384]PMU: pll1 1200 Mhz,PLL6=600 Mhz
AXI=400 Mhz,AHB=200 Mhz, APB1=100 Mhz 
sid read already 
fel key new mode
run key detect
no key found
no key input
dram_para_set start
dram_para_set end
normal mode
[      0.414]DRAM:  512 MiB
relocation Offset is: 15af9000
[box standby] read rtc = 0x0
[box_start_os] mag be start_type no use
user_gpio config
[... 638 lines cut ...]
Debian GNU/Linux 9 DietPi-OrangePi-One ttyS0

DietPi-OrangePi-One login: 
````

Because this boards are running very hot, have change cpu governor to `interactive` and take down dram frequence to `408 MHz`

```
root@DietPi-OrangePi-One:~# cpu
[  OK  ] Root access verified.

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     armv7l
 Temp         |     41'c : 105'f | Optimal temperature.
 Governor     |     interactive

                 Current Freq    Min Freq   Max Freq
 CPU0         |      480 MHz      480 MHz    1200 MHz
 CPU1         |      480 MHz      480 MHz    1200 MHz
 CPU2         |      480 MHz      480 MHz    1200 MHz
 CPU3         |      480 MHz      480 MHz    1200 MHz
```

k-plan, 2018-02-24:
```
root@DietPi-OrangePi-One:~# dmesg
[    0.000000] Booting Linux on physical CPU 0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Linux version 3.4.112 (root@curry) (gcc version 4.6.3 20120201 (prerelease) (crosstool-NG linaro-1.13.1-2012.02-20120222 - Linaro GCC 2012.02) ) #1 SMP PREEMPT Thu Sep 22 15:08:09 CST 2016
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: sun8i
[    0.000000] Ignoring tag cmdline (using the default kernel command line)
[    0.000000] bootconsole [earlycon0] enabled
[    0.000000] cma: CMA: reserved 160 MiB at 56000000
[    0.000000] Memory policy: ECC disabled, Data cache writealloc
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat c1028d80, node_mem_map c1106000
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 129920 pages, LIFO batch:31
[    0.000000] script_init enter!
[    0.000000] script_init exit!
[    0.000000] PERCPU: Embedded 8 pages/cpu @c1598000 s11904 r8192 d12672 u32768
[    0.000000] pcpu-alloc: s11904 r8192 d12672 u32768 alloc=8*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 129920
[    0.000000] Kernel command line: earlyprintk=ttyS0,115200 loglevel=8 initcall_debug=0 console=ttyS0,115200 console=tty0 init=/init
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] allocated 1048576 bytes of page_cgroup
[    0.000000] please try 'cgroup_disable=memory' option if you don't want memory cgroups
[    0.000000] Memory: 512MB = 512MB total
[    0.000000] Memory: 336204k/336204k available, 188084k reserved, 0K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff000000   ( 488 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc09ac288   (9873 kB)
[    0.000000]       .init : 0xc09ad000 - 0xc0face80   (6144 kB)
[    0.000000]       .data : 0xc0fae000 - 0xc102b7f8   ( 502 kB)
[    0.000000]        .bss : 0xc102bfac - 0xc1105a00   ( 871 kB)
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000]  Additional per-CPU info printed with stalls.
[    0.000000] NR_IRQS:544
[    0.000000] Architected local timer running at 24.00MHz.
[    0.000000] Switching to timer-based delay loop
[    0.000000] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 178956ms
[    0.000000] Console: colour dummy device 80x30
[    0.000000] console [tty0] enabled, bootconsole disabled
[    0.019903] Calibrating delay loop (skipped), value calculated using timer frequency.. 4800.00 BogoMIPS (lpj=24000000)
[    0.019958] pid_max: default: 32768 minimum: 301
[    0.020232] Mount-cache hash table entries: 512
[    0.021077] Initializing cgroup subsys cpuacct
[    0.021106] Initializing cgroup subsys memory
[    0.021162] Initializing cgroup subsys devices
[    0.021187] Initializing cgroup subsys freezer
[    0.021210] Initializing cgroup subsys blkio
[    0.021244] Initializing cgroup subsys perf_event
[    0.021306] CPU: Testing write buffer coherency: ok
[    0.021368] ftrace: allocating 26612 entries in 79 pages
[    0.050265] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.050298] [sunxi_smp_prepare_cpus] enter
[    0.050340] Setting up static identity map for 0x406f07c8 - 0x406f0820
[    0.051167] CPU1: Booted secondary processor
[    0.051167] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.051299] CPU2: Booted secondary processor
[    0.051299] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.051299] CPU3: Booted secondary processor
[... 415 lines cut ...]
[   14.022975] gpio name is PA21, ret = 0
[   14.023089] gpio name is PA7, ret = 0
[   14.023201] gpio name is PA8, ret = 0
[   14.023310] gpio name is PG8, ret = 0
[   14.023440] gpio name is PA9, ret = 0
[   14.023546] gpio name is PA10, ret = 0
[   14.023653] gpio name is PG9, ret = 0
[   14.023775] gpio name is PG6, ret = 0
[   14.023893] gpio name is PG7, ret = 0
[   14.024047] gpio_init finish with uesd 1!
[   14.531581] systemd[1]: kmod-static-nodes.service: Failed to set invocation ID on control group /system.slice/kmod-static-nodes.service, ignoring: Operation not supported
[   14.660558] systemd[1]: Starting Create list of required static device nodes for the current kernel...
[   14.682668] systemd[1]: Mounted Debug File System.
[   14.693680] systemd[1]: Started Journal Service.
[   16.247038] systemd-journald[181]: Received request to flush runtime journal from PID 1
[   16.436058] vmouse_input_dev_open
[   16.490113] vmouse_input_dev_close
[   16.972954] gmac0: probed
[   16.979600] gmac0 gmac0: eth0: eth0: PHY ID 00441400 at 0 IRQ poll (gmac0-0:00)
[   20.970220] PHY: gmac0-0:00 - Link is Up - 100/Full
[   32.108896] [ddrfreq] DDR: 672000KHz->408000KHz start
[   32.108911] [ddrfreq] current cpu is cpu2
[   32.111669] [ddrfreq] elapsed: 1050us
[   32.111689] [ddrfreq] DDR: 672000KHz->408000KHz ok!
```

k-plan, 2018-02-26:
@MichaIng 

> Especially with Fourdee time-out right now, I guess it would be fair enough to e.g. provide your image for testing/impressions to our OPi users. We can collect their feedback/issues and see if it has a chance.

Hmm ... have sleeped and think over this suggestion.
At the moment, I think, no. Why?

- Build image will be updated now to v6.2 and so where is:

[image]

Sorry, simple a no-go. 😄 

- Have one one device (OPi-One) to test with.

- It's Fourdee project, so I would like, he is able, to have a look into this, before scatter something with his name.
But it will be okay for me, you are the boss at the moment.

- Have no access to upload something to dietpi.com. (It's okay, please leave it as it is. Don't need or want it.)
Don't want to upload something to a free file hosting service, which is out of your control. (unprofessional)

------------

Do a little bit of research.
It seem to me, provided `debian_xfce_desktop_Orangepione_v0_9_1.img` is based on work of "loboris" (https://github.com/loboris)

http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=342

Where is as well a [`Debian_jessie_mini.img`](https://drive.google.com/drive/folders/0B1hyW7T0dqn6fndnZTRhRm5BaW4zVDVyTGlGMWJES3Z1eXVDQzI5R1lnV21oRHFsWnVwSEU), but with a old Kernel `3.4.39` (29.08.2015).
Last Kernel update, he provided is `3.4.39-01-lobo` (25.10.2015)

Last kernel source from your link is [kernel 3.4.113](https://github.com/orangepi-xunlong/orangepi_h3_linux/tree/master/OrangePi-Kernel) is 25.04.2017.
But have to be compile from source. Nothing I can do.

MichaIng, 2018-02-26:
@k-plan 
- The ARMbian credits are removed for v6.3.
- Jep you are right, we should wait for Fourdee, just gathering possibilities for now.
- Check out: https://github.com/Fourdee/DietPi/issues/1557 In case the Ubuntu image is available only, or Debian image is way older or messy. And this was Ubuntu Artful image, where user noted that it is more different to current Debian than older versions (usually found for OPi/BPi, Xenial). By times I will implement experimental Ubuntu integration into PREP.

k-plan, 2018-02-26:
@MichaIng 

> The ARMbian credits are removed for v6.3.

Yes, see this . :+1:  So we have to wait ...  :wink: 

> Jep you are right, we should wait for Fourdee ...

:+1:  Yes, thank you. :smile: 

> Check out: #1557 In case the Ubuntu image is available only, or Debian image is way older or messy

Yes, read this. Thank you for testing and probing.
Check nearly all available Orange-Pi One images. Ubuntu and Lubuntu be older and have more  outdated kernel.

> ...  just gathering possibilities for now.

Option for improvement so far I can see, maybe the last (newer) kernel, but I'm unable to compile this.
Out there are some kernel patches (see given links), but don't know, how to work it out.
We can add some settings into `dietpi-config` for thermal improvement e.g. dram frequency lowering and / or changing default cpu governor or lowering  default cpu frequency to get a cooler, stable running board.
Changing / Setting HDMI output resolution is a PITA, dealing with `flex` and `script.bin`.

MichaIng, 2018-02-26:
Okay I tested to follow the build instructions from: https://github.com/orangepi-xunlong/orangepi_h3_linux
- Besides the build-essentials (via dietpi-software), following packages are necessary for cross-compiling on VM and python:
`apt install gcc-arm-linux-gnueabi device-tree-compiler python2.7-minimal`
- As somehow the script wants python binary at /usr/bin/python2: `ln -s /usr/bin/python2.7 /usr/bin/python2`
- Creating mainline kernel finally failed with:
```
Traceback (most recent call last):
  File "./tools/binman/binman", line 16, in <module>
    import unittest
ImportError: No module named unittest
Makefile:1102: recipe for target 'u-boot-sunxi-with-spl.bin' failed
make: *** [u-boot-sunxi-with-spl.bin] Error 1
```
- No idea how to resolve: Seems to be related to missing unittest in PYTHONPATH, but actually as I start the script from within source directory...: https://stackoverflow.com/questions/3646307/python-importerror-no-module-named-problem
- Tried it again with fix by PR: https://github.com/orangepi-xunlong/orangepi_h3_linux/pull/4 ... same result.
- Tried to follow instructions for 3.4 kernel, but this fails without further verbose output:
```
root@VM-Stretch:~/orangepi_h3_linux-master/OrangePi-Kernel# ./build_linux_kernel.sh one
Building kernel for OPI-one () ...
  Configuring ...
  Building kernel & modules ...
  Error: KERNEL NOT BUILT.
```
- Maybe I also did something wrong, hehe no experience with all this 😉.
________________________
Following http://linux-sunxi.org/Xunlong_Orange_Pi_One_%26_Lite#Mainline_U-Boot, the way more promising, current and active developed kernel seems to be: https://github.com/megous/linux/tree/orange-pi-4.15
- But the build instructions are too rudimentary for me to start with it this night. Maybe someone has more experience and knows how to do.
________________________
Just found Debian offering u-boot ready images for some Allwinner devices, where then the default Debian installer can run on:
- http://ftp.uk.debian.org/debian/dists/stretch/main/installer-armhf/current/images/netboot/SD-card-images/
- Here another link for daily images: https://d-i.debian.org/daily-images/armhf/daily/u-boot/
- And instructions: https://wiki.debian.org/InstallingDebianOn/Allwinner
________________________
Okay, now really time to sleep 💤.

k-plan, 2018-02-27:
@MichaIng 

many thanks for all this testing and investigation. A big :+1: 

> - Tried to follow instructions for 3.4 kernel, but this fails without further verbose output:
> ```
> root@VM-Stretch:~/orangepi_h3_linux-master/OrangePi-Kernel# ./build_linux_kernel.sh one
> Building kernel for OPI-one () ...
>   Configuring ...
>   Building kernel & modules ...
>   Error: KERNEL NOT BUILT.
> ```
> - Maybe I also did something wrong, hehe no experience with all this :wink 

Same experience on my end.  Sorry.
Now I know, it's not me. :bulb:  :christmas_tree:   :older_man:   |   :smiley: 

--------------------

Have play before a little bit with Debian offered  `u-boot` , `uImage` and `device-tree` stuff for Allwinner.
This don't amount to much (for me). Sorry, no skill.
Documentation and reverence devices are a little bit outdated (2015/2016).

Most things are based on https://linux-sunxi.org/Linux_Kernel or  link [to this](https://linux-sunxi.org/Linux_Kernel#External_links).

----------------------

To bring it on a point:

- we both are unable to build a `kernel/u-boot/uImage` form Xunlong (OrangePi) given sources. 
- we can stay on `kernel 3.4.112 (22.10.2016)` and not able to get   `kernel 3.4.113 (25.04.2017)`.
- we have a untested DietPi v6.x for OPi-One with Debian Stretch, based on Xunlong (OrangePi) provided `debian_xfce_desktop_Orangepione_v0_9_1.img` with `kernel 3.4.112`
- it can be possible (untested), this image can work on more devices, like OPi-PC, -PLUS, -2, ... all based on Allwinner H3 platform.
- for getting a working HDMI output, user have to fiddle around with `fex`  and `script.bin`
- devices are running very hot (not a new finding), we can possible temper this a little bit, but it will cost a bit overall performance, can win stability.

I personally would like to stay away from Mainline Kernel.

MichaIng, 2018-02-27:
@k-plan 
Jep, nice summary 👍.

As far as I could read, the mainline kernel seams to be stabilized with the latest versions. But indeed someone with some kernel building experience would need to try it. I am not even sure, if this cross-compiling script would have worked, the outcome is really something usable, of if you have to take care, to take the right compiler versions etc. etc. or at best do all this on the actual device you want to use the image on 😆.

Generally I agree, if the official 3.4 kernel is proved working stable, why not just use it. There might be some performance or even security concerns, but DietPi does not rely on any new kernel feature as far as I know and at least in terms of stability and functionality (hdmi etc.) it will be probably best.

palik42, 2018-02-28:
I found this maybe some help.. https://github.com/Allwinner-Homlet?tab=repositories

MichaIng, 2018-03-01:
Ah this is the official allwinner firmware releases, so there are non mainline 4.4 + 4.9 beta kernel+bootloader available, do I get this right?

SuBLiNeR, 2018-03-08:
Hi there, i'm getting an orangePi Zero on next Monday, so if there is any chance to help you with some stuff i'm here.
I'd love to see DietPi on this device, as it is currently best image ever for our SBCs.

Fourdee, 2018-06-29:
Marking as closed. 

We will not support a device which lacks a good standard build quality, and, basic level of software support from the manufacture.

As an alternative to OPi, we highly recommend NanoPi devices.

closed 2018-06-29 by Fourdee