MichaIng/DietPi #1551: DietPi image request for NanoPi Duo.

parthgujar, 2018-02-23:
http://www.friendlyarm.com/index.php?route=product/product&product_id=197

## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

kevinshane, 2018-03-10:
+1 yes please, looks like it's very nice sbc

kevinshane, 2018-03-12:
Hi @Fourdee Dan, I've falshed the NanoPi Neo image http://dietpi.com/downloads/images/DietPi_NanoPiNEO-ARMv7-Stretch.7z into the NanoPi Duo, it's working great even the image says support Neo only, please consider modifying the Neo image to suit for Duo, also please add transfer rootfs to ssd function (armbian and ubuntu currently already supported) Thank you so much.

I also try to transfer rootfs to ssd and follow this link to make a ssd rootfs, but it fails
http://www.friendlyarm.com/Forum/viewtopic.php?f=58&t=951

closed 2018-07-09 by Fourdee