MichaIng/DietPi #1552: Image | Odroid Stretch review

Fourdee, 2018-02-23:
Meveric's images:
- C2 | https://forum.odroid.com/viewtopic.php?f=138&t=27449#p194153
- XU4 | 🈴 | Kodi https://forum.odroid.com/viewtopic.php?t=26608#p185562
- C1 | 🈴 | https://forum.odroid.com/viewtopic.php?f=114&t=30053

---

- XU4 4.14: https://github.com/hardkernel/linux/tree/odroidxu4-4.14.y

LexiconCode, 2018-03-09:
If Kodi/GPU is holding up Odroid Stretch could you release the without If could be updated Diet pi later?

Fourdee, 2018-03-09:
@LexiconCode 

C2 Stretch is done (minus GPU/KODI):
http://dietpi.com/downloads/images/DietPi_OdroidC2-ARMv8-Stretch.7z

I've plans to do the XU4 and C1 stretch images aswell. Will post them here when completed.

k-plan, 2018-03-19:
@Fourdee 

> C2 Stretch is done (minus GPU/KODI):

- C2 | 🈯️ img | 🈯️ gpu | 🈴 Kodi | 🈴  pivpn

[image]

LexiconCode, 2018-03-21:
As soon as you have an image for Odroid XU4 an Odroid N2 I contest test well.

Fourdee, 2018-03-31:
XU4 Stretch RC image by Meveric:
http://fuzon.co.uk/meveric/images/Stretch/Debian-Stretch-1.0~RC1-20180324-XU3-XU4.img.xz

LexiconCode, 2018-04-13:
`apt update`
> Err:34 http://ftp.debian.org/debian stretch/main Translation-en
>   Write error - write (28: No space left on device)
> Fetched 14.4 MB in 9s (1,512 kB/s)
> Reading package lists... Done
> E: Failed to fetch http://ftp.debian.org/debian/dists/stretch/main/i18n/Translation-en  Write error - write (28: No space left on device)
> E: Some index files failed to download. They have been ignored, or old ones used instead.

> Filesystem      Size  Used Avail Use% Mounted on
> udev            931M     0  931M   0% /dev
> tmpfs           200M   11M  189M   6% /run
> /dev/mmcblk1p2  464M  422M   39M  92% /
> tmpfs           998M     0  998M   0% /dev/shm
> tmpfs           5.0M     0  5.0M   0% /run/lock
> tmpfs           998M     0  998M   0% /sys/fs/cgroup
> /dev/mmcblk1p1   71M   31M   41M  43% /boot

Fourdee, 2018-04-13:
@LexiconCode 

Please paste the rootFS resize logs:
```
cat /var/tmp/dietpi/logs/fs_*
```

LexiconCode, 2018-04-13:
XU4 Stretch RC image by Meveric on the Odroid N2. without dietpi scripts. Which now that I've looked at it resize the partitions.

Fourdee, 2018-05-04:
XU4 stretch + 4.14:
- if ```console=ttySAC2,115200n8``` is missing from ```boot.ini```, hangs after ```starting kernel``` and does not continue to boot.

Fourdee, 2018-05-05:
XU4/C2 stretch images added to dietpi.com:
https://twitter.com/DietPi_/status/992787751262179328

Just need to verify kodi packages again, then check/do C1.

Fourdee, 2018-05-05:
Kodi status:
- C2 🈯️ 
- XU4 🈯️ 

Great work Meveric!!!

Fourdee, 2018-05-05:
I'am going to leave the C1 as Jessie for now. Lack any Debian Stretch image, and, time to really invest in achieving it.

Marking as completed.

closed 2018-05-05 by Fourdee

LexiconCode, 2018-05-05:
If were used an image down loaded to day do I need to redownload?

Fourdee, 2018-05-05:
> If were used an image down loaded to day do I need to redownload?

If the following says "Jessie", then yes
```
cat /etc/os-release
```