MichaIng/DietPi #1554: Configuration de la sécurité des transports Strict

Virusdead49, 2018-02-23:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Virusdead49, 2018-02-23:
would it be possible to get help on this subject 
thanks
http://dietpi.com/phpbb/viewtopic.php?f=11&t=2804

MichaIng, 2018-02-23:
Answered: http://dietpi.com/phpbb/viewtopic.php?f=11&t=2804&p=11153#p11153

Virusdead49, 2018-02-24:
I think I have a new problem creating a new certificate.

[image]

MichaIng, 2018-02-24:
@Virusdead49 
Respect the side note I gave 😉: Please replace the strings `old.domain.com` and `new.domain.com` with your **actual old and new domain names** 🙂. Especially `old.domain.com` explicitly NEEDs to be exactly the same domain, that you used, when you ran `dietpi-letsencrypt` first!

If this went fine, then try to access your Nextcloud via new domain. The Apache configuration can be left out, I think. Otherwise to edit those, you need to open via text editor, e.g. `nano`:
`nano /etc/apache2/apache.conf`, but you should know what you are doing there, so let me give you some safer replacement commands:

```
sed -i 's/^[[:blank:]]*ServerName/ServerName nextcloud-myweb.ddns.net/' /etc/apache2/apache2.conf
sed -i 's/^[[:blank:]]*ServerName/ServerName nextcloud-myweb.ddns.net/' /etc/apache2/sites-available/000-default.conf
sed -i 's/^[[:blank:]]*ServerName/ServerName nextcloud-myweb.ddns.net/' /etc/apache2/sites-available/000-default-le-ssl.conf
```

Virusdead49, 2018-02-24:
So I wrote down the address on dietpi-letsencrypt, but I'm looking for one on google.
I wrote on the order:
cd /etc/apache2/sites-available-sites
Is
sudo nano 000-default. conf
[image]
[image]

MichaIng, 2018-02-24:
@Virusdead49 
So looks good, the redirection method is not the one, implemented by CertBot by default (via mod_rewrite), but it is even better/native (via mod_alias), if there is no special configuration necessary.

I would remove the line `ErrorLog /error.log`, as this is net already to the more common `/var/log/apache2/error.log` within `/etc/apache2/apache2.conf`: https://github.com/Fourdee/DietPi/blob/master/dietpi/conf/apache2_jessie.conf#L18

Btw:
If you only use Nextcloud on your server, and no other web service besides, you could make your Nextcloud reachable via `nextcloud-myweb.ddns.net` instead of `nextcloud-myweb.ddns.net/nextcloud`, by changing the Apache2 document root to the nextcloud subfolder:
```
sed -i 's|^[[:blank:]]*DocumentRoot|DocumentRoot /var/www/nextcloud|' /etc/apache2/apache2.conf
sed -i 's|^[[:blank:]]*DocumentRoot|DocumentRoot /var/www/nextcloud|' /etc/apache2/sites-available/000-default.conf
sed -i 's|^[[:blank:]]*DocumentRoot|DocumentRoot /var/www/nextcloud|' /etc/apache2/sites-available/000-default-le-ssl.conf
sed -i '/^[[:blank:]]*Alias/d' /etc/apache2/sites-available/nextcloud.conf
```
But this will break other web services and their installations via dietpi-software, as for flexibility we expect and place most web services into their own sub folder.

ludji49, 2018-02-24:
I _just deleted ErrorLog /error. log

Then I have to write it's cmd line._

> sed -i '\|^[[:blank:]]*DocumentRoot|c\DocumentRoot /var/www/nextcloud' /etc/apache2/apache2.conf
> sed -i '\|^[[:blank:]]*DocumentRoot|c\DocumentRoot /var/www/nextcloud' /etc/apache2/sites-available/000-default.conf
> sed -i '\|^[[:blank:]]*DocumentRoot|c\DocumentRoot /var/www/nextcloud' /etc/apache2/sites-available/000-default-le-ssl.conf
> sed -i '/^[[:blank:]]*Alias/d' /etc/apache2/sites-available/nextcloud.conf

MichaIng, 2018-02-24:
@ludji49 
Jep, then your web server root should be switched to /nextcloud subdirectory, thus Nextcloud accessible directly at your domain.

One thing to do, if it works correctly, switch Nextcloud cli requests to root as well then:
`sed -i 's|localhost/nextcloud|localhost|' /var/www/nextcloud/config/config.php`

ludji49, 2018-02-24:
I just put the line commands everything seems to be working normally. 
something I don't understand:
`if it works correctly, switch Nextcloud cli requests to root `
by I have to pass or must I put this line command `sed -i' s|localhost/nextcloud|localhost|' /var/www/nextcloud/config. php`

MichaIng, 2018-02-24:
Yes, run this please, otherwise you will see errors in Nextcloud log, on every cron job execution:
`sed -i 's|localhost/nextcloud|localhost|' /var/www/nextcloud/config/config.php`

ludji49, 2018-02-25:
Thank you a lot for your valued assistance

MichaIng, 2018-02-25:
@ludji49 
About memory caching:
Could you please copy&paste the content of your `/var/www/nextcloud/config/config.php`? Leave private information, passwords etc. out of course.

About dietpi-backup:
By default the backup will be stored within `/mnt/dietpi-backup`, which is indeed on your SD card. Please select `dietpi-backup > Location > List`, where you should see all mounted drives, including your USB drive. Select it and run backup. The backup should be stored on your USB drive within `dietpi-backup/` folder.
[image]

ludji49, 2018-02-25:
copy&paste the content of your /var/www/nextcloud/config/config.php
Do I need to remove my SD card?

MichaIng, 2018-02-25:
@ludji49 
Nope, leave your SD card in place, just do
`cat /var/www/nextcloud/config/config.php` and paste the text output on your terminal here, erasing sensible information.

ludji49, 2018-02-25:
_here but information_

[image]

MichaIng, 2018-02-25:
Huh, okay, yeah I see now the problem. The guide you used (https://www.novaspirit.com/2018/01/19/raspberry-pi-owncloud-install-with-diet-pi/) is outdated (even that it was posted not too long ago), in terms that it suggests to do all the database creation and userdata manually, outside of dietpi_userdata. Since end of last year already, this is not necessary any more, furthermore the steps from the guide make it difficult to recreate intended installation state of ownCloud/Nextcloud, the deletion of `/var/www/owncloud/config/config.php` purge all settings and optimizations: memory cache, redis file locking and some others...
I left a comment about that and hope for rework.

Okay let me see how to fix this best for you:
```
sed -i '\|SOFTWARE_NEXTCLOUD_DATADIR|c\SOFTWARE_NEXTCLOUD_DATADIR=/datastore' /DietPi/dietpi.txt
sed -i "\|'overwrite.cli.url'|c\  'overwrite.cli.url' => 'https://nextcloud-myweb.ddns.net'," /var/www/nextcloud/config/config.php
dietpi-software reinstall 114
```

Please report, if you see any error message of failure within reinstallation process. But as far as I reviewed the guide and our installation script, this should work well.

ludji49, 2018-02-26:
I have the page but it doesn't open.

[image]

and if I go through the windows software I have an error

[image]

MichaIng, 2018-02-26:
@ludji49 
Okay installation looks good, access via https generally works (webserver works), this seems to be now some left PHP/Nextcloud config issue:

Could you please give output of:
```
php -m
cat /var/www/nextcloud/config/config.php
tail /var/log/apache2/error.log
tail /datastore/nextcloud.log
```

ludji49, 2018-02-26:
Here is the result

[image]

[image]

MichaIng, 2018-02-26:
@ludji49 
€: Ah found one issue in combination with nextcloud in web server root, do:
sed -i "s|'/nextcloud',|'/',|" /var/www/nextcloud/config/config.php
ncc maintenance:update:htaccess
____
Hmm no current error. Your apache error log seems on a different location? Check out /var/log where it is or try
journalctl -t apache2 

Otherwise maybe it's just browser cache: try ctrl+F5 to force reload with clearing browser cache.

ludji49, 2018-02-26:
by putting this line the site is open + /var/log
sed -i "s|'/nextcloud',|'/',|" /var/www/nextcloud/config/config.php

[image]
[image]

MichaIng, 2018-02-26:
@ludji49 
Ah great finally 😄.
As this, making Nextcloud available directly on domain, instead of `my.domain.org/nextcloud` is a usual reasonable wish, I will collect the necessary steps here.
But as this currently breaks other web service software we offer (relying on web server root at /var/www), we need to make this clear, with need to revert the previous steps to re-enable installation of other web software
________________________
To format your USB flash drive, start:
`dietpi-drive_manager`

Select your USB flash there, assure that it is really the desired USB drive not root, boot or your nextcloud data storage 😉. Then you can format there to ext4 filesystem.

ludji49, 2018-02-26:
I don't understand what happened the site is inaccessible

ludji49, 2018-02-26:
_format your USB flash drive:_
_But after formatting my USB flash drive is still not in ext4_

[image]
[image]
[image]

MichaIng, 2018-02-26:
@ludji49 
You first need to unmount the drive, as noted on the screen, after hitting "Format" 😉.
Select "Unmount" above first, then "Format".

I guess the mount point gets changed, thus within dietpi-backup afterwards, you need to select the Location again.

ludji49, 2018-02-27:
Hi, I can't see my flash drive anymore. 
[image]

MichaIng, 2018-02-27:
@ludji49 
Ah, jep after unmounting, if you are going back into overview, you see just the mounted drives. Exit drive manager, then remount:
`mount /dev/sdb5 /mnt/346D-27AA`
`dietpi-drive_manager`
then do the same, BUT: after unmount the drive, directly go on format:
[image]

ludji49, 2018-02-28:
I did it again from the beginning, according to you. 
[image]

MichaIng, 2018-02-28:
@ludji49 
Jep 👍. As you can see, the name/mount point of the drive has changed into a long so called "UUID". You have to reset your backup location to this according: https://github.com/Fourdee/DietPi/issues/1554#issuecomment-368311492

ludji49, 2018-02-28:
Hi, tell me if I'm on the right track.
[image]

MichaIng, 2018-02-28:
@ludji49 
Jep that is right

ludji49, 2018-02-28:
He's working, I hope it'll be good and no mistakes.
[image]

ludji49, 2018-02-28:
Do you think that this program is useful or not I installed it with everything but program like nextcloud, apache..........
[image]

MichaIng, 2018-03-01:
@ludji49 
Your USB flash drive has not enough space to take whole system backup. Actually I see now the issue with placing Nextcloud data directory mount point to `/datastore` instead of into usual mount point location `/mnt/...`. The backup tries to copy your whole Nextcloud data onto the USB flash drive as well.

But dietpi-backup has an own include/exclude method: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=256#p255
In your case do the following:
```
echo '/datastore' >> /DietPi/dietpi/.dietpi-backup_exclude
rm -R /mnt/62ce04db-11c1-4675-93fa-176939b6eb45/*
dietpi-backup 1
```

MichaIng, 2018-03-13:
I'll mark this as closed, as TOs issues have been resolved. Feel free to reopen if needed.

closed 2018-03-13 by MichaIng