MichaIng/DietPi #1555: INFO: On leave

Fourdee, 2018-02-24:
Hi all,

I will be taking a break from DietPi completely for at least a week, due to RL + ill health. I may be able to come back sooner, unknown at this time.

@MichaIng 
I've removed requirements for PR reviews. You should be able to accept merges from now on 👍 

@k-plan @MichaIng 
I'd be grateful if you could both continue with the work you do, and, help keep the ship running if/when you can. However, its never a requirement, as always in your own time, when/if you can.

k-plan, 2018-02-24:
@Fourdee 

Think nothing of it! It's okay. No problem.

Get well soon!

MichaIng, 2018-02-24:
@Fourdee 
Jep, take care your healthiness, children and wife 😃!

Thanks for merge permissions. I will do careful use. Waiting for your return before v6.3 release anyway 😉.

symbios24, 2018-02-25:
Be well soon :)

SuBLiNeR, 2018-02-25:
@Fourdee Take your time and get well soon! wink

Invictaz, 2018-02-25:
Get well soon :)

ywjheart, 2018-02-27:
Get well soon! :)

blogabe, 2018-02-27:
enjoy the time away with yours

Fourdee, 2018-02-28:
Thanks all.

Can't keep away, need something to do in-between 🤢 and 🛌. Take each day as it comes :)

closed 2018-02-28 by Fourdee

k-plan, 2018-02-28:
Nice to see you back, Fourdee. :+1:   :heart: