MichaIng/DietPi #1556: Installing Home Assistant on ODROID C1 leads to ruin

WineSoaked, 2018-02-25:
#### Required Information:
- DietPi Version | v6.2
- SBC Device     | Odroid C1 (armv7l) (index=10)
- Distro         | jessie (index=3)
- Command        | G_AGI: sqlite3
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:
Was running the dietpi-software on first install. Expected an error-free installation.

#### Actual behaviour:
On dietpi-software (first install) and then on a retry with dietpi-launcher, got several "error handler" messages (that were resumable) for ffmpeg and a raft of dev tools:

[ SUB1 ] DietPi-Software > Installing FFmpeg: audio & visual libary
[ INFO ] DietPi-Software | APT installation for: ffmpeg, please wait...
E: Unable to correct problems, you have held broken packages.

[FAILED] DietPi-Software | DietPi-Software: G_AGI: ffmpeg
[ INFO ] DietPi-Software | exit_code = 100
[ INFO ] DietPi-Software | VERSION:v6.2 | HW_MODEL:10 | HW_ARCH:2 | DISTRO:3
[ INFO ] DietPi-Software | Log file contents:
E: Unable to correct problems, you have held broken packages.
[ INFO ] DietPi-Software | If problems persist, please report this to DietPi for investigation, including a screenshot of this error! (https://github.com/Fourdee/DietPi/issues).
[ SUB1 ] DietPi-Software > Installing Home Assistant: open-source home automation platform
[ SUB1 ] DietPi-Software > Installing Home Assistant: open-source home automation platform
[ INFO ] DietPi-Software | HA_USER: homeassistant
[ INFO ] DietPi-Software | HA_USERROOT: /home/homeassistant
[ INFO ] DietPi-Software | HA_SRVROOT: /srv/homeassistant
[ INFO ] DietPi-Software | HA_PYENV_ACTIVATION: export PATH="/home/homeassistant/.pyenv/bin:$PATH"; eval "$(pyenv init -)"; eval "$(pyenv virtualenv-init -)"
[ INFO ] DietPi-Software | HA_PYTHON_VERSION: 3.6.3
[ INFO ] DietPi-Software | APT installation for: libssl-dev git cmake libc-ares-dev uuid-dev daemon curl libgnutls28-dev libgnutlsxx28 nmap net-tools sudo libglib2.0-dev libudev-dev swig libssl-dev libusb-1.0-0 gcc libssl-dev libffi-dev libbz2-dev zlib1g-dev libreadline-dev libsqlite3-dev libncurses5-dev libncursesw5-dev, please wait...
E: Unable to correct problems, you have held broken packages.

[FAILED] DietPi-Software | DietPi-Software: G_AGI: libssl-dev git cmake libc-ares-dev uuid-dev daemon curl libgnutls28-dev libgnutlsxx28 nmap net-tools sudo libglib2.0-dev libudev-dev swig libssl-dev libusb-1.0-0 gcc libssl-dev libffi-dev libbz2-dev zlib1g-dev libreadline-dev libsqlite3-dev libncurses5-dev libncursesw5-dev
[ INFO ] DietPi-Software | exit_code = 100
[ INFO ] DietPi-Software | VERSION:v6.2 | HW_MODEL:10 | HW_ARCH:2 | DISTRO:3
[ INFO ] DietPi-Software | Log file contents:
E: Unable to correct problems, you have held broken packages.
[ INFO ] DietPi-Software | If problems persist, please report this to DietPi for investigation, including a screenshot of this error! (https://github.com/Fourdee/DietPi/issues).
[ INFO ] DietPi-Software | APT installation for: libmysqlclient-dev, please wait...

#### Steps to reproduce:
1. Take an ODROID C1
2. Do a vanilla install of DietPi, selecting Home Assistant as an additional package
3. Wait while a lot of software gets downloaded

#### Additional logs:
```
Log file contents:
E: Unable to correct problems, you have held broken packages.
```

WineSoaked, 2018-02-26:
Additional notes:

Installing libglib2.0-dev yields the following:

```
# apt-get install libglib2.0-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libglib2.0-dev : Depends: libglib2.0-0 (= 2.42.1-1+b1) but 2.48.0-1~bpo8+1 is to be installed
                  Depends: libglib2.0-bin (= 2.42.1-1+b1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

WineSoaked, 2018-02-26:
Additional notes:

Installing libsqlite3-dev yields the following:

```
# apt-get install libsqlite3-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libsqlite3-dev : Depends: libsqlite3-0 (= 3.8.7.1-1+deb8u2) but 3.16.2-3~bpo8+1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2018-02-26:
@WineSoaked 
Thanks for your report.

Hmm packages from different repos seem to be mixed, Debian stable and backports at least. Usually this should be no problem, as backports contain all necessary dependencies, but if other packages then depend on the stable version packages... should no occur on just Debian repo as far as I know, but in your case Odroid repos join as well.

I will try to replicate on Jessie VM. If no issue there, it is due to Odroid specific repo and we have to find a way to avoid those mixtures.

MichaIng, 2018-02-26:
@k-plan 
The additional repos should be generally no problem. But if some package is pulled from backports with it's dependencies (`libsqlite3-0` v3.16.2-3~bpo8+1) and then another software is installed without `-t jessie-backports` (`libsqlite3-dev`), then it want's to pull it's version similar dependency from non-backports and fails, as they were already needed by installed packages... Now I wrote a knot into my brain 🤣.

I will check our installation scripts about that.

WineSoaked, 2018-02-26:
I wanted to note that some of the cross-dependency problems look like they
require a very specific version of a dependency, such that when a more
recent version of the package is already loaded, it starts raining frogs.

I am well-versed in building ffmpeg for SBC's from scratch because the
package managers don't include it.

Thanks for all the knot-tying. There's no raging hurry on this, as this is
me experimenting with DietPi, and I have several other SBC's that can be
put into service in the meantime.

MichaIng, 2018-02-26:
@WineSoaked 
Jep, exactly the cross-dependencies seem to be the issue in your case. As said I am just wondering, why this does not happen on my VM. `ffmpeg` is available on default Debian repo since Stretch and for Jessie as said we pull it from backports. But the as well pulled dependency libraries are not related to sqlite or libglib where you face the issues.

Do you have a change to share the output of `apt-cache showpkg ffmpeg`, so I can see if there is something different on Odroids (with their additional repos).

WineSoaked, 2018-02-26:
I can do that tonight, I'm away from the Odroid right now.

WineSoaked, 2018-02-27:
For your viewing pleasure:
```
Package: ffmpeg
Versions:
7:3.4-3~bpo8u1 (/var/lib/apt/lists/fuzon.co.uk_meveric_dists_jessie_backports_binary-armhf_Packages)
 Description Language:
                 File: /var/lib/apt/lists/deb.debian.org_debian_dists_jessie-backports_main_binary-armhf_Packages
                  MD5: 032ff4ee68b923f5137379a7857cb8a8
 Description Language: en
                 File: /var/lib/apt/lists/deb.debian.org_debian_dists_jessie-backports_main_i18n_Translation-en
                  MD5: 032ff4ee68b923f5137379a7857cb8a8

7:3.2.10-1~deb9u1~bpo8+1 (/var/lib/apt/lists/deb.debian.org_debian_dists_jessie-backports_main_binary-armhf_Packages)
 Description Language:
                 File: /var/lib/apt/lists/deb.debian.org_debian_dists_jessie-backports_main_binary-armhf_Packages
                  MD5: 032ff4ee68b923f5137379a7857cb8a8
 Description Language: en
                 File: /var/lib/apt/lists/deb.debian.org_debian_dists_jessie-backports_main_i18n_Translation-en
                  MD5: 032ff4ee68b923f5137379a7857cb8a8

Reverse Depends:
  libav-tools,ffmpeg
  zoneminder,ffmpeg
  youtube-dl,ffmpeg
  vokoscreen,ffmpeg
  libav-tools,ffmpeg
  youtube-dl,ffmpeg 4:0.6
  youtube-dl,ffmpeg
  winff-qt,ffmpeg
  winff-gtk2,ffmpeg
  stopmotion,ffmpeg
  python-scitools,ffmpeg
  ripit,ffmpeg
  pacpl,ffmpeg
  motion,ffmpeg
  kmplayer,ffmpeg
  kino,ffmpeg
  jsymphonic,ffmpeg
  imagination,ffmpeg
  imagemagick-6.q16,ffmpeg
  multimedia-video,ffmpeg
  connectome-workbench,ffmpeg
  bitpim,ffmpeg
Dependencies:
7:3.4-3~bpo8u1 - libavcodec57 (18 7:3.4) libavcodec-extra57 (2 7:3.4) libavdevice57 (2 7:3.4) libavfilter6 (18 7:3.4) libavfilter-extra6 (2 7:3.4) libavformat57 (2 7:3.4) libavresample3 (2 7:3.4) libavutil55 (2 7:3.4) libc6 (2 2.7) libpostproc54 (2 7:3.4) libsdl2-2.0-0 (0 (null)) libswresample2 (2 7:3.4) libswscale4 (2 7:3.4) ffmpeg-doc (0 (null)) libav-tools (3 6:12~~) qt-faststart (3 7:2.7.1-3~) libav-tools (3 6:12~~) qt-faststart (3 7:2.7.1-3~)
7:3.2.10-1~deb9u1~bpo8+1 - libavcodec57 (18 7:3.2.10) libavcodec-extra57 (2 7:3.2.10) libavdevice57 (2 7:3.2.10) libavfilter6 (18 7:3.2.10) libavfilter-extra6 (2 7:3.2.10) libavformat57 (2 7:3.2.10) libavresample3 (2 7:3.2.10) libavutil55 (2 7:3.2.10) libc6 (2 2.7) libpostproc54 (2 7:3.2.10) libsdl2-2.0-0 (2 2.0.0) libswresample2 (2 7:3.2.10) libswscale4 (2 7:3.2.10) libva1 (2 1.0.3) ffmpeg-doc (0 (null)) libav-tools (3 6:12~~) qt-faststart (3 7:2.7.1-3~) libav-tools (3 6:12~~) qt-faststart (3 7:2.7.1-3~)
Provides:
7:3.4-3~bpo8u1 -
7:3.2.10-1~deb9u1~bpo8+1 -
Reverse Provides:
```

MichaIng, 2018-02-27:
Looks good, dependencies are identical, just version number of Meverics is higher, but actually we lowered it's backports pin priority in the past due to (obviously temporary) dependency errors (others than here): https://github.com/Fourdee/DietPi/issues/1273

I am not sure how it is handled now with different version numbers, but all of this should have no effect on `libsqlite3-0` being installed from backports: Separate APT call without any `-t` option, SQLite + deps should be pulled just from default Jessie repo.

Someone with Odroid has to investigate.

WineSoaked, 2018-02-27:
I'd be happy to investigate, but as I know nothing of the backport repo you
all have, I'm lacking a good bit of context. Oh, and free time, but as this
is my ticket, I'm somewhat invested.

Any rtfm available for your backport  dependency strategy?

MichaIng, 2018-02-27:
@WineSoaked 
I just checked, an we lower apt pin priority for the Odroid specific backports only for XU4 somehow. So on your device, it should be used instead of debian default backports. But I would wonder, if this generally causes errors, as Meverics image+repos is generally well tested/much used and shouldn't cause such problems on default apt installations, without much coding magic, which ffmpeg, sqlite, git and build-essentials all basically are.

But just to test/exclude, if you find time, try:
```
		cat << _EOF_ > /etc/apt/preferences.d/backports
Package: *
Pin: release a=jessie-backports
Pin: origin "fuzon.co.uk"
Pin-Priority: 99
_EOF_
```

Maybe you need to clean possible installed packages from those first, so no backports packages are left (check via `dpkg -l | grep bpo8`).
Then apt clean+update+retry installation via dietpi-software.

WineSoaked, 2018-02-28:
Will do, probably won't have a report until tomorrow (about 24 hours from
now).

WineSoaked, 2018-02-28:
In fact, I will reflash the emmc on the C1 with the image from the site, make the change to backports after the initial installation (i.e. I won't install any optional packages with `dietpi-software`) and will paste the results of trying to install HA here once I have done it. Again, not likely to happen until about 24 hours from now.

WineSoaked, 2018-03-01:
I am pleased to report that the updating the pin priority for apt was the trick. I did the following:

- Reflashed the image (`DietPi_v6.0_OdroidC1-ARMv7-Jessie.img`) to the 16GB eMMC that I am using as the root filesystem storage
- Ran the vanilla installation, but did not install anything additional using `dietpi-software`
- After the reboot, added the `/etc/apt/preferences.d/backports` file as specified above by @MichaIng 
- Installed Home Assistant using `dietpi-software`, the very lengthy (but successful) installation log follows

```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software

[ INFO ] DietPi-Software | SQlite will be installed
[ INFO ] DietPi-Software | Git will be installed
[ INFO ] DietPi-Software | Build-Essential will be installed
[ INFO ] DietPi-Software | FFmpeg will be installed
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Build-Essentials: common packages for compile

[ INFO ] DietPi-Software | APT installation for: build-essential make autoconf automake --no-install-recommends, please wait...
Extracting templates from packages: 100%
Selecting previously unselected package libgdbm3:armhf.
(Reading database ... 14648 files and directories currently installed.)
Preparing to unpack .../libgdbm3_1.8.3-13.1_armhf.deb ...
Unpacking libgdbm3:armhf (1.8.3-13.1) ...
Selecting previously unselected package perl-modules.
Preparing to unpack .../perl-modules_5.20.2-3+deb8u9_all.deb ...
Unpacking perl-modules (5.20.2-3+deb8u9) ...
Selecting previously unselected package perl.
Preparing to unpack .../perl_5.20.2-3+deb8u9_armhf.deb ...
Unpacking perl (5.20.2-3+deb8u9) ...
Selecting previously unselected package libasan1:armhf.
Preparing to unpack .../libasan1_4.9.2-10+deb8u1_armhf.deb ...
Unpacking libasan1:armhf (4.9.2-10+deb8u1) ...
Selecting previously unselected package libatomic1:armhf.
Preparing to unpack .../libatomic1_4.9.2-10+deb8u1_armhf.deb ...
Unpacking libatomic1:armhf (4.9.2-10+deb8u1) ...
Selecting previously unselected package libisl10:armhf.
Preparing to unpack .../libisl10_0.12.2-2_armhf.deb ...
Unpacking libisl10:armhf (0.12.2-2) ...
Selecting previously unselected package libcloog-isl4:armhf.
Preparing to unpack .../libcloog-isl4_0.18.2-1+b2_armhf.deb ...
Unpacking libcloog-isl4:armhf (0.18.2-1+b2) ...
Selecting previously unselected package libgomp1:armhf.
Preparing to unpack .../libgomp1_4.9.2-10+deb8u1_armhf.deb ...
Unpacking libgomp1:armhf (4.9.2-10+deb8u1) ...
Selecting previously unselected package libmpfr4:armhf.
Preparing to unpack .../libmpfr4_3.1.2-2_armhf.deb ...
Unpacking libmpfr4:armhf (3.1.2-2) ...
Selecting previously unselected package libsigsegv2:armhf.
Preparing to unpack .../libsigsegv2_2.10-4+b1_armhf.deb ...
Unpacking libsigsegv2:armhf (2.10-4+b1) ...
Selecting previously unselected package libubsan0:armhf.
Preparing to unpack .../libubsan0_4.9.2-10+deb8u1_armhf.deb ...
Unpacking libubsan0:armhf (4.9.2-10+deb8u1) ...
Selecting previously unselected package libmpc3:armhf.
Preparing to unpack .../libmpc3_1.0.2-1_armhf.deb ...
Unpacking libmpc3:armhf (1.0.2-1) ...
Selecting previously unselected package m4.
Preparing to unpack .../archives/m4_1.4.17-4_armhf.deb ...
Unpacking m4 (1.4.17-4) ...
Selecting previously unselected package patch.
Preparing to unpack .../patch_2.7.5-1_armhf.deb ...
Unpacking patch (2.7.5-1) ...
Selecting previously unselected package xz-utils.
Preparing to unpack .../xz-utils_5.1.1alpha+20120614-2+b3_armhf.deb ...
[... 1044 lines cut ...]

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Optimize and configure software

[ INFO ] DietPi-Software | Applying DietPi optimizations and configurations for Odroid C1 (armv7l), please wait...

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring Home Assistant: open-source home automation platform

[ SUB1 ] DietPi-Services > dietpi_controlled
[  OK  ] DietPi-Services | dietpi_controlled : cron
[  OK  ] DietPi-Services | dietpi_controlled : home-assistant

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installation completed

[  OK  ] DietPi-Software | The system will now reboot.
 This completes the DietPi-Software installation.
```

---
And so it seems that we have a solution of sorts. There was an existing file in `/etc/apt/preferences.d` that I did not modify or remove, just added the `backports` file as specified above, and all seems to be running normally, for now.

MichaIng, 2018-03-01:
@WineSoaked 
Thank you for testing. Jep our backports file seems to overwrite the existing due to alphabetical order 😉. So for now we need to add this file not just for XU4 but for all Odroid devices to avoid dependency issues.
`Preparing to unpack .../ffmpeg_7%3a3.2.10-1~deb9u1~bpo8+1_armhf.deb ...`
According to version, as expected Debian backports is used now.

I am still wondering why Meverics backports produce errors here. Would be interesting, what is actually different on his untouched image, as there for sure ffmpeg installation will work... Would be nice, if we could also manage this, as his repo simply provides the more current version and possibly optimized for Odroid devices.

@Fourdee 
If you find time, maybe you can have a look. Until then I will force non-Meveric backports for all Ordoids for v6.3.

Fourdee, 2018-03-01:
Reference: https://github.com/Fourdee/DietPi/pull/1571#issuecomment-369699010

WineSoaked, 2018-03-01:
Closing this now that the merge has happened. Glad it worked out smoothly.

closed 2018-03-01 by WineSoaked

alexthe-red, 2018-12-14:
FWIW I experienced the error: `DietPi homeassistant-start.sh[2025]: ModuleNotFoundError: No module named 'pytz'`
today while updating homeassistant and dietpi on Odroid C2. @WineSoaked 's suggestion of `cat << _EOF_ > /etc/apt/preferences.d/backports
Package: *
Pin: release a=jessie-backports
Pin: origin "fuzon.co.uk"
Pin-Priority: 99
_EOF_`
followed by running `sh updatehs.sh` from /srv/homeassistant/ fixed it. Thanks!!