MichaIng/DietPi #1557: DietPi-Prep | Test installation on top of Ubuntu image

MichaIng, 2018-02-26:
Jep, as many SBC producers offer Ubuntu, but no Debian images, I was interested in how far the Distros are actual different and if it is possible to convert them and install DietPi on top, via DietPi-Prep.

And finally, jep it is actually not a too big deal:
- First I checked, which Debian version was closest to the Ubuntu one: Ubuntu Artful -> Debian Stretch
- Allow our script to recognize Artful and go on as if Stretch was installed.
- `debian-archive-keyring` needs to be installed manually first of course: https://packages.debian.org/stretch/all/debian-archive-keyring/download
- Ubuntu seems to handle the locales a bid different: dpkg-reconfigure/update-gen seems to not use the list within `/etc/locale.gen`, but from somewhere else. `locale-gen --purge en_GB.UTF-8` works instead, purging everything else, but keeping/generating en_GB.UTF-8.
- I sadly didn't save some backup of original Ubuntu `/etc/environment` but after first PREP attempt, and reboot in between the PATH variable got not applied on SSH access:
```
root@DietPi:~# echo $PATH
/usr/bin:/bin
```
- Adding it to /etc/environment like this `PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games"` didn't help on SSH, okay got it fixed by adding as export to `~/.bashrc` for now.
- PREP runs fine then until trying to install `wput`. Its dependency tail ends at an Debian repo version of `libgnutls30`, but an ubuntu version is installed was stay unable to get overwritten. Also changing priorities and removing all somehow preserving apt settings didn't help:
```
root@DietPi:~# cat /etc/apt/preferences.d/ubuntu
Package: *
Pin: release a=stretch
Pin-Priority: 1100
```
- It is possible to download the related package manually, install via `dpkg -i` and then install `wput` afterwards.
- I found no way to somehow convince APT to overwrite, in case downgrade, all those ubuntu packages.
- Found `/etc/dpkg/origins/default` still showing ubuntu, but even after copying debian in place there, didn't change something about APT behaviour.
- I finally upgraded further to Buster and at least most packages got upgraded then, it seems to be really connected to downgrades. But `--forces-yes`, `--allow-downgrades` and several other options don't help:
```
root@DietPi:~# apt install --reinstall -t buster iputils-ping --allow-downgrades --allow-change-held-packages --allow-remove-essential --force-yes
Reading package lists... Done
Building dependency tree
Reading state information... Done
Reinstallation of iputils-ping is not possible, it cannot be downloaded.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
W: --force-yes is deprecated, use one of the options starting with --allow instead.
```
- Purging (in case via dpkg -P) and reinstalling works for some packages, but for core packages even `dpkg -P` refuses to purge. So I am left with this for now:
```
root@DietPi:~# dpkg -l | grep ubuntu
ii  fuse                       2.9.7-1ubuntu1                    amd64        Filesystem in Userspace
ii  gettext-base               0.19.8.1-4ubuntu1                 amd64        GNU Internationalization utilities for the base system
ii  iputils-ping               3:20161105-1ubuntu2               amd64        Tools to test the reachability of network hosts
ii  libfuse2:amd64             2.9.7-1ubuntu1                    amd64        Filesystem in Userspace (library)
ii  libsasl2-2:amd64           2.1.27~101-g0780600+dfsg-3ubuntu1 amd64        Cyrus SASL - authentication abstraction library
ii  libsasl2-modules-db:amd64  2.1.27~101-g0780600+dfsg-3ubuntu1 amd64        Cyrus SASL - pluggable authentication modules (DB)
ii  libseccomp2:amd64          2.3.1-2.1ubuntu3                  amd64        high level interface to Linux seccomp filter
ii  linux-base                 4.5ubuntu1                        all          Linux image base package
pi  zlib1g:amd64               1:1.2.11.dfsg-0ubuntu2            amd64        compression library - runtime
```
- As sad: Manually downloading these packages from https://packages.debian.org and dpkg -i works, in case dependencies need to be installed before dependants (libfuse2 > fuse).

I was looking for a way to force automated reinstallation/downgrading of every single package to Debian repo, but found nothing that worked here. On the other hand, these remaining ubuntu versions don't seem to hurt. Everything is running fine.
- Finally also found, where Debian is setting PATH and Ubuntu doesn't: `/etc/profile`

So this for me looks like an alternative, e.g. for creating an EspressoBin image, or another one, where just Ubuntu base images are available, especially if we want to avoid ARMbian. Someone with EspressoBin there and in mood to try out with: http://espressobin.net/wp-content/uploads/2017/10/ebin-ubuntu-16.04.3.zip ? 😃
Next I am checking out the procedure to build an image from scratch with just bootloader, kernel, firmware.
_____________________________
Side note: `sed: can't read /DietPi/dietpi/.version: No such file or directory`
- Error handler/globals need to check for existence and otherwise use something like `non-DietPi` or `on-installation` or just the version of PREP script (?) as version string within error message.

LeadProphet, 2018-02-26:
Ubuntu 17 and forward will be very, very different from the latest Debian, because of the move to Wayland. Recommend you try the above with Ubuntu 16.04 LTS, since this will be closer to Debian and because it is more stable/has a longer support term (5 years).

MichaIng, 2018-02-26:
@LeadProphet 
Thanks for the hint. As my image was Ubuntu 17.10. this gives me hope, that with e.g. 16.04. it is even easier.

Hmm haven't noticed something about wayland, maybe that just matters for desktop images? However X.org seems to be still supported/available on Ubuntu, so should be no problem to switch for DietPi compatibility: https://itsfoss.com/switch-xorg-wayland/

The LTS for us is not important, we anyway leave not much left from the Ubuntu system, at best nothing. In the end it's currently Debian Stretch and necessarily the SBC firmware providers support that counts.

LeadProphet, 2018-02-26:
OK, that makes sense; you must have been working off a 17.10 server
release? That wouldn't have anything with Wayland in it, I'd imagine.

That makes sense. Thanks for all your help vis the EspressoBin; it's too
bad they don't support debian directly...

MichaIng, 2018-02-26:
@LeadProphet 
Jep, I used this server image: https://www.ubuntu.com/download/server 17.10.1

MichaIng, 2018-03-18:
Note for myself: https://www.linuxquestions.org/questions/debian-26/using-dist-upgrade-to-upgrade-from-debian-to-ubuntu-583736/

LexiconCode, 2018-05-18:
Why avoid using Armbian?

Fourdee, 2018-05-18:
@LexiconCode 

> Why do you want to avoid using Armbian?

Our projects, goals, and levels of professionalism are not compatible.

To ensure the best for both our projects, I took the decision to separate our projects and remove all ARMbian images. This will not be reverted or changed.

You are welcome to use ARMbian at your own agenda, however, I will not endorse or support it officially with DietPi.

Fourdee, 2018-07-09:
@MichaIng 

I'll mark this as closed, as we are only focused on pure Debian.

However, please reopen if you want to continue work on this. I'am just trying to clean up some old tickets :)

closed 2018-07-09 by Fourdee