MichaIng/DietPi #1558: PCManFM has broken icons

gammaleak, 2018-02-26:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? v6.2
- SBC Device (EG: RPi 3)? RPi3
- Power supply used (EG: 5v 1A RAVpower)? 5V 2.5A
- SD card used (EG: Sandisk ultra)? LoveRPi branded
- Distro (EG: Jessie) | ```uname -a```? Stretch

#### Additional Information (if applicable):
- Software title? N/A
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Expected behaviour:

If you install LXDE, the basic applications that come with it should all have fully functioning icons.

#### Actual behaviour:

PCManFM has broken icons for the "View" toggles (i.e., "Large Icons," "Small Icons," "Detailed List," etc.)

#### Steps to reproduce:

1. Install DietPi with LXDE
2. Launch LXDE
3. Observe broken icons in toolbar

#### Did you submit a ```dietpi-bugreport```?

I didn't submit one.

#### Extra details:

I tried adding a new icon set (Oxygen) from the repository, but that didn't fix the issue. Seems like maybe PCManFM is expecting to find those icons somewhere specific and they aren't there?

MichaIng, 2018-02-26:
@gammaleak  
Thanks for your report, I will try to replicate on VM.

MichaIng, 2018-02-26:
@gammaleak 
I couldn't replicate on Stretch VM. LXDE and PCManFM working and icons looking as they should.

Do you see any error messages within journalctl? Don't know where PCManFM/Openbox would drop their error messages actually, didn't use a desktop for a long time 😅.

In case someone with testing RPi has to try replicate.

Fourdee, 2018-02-27:
RPi 3 Stretch LXDE fresh install:
- 🈴 Confirmed
[image]

Fourdee, 2018-02-27:
```
** (pcmanfm:9863): WARNING **: The directory '/root/Templates' doesn't exist, ignoring it

(pcmanfm:9863): Gtk-CRITICAL **: IA__gtk_menu_shell_insert: assertion 'GTK_IS_MENU_SHELL (menu_shell)' failed
```

```
root@DietPi:~# dpkg -l | grep icon
ii  adwaita-icon-theme             3.22.0-1+deb9u1              all          default icon theme of GNOME
ii  gtk-update-icon-cache          3.22.11-1+rpi3               armhf        icon theme caching utility
ii  hicolor-icon-theme             0.15-1                       all          default fallback theme for FreeDesktop.org icon themes
ii  libtext-iconv-perl             1.7-5+b8                     armhf        converts between character sets in Perl
ii  lxde-icon-theme                0.5.1-2                      all          LXDE standard icon theme
```

Favorites tab missing aswell.

Fourdee, 2018-03-25:
Notes:
- As far as I can tell, the only reason this may be occurring, is due to removal of ```dbus``` in PREP. However, it is installed via APT during LXDE install. Is it possible removing Dbus initially is the cause?
I'll need to confirm/exclude this by doing a PREP without removal of DBUS and retest.

Fourdee, 2018-05-18:
Tests:
- Debian pcmanfm 🈯️ 
```
wget http://ftp.us.debian.org/debian/pool/main/p/pcmanfm/pcmanfm_1.2.5-3_armhf.deb
dpkg -i *.deb
```

RPi version of ```pcmanfm``` is the issue, most likely customized towards the RPi desktop, either way, we'll pull in the Debian version to resolve.

Fourdee, 2018-05-18:
Look and feel is still an issue, hangs on open:
```
root@DietPi:~# dpkg -l | grep lx | grep rpi
ii  libgl1-mesa-glx:armhf          13.0.6-1+rpi2                armhf        free implementation of the OpenGL API -- GLX runtime
ii  lxde                           9+rpi1                       all          Metapackage for LXDE
ii  lxde-core                      9+rpi1                       all          Metapackage for the LXDE core
ii  lxinput                        0.3.5-1+rpi4                 armhf        LXDE keyboard and mouse configuration
ii  lxpanel                        0.9.3-1+rpi6                 armhf        LXDE panel
ii  lxpanel-data                   0.9.3-1+rpi6                 all          LXDE panel (data files)
ii  lxterminal                     0.3.0-2+rpi1                 armhf        LXDE terminal emulator
```

Fourdee, 2018-05-21:
Completed, new ticket for the other issues with lxappearance:
https://github.com/Fourdee/DietPi/issues/1791#issue-325031923

puccaso, 2018-11-02:
hey guys - you know, im having the same issue.
i've documented the issue on raspberry pi forums, https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=225936&p=1387128#p1387128

the PIX icon theme works, but nothing else.

puccaso, 2018-11-02:
[image]
[image]
[image]
Here is my current output.
im on the most recent raspbian - how might i patch the rasbpian code with the Dietpi fix?

Fourdee, 2018-11-02:
@puccaso 

Hi, 

We forgot to hold the APT package during install (fixed in v6.18). Raspbian repo updates will replace the pcmanfm package and broken icons will re-appear.

Workaround, is to reinstall the Debian repo pcmanfm with working icons:
```
wget https://dietpi.com/downloads/binaries/all/pcmanfm_1.2.5-3_armhf.deb
dpkg -i *.deb
```

puccaso, 2018-11-03:
hello @Fourdee , 

thanks for getting back to me, 

yea, that just segments for me,
im looking in the deb package itself for what is different.. 
what exactly has been/needs to be change in the source for the RPI guys, maybe i can follow this up with them..

MichaIng, 2018-11-03:
@puccaso 
This would be great. Always better to solve this dev/maintainer side instead of keeping an unideal workaround.

MichaIng, 2019-10-08:
Same issue present on current current RPi Buster image.

irenegr, 2019-10-10:
Same problem on Raspbian Buster. I found a workaround (?): Pcmanfm->Edit->Preferences->Layout->disable "Display simplified user interface and menus".
Toolbar icons are back again (Papirus icons).

MichaIng, 2020-08-31:
Similar issue with lxpanel from RPi repo where the LXDE menu icon is missing. Very strange thing there, one can open the panel menu settings and the correct icon is pre-selected and exists. One can change padding there, changing it back and forth makes the correct space is taken by the menu, while before only a minimal very small space is reserved, so somehow the icon info is used, but the icon image itself still does not appear. Opening it with image viewer works fine.

I installed `pixflat-icons` icon theme, just to test around some solution (better then using Debian/Raspbian packages and breaking the official RPi desktop by this, if someone wants to install it) and switching to this icon theme restores the icons in PCManFM, but not the menu icon... Messy RPi desktop packages I have to say, I'll go on playing around with it to get it running without the need to manually change settings.

MichaIng, 2020-08-31:
There is a Raspberry Pi UI (desktop) orga with all related repos on GitHub: https://github.com/raspberrypi-ui/
Didn't know that.

MichaIng, 2020-08-31:
@irenegr 
Disabling "`Display simplified user interface and menus`" does not show the toolbar icons, it simply hides the additional icons that lack an image. Those additional buttons provided by this setting allow to change the view of current directory items and open a new PCManFM window. Checking now whether Raspbian/Debian version of PCManFM as well doesn't have these icons + this setting disabled by default or if they have those icons for some reason.

MichaIng, 2020-08-31:
Using Raspbian or Debian package on Buster does not work at all, it just pops up for a microsecond and fails/exits immediately without any error message (nothing when started from console as well), I just tested what was reported here already: https://github.com/MichaIng/DietPi/issues/3570#issuecomment-678048253 _(same with the package from Debian repo)_

MichaIng, 2020-09-11:
Lonely issue about the missing icons on PCManFM: https://github.com/raspberrypi-ui/pcmanfm-buster/issues/2

MichaIng, 2020-09-12:
Sad, zero ambition to make the RPi repo LXDE packages compatible with native LXDE installs (`apt install lxde` instead of `apt install raspberrypi-ui-mods`) and they seem to be so much different from upstream versions now that merging the reasonable changes will never happen. So we have to accept the fact that all raspberrypi-ui packages are working with complete RPi default desktop only. I'll try to replicate some of the issues with native RPi desktop now (the blurry icons are almost certainly an issue there as well, simply no recognised due to missing comparison) to have any chance for attention. As workaround at least we know the packages that we need to pin for now.

MichaIng, 2020-09-24:
Final request to again separate the RPi desktop from the main repository component: #203
I see this the last chance to have a reliable native LXDE experience on RPi, as maintaining a pin list is at best fragile and while fixing one thing we might run into other issues, e.g. due to changing package dependencies over overlaps between desktop and X server packages.

MichaIng, 2020-09-30:
Another day of testing and debugging and finally I found a solution for hanging `lxappearance` on TigerVNC LXDE sessions. Don't ask me for details why this is required in this specific circumstance, but [apt install dbus-user-session](https://packages.debian.org/en/buster/dbus-user-session) solves the issue.
- Basically **without** this package, when starting an X session, a `dbus-daemon` + `dbus-launch` instance is started with a specific ID. On `lxappearance` start, the exact same `dbus-launch` (same cmd options, same ID) is attempted to be started again, which hangs. Only a single instance is allowed, which is somewhat explained in the package details.
- **With** this package, the `dbus-daemon` instance is started a `systemd --user` child process, the `dbus-launch` is not visible. This seems to allow the concurrent `dbus-launch` instance(s) then.

Joulinar, 2020-10-01:
@MichaIng 
As I'm not sure, will the missing menu item be fixed in 6.33 as well? I know we discussed it on the RPi 64bit issue already and you provided a workaround https://github.com/MichaIng/DietPi/issues/3570#issuecomment-675269253.

I'm asking because we have a new report on our forum https://dietpi.com/phpbb/viewtopic.php?p=27620#p27620

MichaIng, 2020-10-01:
Yes I applied the workaround already:
- https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-software#L3000-L3008
- Changelog: https://github.com/MichaIng/DietPi/blob/dev/CHANGELOG.txt#L27

MichaIng, 2020-10-02:
Okay all done that we can do our side. I'll leave it marked as workaround, as this is what it is, sadly not more.

closed 2020-10-02 by MichaIng