MichaIng/DietPi #1559: Open Media Vault

symbios24, 2018-02-26:
Hello,

I cannot see Open Media Vault in the latest dietpi Stretch on a raspberry pi 3
it should be on the System Stats / Management: but its not there or anywhere else
is it removed?

Thanks

MichaIng, 2018-02-26:
@symbios24 
I am sorry, but with v148 we had to remove OMV support due to too many incompatibilities with DietPi: https://github.com/Fourdee/DietPi/blob/testing/CHANGELOG.txt#L582

You could check out our old code, how we handled it that time: https://github.com/Fourdee/DietPi/blob/dfb9a4ff378b3e6393ae844f1875055e4613ec76/dietpi/dietpi-software
Just search by "OMV" via browser, it's not too many entries.

As you are on Debian Stretch, you need OMV version 4 at least: https://github.com/Fourdee/DietPi/issues/1447#issuecomment-362333926

I guess you will find a way to install it and it works. It might just break other parts/software of DietPi and we can't provide support there.

symbios24, 2018-02-26:
thanks for the quick responce
also for the Raspotify there is not any info how to connect

k-plan, 2018-02-26:
@symbios24 

> also for the Raspotify there is not any info how to connect

Hmm .... 

```
root@RPi-Zero:~# dietpi-software list | grep raspotify
id 167 | =0 | raspotify: spotify connect client | +alsa | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=9368#p9368
```

http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=9368#p9368

- you need a working audio output on your device
- you need a Spotify PREMIUM account ( I have non!)

https://www.spotify.com/connect/

-------------------

Your initial question Open Media Vault #1559 is completely answer.
So issue closed.

closed 2018-02-26 by k-plan