MichaIng/DietPi #1560: services inactive - restart fails

d4f2, 2018-02-26:
#### Required Information:
- DietPi Version | v6.2
- SBC Device     | RPi 3 Model B (armv7l) (index=3)
- Distro         | stretch (index=4)
- Command        | ncc maintenance:mode --on
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Services

#### Expected behaviour:
services to start after reboot

#### Actual behaviour:
services are not running after reboot

# dietpi-services status
[  OK  ] Root access verified.

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[ INFO ] DietPi-Services | cron inactive (dead)
[ INFO ] DietPi-Services | lighttpd     inactive (dead)
[ INFO ] DietPi-Services | php7.0-fpm   inactive (dead)
[ INFO ] DietPi-Services | mysql        inactive (dead)
[ INFO ] DietPi-Services | minidlna     inactive (dead)
[ INFO ] DietPi-Services | pihole-FTL   inactive (dead)
[ INFO ] DietPi-Services | redis-server inactive (dead)
[  OK  ] DietPi-Services | dnsmasq      active (running) since Mon 2018-02-26 14:28:58 CET; 18s ago
[  OK  ] DietPi-Services | openvpn      active (exited) since Mon 2018-02-26 14:28:58 CET; 19s ago

attempt to restart results in this error asking me to open a ticket

#### Steps to reproduce:
installed nextcloud and updated to v13
installed pi-hole

#### Additional logs:
```
Log file contents:
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(151): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\Driver\PDOMySql\Driver), Object(Doctrine\DBAL\Configuration), Object(Doctrine\CommonentManager))
#6 /var/www/nextcloud/lib/private/DB/ConnectionFactory.php(152): Doctrine\DBAL\DriverManager::getConnection(Array, Object(Doctrine\DBAL\Configuration), Object(Doctrine\CommonentManager))
#7 /var/www/nextcloud/lib/private/Server.php(618): OC\DB\ConnectionFactory->getConnection('mysql', Array)
#8 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#9 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OCP\IDBConnecti...')
#10 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\IDBConnecti...')
#11 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164): OC\ServerContainer->query('OCP\IDBConnecti...')
#12 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#13 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('DatabaseConnect...')
#14 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('DatabaseConnect...')
#15 /var/www/nextcloud/lib/private/Server.php(1490): OC\ServerContainer->query('DatabaseConnect...')
#16 /var/www/nextcloud/lib/private/Server.php(331): OC\Server->getDatabaseConnection()
#17 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#18 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OC\Authenticati...')
#19 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\Authenticati...')
#20 /var/www/nextcloud/lib/private/Server.php(335): OC\ServerContainer->query('OC\Authenticati...')
#21 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#22 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OC\Authenticati...')
#23 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\Authenticati...')
#24 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164): OC\ServerContainer->query('OC\Authenticati...')
#25 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#26 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OC\Authenticati...')
#27 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\Authenticati...')
#28 /var/www/nextcloud/lib/private/Server.php(351): OC\ServerContainer->query('OC\Authenticati...')
#29 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#30 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OCP\IUserSessio...')
#31 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\IUserSessio...')
#32 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164): OC\ServerContainer->query('OCP\IUserSessio...')
#33 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#34 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('UserSession')
#35 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('UserSession')
#36 /var/www/nextcloud/lib/private/Server.php(1359): OC\ServerContainer->query('UserSession')
#37 /var/www/nextcloud/lib/private/Server.php(678): OC\Server->getUserSession()
#38 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#39 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OC\App\AppManag...')
#40 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\App\AppManag...')
#41 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164): OC\ServerContainer->query('OC\App\AppManag...')
#42 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#43 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('AppManager')
#44 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('AppManager')
#45 /var/www/nextcloud/lib/private/Server.php(1663): OC\ServerContainer->query('AppManager')
#46 /var/www/nextcloud/lib/private/legacy/app.php(330): OC\Server->getAppManager()
#47 /var/www/nextcloud/lib/private/legacy/app.php(113): OC_App::getEnabledApps()
#48 /var/www/nextcloud/lib/base.php(661): OC_App::loadApps(Array)
#49 /var/www/nextcloud/lib/base.php(1080): OC::init()
#50 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#51 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#52 {main}
```

--------------------------------------------------------------------

[FAILED] DietPi-Services | Unable to continue, the program will now terminate.

d4f2, 2018-02-26:
issue just disappeared after repeated reboot / restart attempts
no changes 
?

MichaIng, 2018-02-26:
@d4f2 
Thanks for your report. Means the issue is resolved for you, now?

As nearly all services were found dead, it looks like a deeper system issue. A look into `systemctl` could reveal errors, e.g. where systemd starts.
Otherwise check logs of related services from within `/var/log/`.

I just tested Nextcloud + PiHole working fine (including web ui) on Stretch VM.
I see RealMedia and OpenVPN running besides at your system. Also tested this on VM, which works generally fine:
```
 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] DietPi-Services | cron active (running) since Mon 2018-02-26 19:08:36 CET; 34s ago
[  OK  ] DietPi-Services | lighttpd     active (running) since Mon 2018-02-26 19:08:36 CET; 34s ago
[  OK  ] DietPi-Services | php7.0-fpm   active (running) since Mon 2018-02-26 19:08:37 CET; 33s ago
[  OK  ] DietPi-Services | mysql        active (running) since Mon 2018-02-26 19:08:38 CET; 32s ago
[  OK  ] DietPi-Services | minidlna     active (running) since Mon 2018-02-26 19:08:38 CET; 32s ago
[  OK  ] DietPi-Services | pihole-FTL   active (running) since Mon 2018-02-26 19:08:38 CET; 32s ago
[  OK  ] DietPi-Services | redis-server active (running) since Mon 2018-02-26 19:08:38 CET; 32s ago
[  OK  ] DietPi-Services | dnsmasq      active (running) since Mon 2018-02-26 19:08:36 CET; 34s ago
[  OK  ] DietPi-Services | openvpn      active (exited) since Mon 2018-02-26 19:08:35 CET; 35s ago
```
But RPi is a different APT repository.

If error still occurs on your system, besides error logs (above), you could try to purge and reinstall the software 1 by 1 and see on which the issue occurs.
Somebody with testing ready RPi can try to replicate, sadly mine is production server 😅.

d4f2, 2018-02-27:
Hello, unfortunatelly it is not resolved

> checked systemctl

# systemctl --failed
  UNIT           LOAD   ACTIVE SUB    DESCRIPTION

 dhcpcd.service loaded failed failed dhcpcd on all interfaces

# systemctl status dhcpcd.service
dhcpcd.service - dhcpcd on all interfaces
   Loaded: loaded (/lib/systemd/system/dhcpcd.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/dhcpcd.service.d
           └─wait.conf
   Active: failed (Result: exit-code) since Mon 2018-02-26 15:02:17 CET; 17h ago

Feb 26 15:02:17 DietPi64G systemd[1]: Starting dhcpcd on all interfaces...
Feb 26 15:02:17 DietPi64G dhcpcd[293]: Not running dhcpcd because /etc/network/interfaces
Feb 26 15:02:17 DietPi64G dhcpcd[293]: defines some interfaces that will use a
Feb 26 15:02:17 DietPi64G dhcpcd[293]: DHCP client or static address
Feb 26 15:02:17 DietPi64G systemd[1]: dhcpcd.service: Control process exited, code=exited status=6
Feb 26 15:02:17 DietPi64G systemd[1]: Failed to start dhcpcd on all interfaces.
Feb 26 15:02:17 DietPi64G systemd[1]: dhcpcd.service: Unit entered failed state.
Feb 26 15:02:17 DietPi64G systemd[1]: dhcpcd.service: Failed with result 'exit-code'.
#

I started all over again from scratch with a clean install "DietPi_v6.2_RPi-ARMv6-Stretch.img"
**and it is the pivpn installation causing this**
from previous installations I noticed the setup part AFTER the key generation, where the publicIP or DNS is set is missing during the installation process!
is dnsmasq installation missing?

# cat  /etc/pivpn/install.log
	:::
	::: Stopping OpenVPN service... done.
	:::
	::: Installing scripts to /opt/pivpn... done.
	::: Using protocol: udp
	::: Building CA...

	::: CA Complete.

	Note: using Easy-RSA configuration from: ./vars

	Note: using Easy-RSA configuration from: ./vars

	DH parameters of size 2048 created at /etc/openvpn/easy-rsa/pki/dh.pem

	Note: using Easy-RSA configuration from: ./vars

	An updated CRL has been created.
	CRL file: /etc/openvpn/easy-rsa/pki/crl.pem

	net.ipv4.ip_for
	#

Here I get stuck due to my limited know how
thank you very much for your help

MichaIng, 2018-02-27:
@d4f2 
Thanks for your retry, I will investigate our installation script and recheck on VM, until someone with RPi find time to reproduce.

pedrom34, 2018-02-28:
Dunno if related, but I found that every services where crashed except Kodi this morning.  
After updating to 6.2, and a reboot, everything is back to normal... except sickrage. And trying to launch it manually doesn't work.

`[FAILED] DietPi-Services | sickrage failed (Result: exit-code) since Wed 2018-02-28 07:27:07 CET; 17min ago
● sickrage.service - SickRage
Loaded: loaded (/etc/systemd/system/sickrage.service; disabled; vendor preset: enabled)
Active: failed (Result: exit-code) since Wed 2018-02-28 07:27:07 CET; 17min ago
Process: 1815 ExecStart=/usr/bin/python /etc/sickrage/SickBeard.py -q --daemon --nolaunc h --datadir=/mnt/dietpi_userdata/sickrage (code=exited, status=1/FAILURE)

Feb 28 07:27:06 DietPi systemd[1]: Starting SickRage...
Feb 28 07:27:07 DietPi python[1815]: Traceback (most recent call last):
Feb 28 07:27:07 DietPi python[1815]: File "/etc/sickrage/SickBeard.py", line 71, in <module>
Feb 28 07:27:07 DietPi python[1815]: from sickbeard.tv import TVShow
Feb 28 07:27:07 DietPi python[1815]: ImportError: cannot import name TVShow
Feb 28 07:27:07 DietPi systemd[1]: sickrage.service: Control process exited, code=exited status=1
Feb 28 07:27:07 DietPi systemd[1]: Failed to start SickRage.
Feb 28 07:27:07 DietPi systemd[1]: sickrage.service: Unit entered failed state.
Feb 28 07:27:07 DietPi systemd[1]: sickrage.service: Failed with result 'exit-code'.`
  
Reboot doesn't help. Raspberry pi 3.  
  
If my comment isn't related, please delete it.

pedrom34, 2018-02-28:
On another reboot mysql failed to start as well.

> [FAILED] DietPi-Services | mysql failed (Result: exit-code) since Wed 2018-02-28 07:53:34 CET; 39s ago
● mysql.service - LSB: Start and stop the mysql database server daemon
Loaded: loaded (/etc/init.d/mysql; generated; vendor preset: enabled)
Active: failed (Result: exit-code) since Wed 2018-02-28 07:53:34 CET; 39s ago
Docs: man:systemd-sysv-generator(8)
Process: 1305 ExecStart=/etc/init.d/mysql start (code=exited, status=1/FAILURE)

Feb 28 07:52:56 DietPi mysqld[1509]: 2018-02-28 7:52:56 1988493312 [Note] Plugin 'FEEDBACK' is disabled.
Feb 28 07:52:56 DietPi mysqld[1509]: 2018-02-28 7:52:56 1988493312 [ERROR] Can't init tc log
Feb 28 07:52:56 DietPi mysqld[1509]: 2018-02-28 7:52:56 1988493312 [ERROR] Aborting
Feb 28 07:52:56 DietPi mysqld[1509]:
Feb 28 07:53:08 DietPi mysqld_safe[1784]: mysqld from pid file /var/run/mysqld/mysqld.pid ended
Feb 28 07:53:34 DietPi mysql[1305]: Starting MariaDB database server: mysqld ............failed!
Feb 28 07:53:34 DietPi systemd[1]: mysql.service: Control process exited, code=exited stat us=1
Feb 28 07:53:34 DietPi systemd[1]: Failed to start LSB: Start and stop the mysql database server daemon.
Feb 28 07:53:34 DietPi systemd[1]: mysql.service: Unit entered failed state.
Feb 28 07:53:34 DietPi systemd[1]: mysql.service: Failed with result 'exit-code'.

MichaIng, 2018-02-28:
@pedrom34 
I don't think it is related, but let's see if I can help. Several services crashing over night can have several reasons, often filled memory when e.g. daily cron tasks add to memory usage. I had that once with some rsync backup and mysql dump and sadly that time without swap file on my RPi2.
But also sdcard corruption is possible.

The crashed MariaDB then often leads to file system, database and or log corruption, preventing it from regular startup. I also found especially the database quite sensitive here.

About the specific `Can't init tc log` I mostly find the quick solution to remove `/var/lib/mysql/tc.log`. In case it doesn't work, better move it:
`mv /var/lib/mysql/tc.log /var/lib/mysql/tc.log.bak`

If MariaDB starts up again, it cannot be wrong to do a database corruption check. I personally do this after every MariaDB crash or unexpected exit:
`mysqlcheck --all-databases` in case
`mysqlcheck --auto-repair --all-databases`

If you don't have a swapfile, please add one with at least 1G via `dietpi-config`>`Advanced`. In case you have much software/services active, a higher value could be necessary. Check usage via `htop`, especially on high load (access to your servers services).

Finally you can force a file system check on reboot:
`touch /forcefsck` or to do on every reboot
add `fsck.mode=force` to your cmdline: `nano /boot/cmdline.txt`

d4f2, 2018-02-28:
Sadly I  can reproduce my pivpn installation problem over and over again
tried several approaches but it comes down to:
- use install image 6.2
- change ethernet network settings to static ip
- complete installation
- reboot
- install pivpn
- reboot 
> dhcpcd.service loaded failed failed dhcpcd on all interfaces - see above

(might want to change the subject of the issue to point at the pivpn installation causing it?)

	:::
	::: You are root.
	::: Verifying free disk space...
	:::
	::: Checking apt-get for upgraded packages.... done!
	:::
	::: Your system is up to date! Continuing with PiVPN installation...
	:::
	::: Setting IP to 192.168.200.100.  You may need to restart after the install is complete.
	:::
	::: Using User: pivpn
	:::
	::: Checking for existing base files...
	:::    Checking /etc/.pivpn is a repo...:::    Cloning https://github.com/pivpn/pivpn.git into /etc/.pivpn... done!
	:::
	::: Stopping OpenVPN service... done.
	:::
	::: Installing scripts to /opt/pivpn... done.
	::: Using protocol: udp
	::: Building CA...
	Generating a 2048 bit RSA private key
	........................+++
	..+++
	writing new private key to '/etc/openvpn/easy-rsa/pki/private/ca.key.BXam5r5pmt'
	-----

	::: CA Complete.

	Note: using Easy-RSA configuration from: ./vars
	rand: Use -help for summary.
	Generating a 2048 bit RSA private key
	.....................+++
	.......................+++
	writing new private key to '/etc/openvpn/easy-rsa/pki/private/server_'
	-----
	Using configuration from /etc/openvpn/easy-rsa/openssl-1.0.cnf
	Can't open /etc/openvpn/easy-rsa/pki/index.txt.attr for reading, No such file or directory
	1995662752:error:02001002:system library:fopen:No such file or directory:../crypto/bio/bss_file.c:74:fopen('/etc/openvpn/easy-rsa/pki/index.txt.attr','r')
	1995662752:error:2006D080:BIO routines:BIO_new_file:no such file:../crypto/bio/bss_file.c:81:
	Check that the request matches the signature
	Signature ok
	The Subject's Distinguished Name is as follows

	Certificate is to be certified until Feb 26 12:02:07 2028 GMT (3650 days)

	Write out database with 1 new entries
	Data Base Updated

	Note: using Easy-RSA configuration from: ./vars
	Generating DH parameters, 2048 bit long safe prime, generator 2
	This is going to take a long time
	............................

	DH parameters of size 2048 created at /etc/openvpn/easy-rsa/pki/dh.pem

	Note: using Easy-RSA configuration from: ./vars
	Using configuration from /etc/openvpn/easy-rsa/openssl-1.0.cnf

	An updated CRL has been created.
	CRL file: /etc/openvpn/easy-rsa/pki/crl.pem

	net.ipv4.ip_forward = 1
	run-parts: executing /usr/share/netfilter-persistent/plugins.d/15-ip4tables save
	run-parts: executing /usr/share/netfilter-persistent/plugins.d/25-ip6tables save
	ip6tables-save v1.6.0: Cannot initialize: Address family not supported by protocol

	run-parts: /usr/share/netfilter-persistent/plugins.d/25-ip6tables exited with return code 1
	::: Install Complete...
	::: Restarting services...
	Synchronizing state of openvpn.service with SysV service script with /lib/systemd/systemd-sysv-install.
	Executing: /lib/systemd/systemd-sysv-install enable openvpn
	::: done.
	:::
	::: Installation Complete!
	::: Now run 'pivpn add' to create an ovpn profile for each of your devices.
	::: Run 'pivpn help' to see what else you can do!
	::: It is strongly recommended you reboot after installation.
	:::
	::: The install log is located at: /etc/pivpn/install.log
	[ SUB1 ] DietPi-Services > stop
	[  OK  ] DietPi-Services | stop : cron

	 DietPi-Software
	─────────────────────────────────────────────────────
	 Mode: Optimize and configure software

	[ INFO ] DietPi-Software | Applying DietPi optimizations and configurations for RPi 3 Model B (armv7l), please wait...

	[ SUB1 ] DietPi-Services > dietpi_controlled
	[  OK  ] DietPi-Services | dietpi_controlled : cron

	 DietPi-Software
	─────────────────────────────────────────────────────
	 Mode: Installation completed

	[  OK  ] DietPi-Software | The system will now reboot.
	 This completes the DietPi-Software installation.

pedrom34, 2018-03-01:
@MichaIng I deleted my comments as it's unrelated.
Thanks for the help, but I managed to fix everything by uninstalling and re-installing MariDB and Sckrg.

Thanks!

Fourdee, 2018-03-01:
@MichaIng @d4f2 
In regards to PiHole + OpenVPN.
This is a known issue as PiHole + OpenVPN require additional manual configuration at this time.
REF: https://github.com/Fourdee/DietPi/issues/1245.

MichaIng, 2018-03-01:
@Fourdee @d4f2 
Okay seems to be 2 mixed things here now:
1. PiHole not being compatible with PiVPN/OpenVPN without manual configuration, possibly causing the service breaks?
2. PiVPN leading to error messages.

I tried again the following, as before I installed (just) OpenVPN:
1. Switch to static IP.
2. Install PiVPN
- Works well, didn't really test a VPN connection, but openvpn starts without errors and `pivpn` command works.
- Indeed I also see `dhcpcd.service` failing, but this makes sense as we disabled DHCP to use PiVPN. On my preconfigured/but not really loaded WLAN interface it still shows up and seems that dhcpcd notices this, showing error, but this should not hurt in any way.
- Maybe we should just `systemctl disable dhcpcd` on PiVPN installation, as it is anyway unused then/doesn't work, avoiding the systemd messages about it? Maybe we should already disable it when switching to STATIC on all enabled+connected adapters and re-enable when switching to DHCP?

@d4f2 
- Do you actually face any issues with (just) PiVPN installed or is it just the error messages/warnings on boot?
- If it's just the messages, do `systemctl disable dhcpcd` to avoid them, but note that currently you need to `systemctl enable dhcpcd` again then, if you at any time want to switch to DHCP network again.
- Please try to install everything besides PiHole then and see if it works, `dietpi-services status` showing all services up as expected.
- For PiHole, follow this guide to configure it properly in combination with PiVPN/OpenVPN: https://itchy.nl/raspberry-pi-3-with-openvpn-pihole-dnscrypt
- Feel free to add your voice on the feature request for auto configuration in DietPi: https://github.com/Fourdee/DietPi/issues/1245

d4f2, 2018-03-02:
thank you vey much for all your efforts!
to point this out - this has got __nothing to do with PiHole !__
- this a __PiVpn installation problem__ only
the installation breaks off after key generation and does not continue to configure the server thus adding a client `pivpn -a` fails
    
      cat: Default.txt: No such file or directory
    
looking at: https://github.com/pivpn/pivpn/blob/master/auto_install/install.sh

`confOpenVPN` - appears to finish fine - the server.conf is created
but the __`netfilter-persistent save`__  (PLAT=Raspbian) in `confNetwork` brakes off with:
    
    run-parts: executing /usr/share/netfilter-persistent/plugins.d/15-ip4tables save
    run-parts: executing /usr/share/netfilter-persistent/plugins.d/25-ip6tables save
    ip6tables-save v1.6.0: Cannot initialize: Address family not supported by protocol

    run-parts: /usr/share/netfilter-persistent/plugins.d/25-ip6tables exited with return code 1
    
https://github.com/pivpn/pivpn/blob/25aaf24c893175894d0a54d159a10fbb74b6b463/auto_install/install.sh#L987
subsequent functions `confOVPN` / `setClientDNS` ..etc.. are missing

hope this details my problem

d4f2, 2018-03-02:
after these hours trying to get to the bottom of this I  finally found...
https://github.com/pivpn/pivpn/issues/464
enabled IPV6 and installed again...
PiVPN installation problem solved !

though still get the:

		root@DietPi:~# systemctl --failed
		  UNIT           LOAD   ACTIVE SUB    DESCRIPTION

		● dhcpcd.service loaded failed failed dhcpcd on all interfaces

		LOAD   = Reflects whether the unit definition was properly loaded.
		ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
		SUB    = The low-level unit activation state, values depend on unit type.

		1 loaded units listed. Pass --all to see loaded but inactive units, too.
		To show all installed unit files use 'systemctl list-unit-files'.
		root@DietPi:~# systemctl status dhcpcd.service
		● dhcpcd.service - dhcpcd on all interfaces
		   Loaded: loaded (/lib/systemd/system/dhcpcd.service; enabled; vendor preset: enabled)
		  Drop-In: /etc/systemd/system/dhcpcd.service.d
				   └─wait.conf
		   Active: failed (Result: exit-code) since Fri 2018-03-02 09:29:31 GMT; 1min 14s ago
		  Process: 298 ExecStart=/usr/lib/dhcpcd5/dhcpcd -q -w (code=exited, status=6)

		Mar 02 09:29:31 DietPi systemd[1]: Starting dhcpcd on all interfaces...
		Mar 02 09:29:31 DietPi dhcpcd[298]: Not running dhcpcd because /etc/network/interfaces
		Mar 02 09:29:31 DietPi dhcpcd[298]: defines some interfaces that will use a
		Mar 02 09:29:31 DietPi dhcpcd[298]: DHCP client or static address
		Mar 02 09:29:31 DietPi systemd[1]: dhcpcd.service: Control process exited, code=exited status=6
		Mar 02 09:29:31 DietPi systemd[1]: Failed to start dhcpcd on all interfaces.
		Mar 02 09:29:31 DietPi systemd[1]: dhcpcd.service: Unit entered failed state.
		Mar 02 09:29:31 DietPi systemd[1]: dhcpcd.service: Failed with result 'exit-code'.
		Warning: dhcpcd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
		root@DietPi:~#

which I will negelect as advised

I need a break now .. will continue later with my remaining setup nextcloud etc...
will follow up in case the restart of services issue was not related to this and re-occurs
nevertheless all this point me in the required direction and did help me
thank you

MichaIng, 2018-03-02:
@d4f2 
May I ask how you enabled IPv6? By default we do not disable it, but you can toggle within dietpi-config. But as you run RPi, I recognized some default alias blocking within `/etc/modprobe.d/ipv6` which is installed by default on Raspbian images (where DietPi for RPi is based on). This had no visible effect on my RPi, IPv6 module is still active and used, but was this file the issue for you?

About your dhcpcd error, as said, as PiVPN+OpenVPN needs DHCP to be disabled, just do:
```
systemctl stop dhcpcd
systemctl disable dhcpcd
```
and you are free or error messages.

d4f2, 2018-03-02:
@MichaIng 
confirm I disabled IPv6 in dietpi-confg when setting the static IP
I want back and started again from scratch 
(too much messing with confgurations when during my attempts)
 IPv6 is indeed enabled by default

thank you very much
just for completness: in the old jessie version none of these issues and dhcpd active togehter with PiVPN+OpenVPN installed

MichaIng, 2018-03-02:
@d4f2 
Okay thanks for testing. Maybe the older versions of openvpn handled absence of IPv6 different, without throwing errors. Actually I am wondering if it is possible to use it just with IPv4 as well by pre-configuring/choosing different option within PiVPN install dialogs. Otherwise we could simple force enabling of IPv6 during our installation script and inform user about that. I will have a look into this.

k-plan, 2018-03-02:
@MichaIng 

> Maybe the older versions of openvpn handled absence of IPv6 different

PiVPN installer handle this different in the paste. 
Can't say this exactly, because in dislike and use OpenVPN.

> Starting officially in the 2.3.0 release, OpenVPN supports IPv6 inside the tunnel, and can optionally be configured with IPv6 as a transport protocol for the tunneled data.

https://community.openvpn.net/openvpn/wiki/IPv6

```
root@DietPi:~# cat /DietPi/dietpi/.version
122
```

```
root@DietPi:~# cat /etc/debian_version
8.9

```

```
root@DietPi:~# openvpn --version
OpenVPN 2.3.4 aarch64-unknown-linux-gnu [SSL (OpenSSL)] [LZO] [EPOLL] [PKCS11] [MH] [IPv6] built on Jun 26 2017
```

MichaIng, 2018-03-02:
@k-plan @Fourdee  
Haha, I just found some more things, that got mixed up:
- Why actually `dhcpcd5` is installed and active, at least on Debian images, by default, where we use `ics-dhcp-client` by default? On my Buster Raspbian RPi there at least is no dhcpcd5 installed, seemed to be autoremoved there.
- These are two different DHCP clients, `dhcpcd5` more passiv, without active background process and `ics-dhcp-client` with `dhclient` as active background process, that enables some monitoring features, as far as I understood.
- I can't find a reason why `dhcpcd5` is not autoremoved, since no other package depends on it and it is not essential nor "required". But somehow on my Stretch VM and seems as well d4f2's Stretch RPi it is installed and even with enabled service, besides and doubling dhclient (isc-dhcp-client).
- Actually as `dhcpcd` seems to be more lightweight and passive (htop shows process as well, but with 0.0% memory use). Is there some reason, why we have `isc-dhcp-client` as default (and somehow doubled) with DietPi? In this case we should purge `dhcpcd5` on all systems to avoid interference and error messages as above.

About all this DHCP server stuff:
- All this sounds like, it would be beneficial, if we could resolve this for users. Actually it sounds easiest, if we by default disable DHCP server functionality for dnsmasq and openvpn by default and inform users, if we find multiple DHCP capable software titles? By times we could add further configuration scripts, enabling DHCP within dnsmasq or openvpn, disabling it for the other.
- We could also perform just a DHCP server check with local DHCP client and decide, based on that, if we activate DHCP server functionality for the related software title.
- Affected are PiHole, OpenVPN and PiVPN, which is based as well on OpenVPN, or is there something else?

About IPv6+PiVPN:
- For now we can handle it a bid similar than: https://github.com/Fourdee/DietPi/issues/316
- Check for enabled IPv6, otherwise inform users, that IPv6 need to be enabled and block installation.

k-plan, 2018-03-03:
@MichaIng 

> if we by default disable DHCP server functionality for dnsmasq and openvpn by default and inform users, if we find multiple DHCP capable software titles?

Vielleicht habe ich mich nicht deutlich genug ausgedrückt.
- bei `dnsmasq` von Pi-Hole ist die DHCP Server Funktion per default deaktiviert, bei jeder Installation. Da muss erst der Nutzer etwas aktivieren und konfigurieren bevor dieser läuft.
- `openvpn` Server ist kein klassischer DHCP Server. Der Dienst`openvpn` Server nutzt nur das DHCP Protokoll um IP Adressen den `openvpn` Clients zuzuweisen. Und das auch nur auf den TUN Interface.

Eigentlich kommt sich das nie in die Quere, auch nicht im LAN mit DHCP vom Router.

Selbst wenn man dann noch den HotSpot aktiviert, funktioniert das. `isc-dhcp-server` läuft  dann nur auf dem WLAN Interface.

Es sei denn, man fummelt daran herum und weiß nicht was man da tut.

k-plan, 2018-03-03:
@MichaIng 

-------------

> Haha, I just found some more things, that got mixed up:

yes, nice finding. Noticed this as well.

RPi v6:

```
root@RPi-Zero:~# dpkg -l | grep dhc
ii  dhcpcd5                         1:6.11.5-1+rpt4              armhf        DHCPv4, IPv6RA and DHCPv6 client with IPv4LL support
ii  isc-dhcp-client                 4.3.5-3                      armhf        DHCP client for automatically obtaining an IP address
```

Orange-Pi-One v6 (test-build):

```
root@DietPi-OrangePi-One:~# dpkg -l | grep dhc
ii  isc-dhcp-client           4.3.5-3                        armhf        DHCP client for automatically obtaining an IP address
```

Odroid-C1 v6:

```
root@Odroid-C1-v6:~# dpkg -l | grep dhc
ii  isc-dhcp-client             4.3.1-6+deb8u2                   armhf        DHCP client for automatically obtaining an IP address
ii  isc-dhcp-common             4.3.1-6+deb8u2                   armhf        common files used by all of the isc-dhcp packages
```

-----------

---- ONLY for reference ----

OLD ARMbian based build

```
root@NanoPi-Neo:~# dpkg -l | grep dhc
ii  dhcpcd5                        6.0.5-2                          armhf        DHCPv4, IPv6RA and DHCPv6 client with IPv4LL support
ii  isc-dhcp-client                4.3.1-6+deb8u2                   armhf        DHCP client for automatically obtaining an IP address
ii  isc-dhcp-common                4.3.1-6+deb8u2                   armhf        common files used by all of the isc-dhcp packages
```

OLD Odroid based build

```
root@Odroid-C2:~# dpkg -l | grep dhc
ii  isc-dhcp-client                4.3.1-6+deb8u2                   arm64        DHCP client for automatically obtaining an IP address
ii  isc-dhcp-common                4.3.1-6+deb8u2                   arm64        common files used by all of the isc-dhcp packages
```

k-plan, 2018-03-03:
@MichaIng

> Check for enabled IPv6, otherwise inform users, that IPv6 need to be enabled and block installation.

Wie schon gesagt, ich sehe nur zwei Optionen:
- entweder implementiert man alle IPv6 Konfigurationsmöglichkeiten in `dietpi-config` vollständig.

- oder man lässt IPv6 laufen, da Dual Stack Support heute Standard des Betriebssystems und entfernt die Möglichkeit diesen zu deaktivieren.

Die erste Lösung wird aufwendig, da will im Moment keiner ran.
Außerdem können die vielfältigen Einstellmöglichkeiten unbedarfte Nutzer leicht überfordern und zu weiteren Fehlern führen. Über kurz oder lang wird aber kein Weg daran vorbei führen.

Die zweite Option vermeidet Fehlkonfigurationen und Software Inkompatibilitäten. Man hat aber nicht mehr vollständig im Griff, was läuft und wohin Verbindungen aufgebaut werden, wenn man ein Dual Stack LAN mit Dual Stack ISP Internetzugang oder IPv6 Tunnel Provider hat.

MichaIng, 2018-03-03:
Okay I will close this, as TOs issue is solved by enabling IPv6. I open new issues about the two other topics.

closed 2018-03-03 by MichaIng