MichaIng/DietPi #1561: [Question] - Set two different daily times on cron

DanielSSilva, 2018-02-26:
Hello there! As the title says, I would like to know if it's possible to set two different "daily" cron jobs. I have some lights/screen that I want to be turned on by certain time and then off at certain time. I've the scripts working from the terminal, but I want to set it to run daily at specific times

Thank you in advance!

MichaIng, 2018-02-26:
`crontab -u root -e`
There enter something like:
`* 18 * * * /path/to/script <option to turn on lights>`
which would lead to lights turned on at 18:00.

DanielSSilva, 2018-02-26:
I forgot to mention that the script must be executed as root, more specifically this script `sudo nano examples/strandtest.py`
taken from [here](https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/)

MichaIng, 2018-02-26:
@DanielSSilva 
> I forgot to mention that the script must be executed as root

Therefore I used crontab -u **root** -e 😉, `crontab -u www-data -e` would enable to add scripts, that need/wanted to start by www-data (e.g. necessary to run ownCloud/Nextcloud occ command).

So then your line to add would be:
`0 18 * * * /full/path/to/example/strandtest.py`

DanielSSilva, 2018-02-26:
right... that's what i've also tried but with no luck. Trying to debug the cause but without luck. I'll set the logs to full

MichaIng, 2018-02-26:
@DanielSSilva 
Ah sorry, the command is at least `python /full/path/to/examples/strandtest.py` or even `PYTHONPATH="/full/path/to/lib.linux-armv7l-2.7" python /full/path/to/examples/strandtest.py` or something like this according to your link.

TheAntimist, 2018-02-27:
Another option is to utilize [systemd/timers](https://wiki.archlinux.org/index.php/Systemd/Timers) ([man page](https://www.freedesktop.org/software/systemd/man/systemd.timer.html)). I have used them for something similar before.

MichaIng, 2018-02-28:
Jep, systemd timers/services allow much more flexibility and you can add certain conditions and more. Needs just a bid more reading in order to configure them properly.

MichaIng, 2018-03-13:
@DanielSSilva 
Were you able to get your python script running via cron?

DanielSSilva, 2018-03-14:
Yes I did :) There was also a misunderstanding on my side because i've set it as 1 * * * * * and thought it would run every minute, instead it runs only at the first minute each hour... 
It has been fixed. Thank you :D

Fourdee, 2018-03-14:
Marking as resolved. Please reopen if required.

closed 2018-03-14 by Fourdee