MichaIng/DietPi #1563: HTPC Manager Further info

symbios24, 2018-02-27:
Hello is there somewhere a list with all the compatible programs/software that can be used with HTPC Manager on the raspberry pi?

There is no further info about in here: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=110#p8043

Thanks

k-plan, 2018-02-27:
@symbios24 

Sorry, if this sound harsh now, first google search entry for "HTPC Manager + software"

http://htpc.io/#software

symbios24, 2018-02-27:
i did that and cant find an accurate list

k-plan, 2018-02-27:
... and what please shall we do?

symbios24, 2018-02-27:
i did not say that i asked if you know.thanks

symbios24, 2018-02-27:
its ok i will do a deeper search and give you the list if i find anything so you can put it on the info

symbios24, 2018-02-27:
ok found it, this are all the software for it you can put it on the info page:

couchpotato
deluge
kodi
nzbget
plex server
plexpy
radarr
sabnzdb
squeezebox
sickrage
sonarr
transmission
qbittorrent
rtorrent

only thing is first i install the softwares and then i install HTPS Manager? or First HTPS Manager and then the Softwares?

MichaIng, 2018-02-28:
@symbios24 
Thanks for this. I never had a closer look what HTPC Manager actually is 😆. It makes sense, that we add this list, at least as an example, which software UIs can be merged with HTPC Manager.

I will try it out later, sounds interesting, if it is working reliably.

symbios24, 2018-02-28:
hi i dont know only if first i install the software/s and then i install HTPS Manager? or First HTPS Manager and then the Software/s??

symbios24, 2018-02-28:
Hi, you have to setup everything manual this is the problem is it possible to put in the install software two options one to install only HTPC manager and second to install with all supported software?
i will be glad to help any way i can.

Thanks

symbios24, 2018-02-28:
Hi, you have to setup everything manual from within HTPC Manager, is it possible to put in the install software two options one to install only HTPC manager and second to install with all supported software and automatically configured to work out of the box?
i will be glad to help any way i can.

Thanks

symbios24, 2018-02-28:
Hi, you have to setup everything manual from within HTPC Manager, is it possible to put in the install software two options one to install only HTPC manager and second to install with all supported software and automatically configured to work out of the box?
i will be glad to help any way i can.

and another thing if i install  HTPC Manager  and then the software eg: sonar,transmission etc.
does  HTPC Manager automatically configure the installed software to work with it? or do i need to do it manual?

Thanks

MichaIng, 2018-02-28:
@symbios24 
I think it doesn't make sense to install all supported software automatically, as not everyone needs all of them, some even double (qbittorrent+rtorrent). But what we could try is, to add automatic configuration of already installed supported software, within HTPC manager installation process. If we place those additional scripts into separate files, outside of `dietpi-software` script, we could even call them on installation of supported software.

But this will cost some time. I will have a look and try out HTPC manager the next days.

symbios24, 2018-03-01:
ok it will be great the automation, thanks

MichaIng, 2026-03-29:
I am closing this. Before implementing HTPC Manager more deeply, someone would need to maintain HTPC Manager itself. When gmiranda stopped maintaining it (alone), we created an orga which we are both owner of. But honestly, I do not find the time needed to keep the project up alone either, so it staled: https://github.com/HTPC-Manager/HTPC-Manager

Such a project really is difficult to maintain, as each module needs to be maintained. This is difficult when not using them oneself, or backed a larger active community that reports upstream changes and helps keeping the modules functional. Otherwise they break over time, which is what happened. I do already have a project where I maintain lots of software and hardware implementations that I do not use myself: DietPi 😅

Of course anyone is invited to join and try reviving the project.