MichaIng/DietPi #1564: RPi Cam Control / MotionEye Problem

symbios24, 2018-02-27:
Hello, i Installed both softwares RPi Cam Control - RPi Camera / Web Interface Surveillance
- MotionEye - Camera / Web Interface Surveillance on latest dietpi Stretch in a RaspberryPi3 but upon first boot after installation the dietpiOS keeps rebooting by it self,anyone know whats wrong?

Fourdee, 2018-02-27:
@symbios24 

Hi,

Thanks for the report 👍 

- 🈯️ RPi cam is fixed in v6.3: https://github.com/Fourdee/DietPi/issues/1512

--- 

Motion notes:
- 🈴 Missing ```build-essential```, fails to build. ```zlib1g-dev``` is installed
```
 The headers or library files could not be found for zlib,
    a required dependency when compiling Pillow from source.
```

🈯️ fixed for v6.3

Once ```build-essential``` is installed, installs fine and unable to replicate automatic reboot.

I did experience reboots previously, due to the sunny connector being 50% disconnected. Please ensure the sunny connector and ribbon cable is fully connected.
[image]

I'll mark this as closed, unable to replicate auto reboot. If problems persist after sunny connector re-seat and v6.3, please reopen.

closed 2018-02-27 by Fourdee

symbios24, 2018-02-27:
But 6.3 is not released yet..

Fourdee, 2018-02-28:
@symbios24 

> But 6.3 is not released yet..

Wont be long, eta mid-end next week.