MichaIng/DietPi #1565: No WiFi mschapv2-Support

Gokujo, 2018-02-28:
#### Required Information:
- DietPi Version: 6.2
- SBC Device: RPi 3
- Distro: Linux DietPi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
I'm trying to connect to PEAP MSCHAPV2 WiFi network but I can't. I tried some manuals but nothing helped.
This ist my /etc/wpa_supplicant/wpa_supplicant.conf:
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=GB
network={
	ssid="WLAN_DATA1"
	proto=RSN
	key_mgmt=WPA-EAP
	pairwise=CCMP
	auth_alg=OPEN
	eap=PEAP
	identity="user"
	password="pass"
	phase2="auth=MSCHAPV2"
}
```

#### Expected behaviour:
It should connect to WiFi

#### Actual behaviour:
I get an error that authentification failed. My data are right, i tried them on other devices.

#### Steps to reproduce:
I followed this [instruction](https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=111100)

#### Did you submit a ```dietpi-bugreport```?
No

Fourdee, 2018-02-28:
@Gokujo 

Hi,

- ```WPA-EAP``` support was added to dietpi-automation as of v6.3 (and requires a v6.3+ image, TBA).
- ```dietpi-config``` support is planned for the future (no ETA yet): https://github.com/Fourdee/DietPi/issues/1549

> It should connect to WiFi

From what I can tell, the config looks fine, should be functional.

Any info from wpa_supplicant service?
```
systemctl status wpa_supplicant.service -l
```

Gokujo, 2018-03-01:
@Fourdee 
fresh installed dietpi on rpi3
i tried #1549 and [this](http://dietpi.com/phpbb/viewtopic.php?p=706#p706)
```
root@DietPi:~# ifdown wlan0
Killed old client process
Internet Systems Consortium DHCP Client 4.3.5
Copyright 2004-2016 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/b8:27:eb:5c:59:78
Sending on   LPF/wlan0/b8:27:eb:5c:59:78
Sending on   Socket/fallback
root@DietPi:~# /etc/init.d/networking reload
[....] Reloading network interfaces configuration...Killed old client process
Internet Systems Consortium DHCP Client 4.3.5
Copyright 2004-2016 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/eth0/b8:27:eb:09:0c:2d
Sending on   LPF/eth0/b8:27:eb:09:0c:2d
Sending on   Socket/fallback
DHCPRELEASE on eth0 to 192.168.0.10 port 67
Internet Systems Consortium DHCP Client 4.3.5
Copyright 2004-2016 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/eth0/b8:27:eb:09:0c:2d
Sending on   LPF/eth0/b8:27:eb:09:0c:2d
Sending on   Socket/fallback
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 4
DHCPREQUEST of 192.168.13.22 on eth0 to 255.255.255.255 port 67
DHCPOFFER of 192.168.13.22 from 192.168.13.1
DHCPACK of 192.168.13.22 from 192.168.13.1
bound to 192.168.13.22 -- renewal in 526253 seconds.
done.
root@DietPi:~# ifup wlan0
Internet Systems Consortium DHCP Client 4.3.5
Copyright 2004-2016 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/b8:27:eb:5c:59:78
Sending on   LPF/wlan0/b8:27:eb:5c:59:78
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 2
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
root@DietPi:~#
```

`systemctl status wpa_supplicant.service -l` says:
```
root@DietPi:~# systemctl status wpa_supplicant.service -l
● wpa_supplicant.service - WPA supplicant
   Loaded: loaded (/lib/systemd/system/wpa_supplicant.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-03-01 09:36:50 CET; 5min ago
 Main PID: 5568 (wpa_supplicant)
   CGroup: /system.slice/wpa_supplicant.service
           └─5568 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant

Mär 01 09:36:50 DietPi systemd[1]: Starting WPA supplicant...
Mär 01 09:36:50 DietPi wpa_supplicant[5568]: Successfully initialized wpa_supplicant
Mär 01 09:36:50 DietPi systemd[1]: Started WPA supplicant.
```
but no results

Gokujo, 2018-03-19:
@Fourdee is this feature now integrated in v6.4?

Fourdee, 2018-03-19:
@Gokujo 

Still pending full support: https://github.com/Fourdee/DietPi/issues/1549

Currently, basic support only via PREP and use of dietpi.txt automation.

Outside of that, manual system entries/configuration are required.

Fourdee, 2018-07-09:
Marking as closed in favor of: https://github.com/Fourdee/DietPi/issues/1549

closed 2018-07-09 by Fourdee