MichaIng/DietPi #1566: Sonarr and Radarr Problem

roelfoubert, 2018-02-28:
When I want to install sonarr and radarr on the latest version of dietpi, The installation completed successfully. The problem is that They won't boot and can't get started manually.

symbios24, 2018-02-28:
Hi, di you access the web interface of sonarr for example from another computer connected to the same network thought your internet browser url = http://192.168.0.100:8989 ??

roelfoubert, 2018-02-28:
Yes, but before I updated to the latest version everything worked. So I did a clean install of dietpi and installed sonarr but it wont start. Same problem for radarr. when I look at the service status, I get this as feedback:
● sonarr.service - Sonarr (NzbDrone) Daemon
   Loaded: loaded (/etc/systemd/system/sonarr.service; disabled; vendor preset: 
enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2018-02-28 15
:07:54 GMT; 2s ago
  Process: 3769 ExecStart=/usr/bin/mono --debug /opt/NzbDrone/NzbDrone.exe -nobr
owser (code=exited, status=1/FAILURE)
 Main PID: 3769 (code=exited, status=1/FAILURE)

Feb 28 15:07:54 RaspberryPi systemd[1]: sonarr.service: Main process exi
ted, code=exited, status=1/FAILURE
Feb 28 15:07:54 RaspberryPi systemd[1]: sonarr.service: Unit entered fai
led state.
Feb 28 15:07:54 RaspberryPi systemd[1]: sonarr.service: Failed with resu
lt 'exit-code'.

Fourdee, 2018-02-28:
🈴 RPi 3 test ```dietpi-software install 144 145``` :
- 🈴 Mono Fails install due to ```/tmp tmpfs``` being full. Filled with mono.
```G_AGI: mono-devel mediainfo sqlite3 libmono-cil-dev```

roelfoubert, 2018-02-28:
How can I fix this?

Fourdee, 2018-02-28:
Ok, so this is an issue with Mono version used (```5.10``` latest via http://download.mono-project.com/repo/debian):
- Radarr is pending update to work with latest version of Mono: https://github.com/Radarr/Radarr/issues/2585#issue-300794900
- Sonarr fix is already done, confirmed using develop repo install.

Regardless of above, i'll set Sonarr to develop repo install by default (in line with Radarr).

symbios24, 2018-03-01:
Hello i installed also both radarr and sonarr in latest DietPi stretch in a raspberryPi 3
Sonarr web interface loads fine but Radarr web interface is not loading can this be fixed manual?

Thanks

symbios24, 2018-03-01:
Hello i installed also both radarr and sonarr in latest DietPi stretch in a raspberryPi 3
Sonarr web interface loads fine but Radarr web interface is not loading and as i see on boot the Radarr service is running, can this be fixed manual?

Thanks

roelfoubert, 2018-03-01:
I used this link to fix my problem and Radarr is working again.
[https://forums.sonarr.tv/t/mono-5-10-update-broke-sonarr/17800/19](https://forums.sonarr.tv/t/mono-5-10-update-broke-sonarr/17800/19)

symbios24, 2018-03-01:
Hi, But the fix is for Sonarr not Radarr 
Will this fix Radarr?

roelfoubert, 2018-03-01:
yes just change /opt/NzbDrone/ to /opt/Radarr/

samsepiol59, 2018-03-01:
@roelfoubert I did (Sonarr is working, Radarr is not) but nothing happened, I'm still unable to access radarr at :7878!

symbios24, 2018-03-01:
samsepiol59 you installed both on latest DietPi?

Fourdee, 2018-03-01:
Notes:
Ok, waiting for https://api.github.com/repos/Radarr/Radarr/releases to be updated with 0.2.0.984> version. Used in our scrape to obtain the latest version available.

Once they do this (assuming the same sonarr fixed is applied), the issue of failing to run will no longer occur.

Fourdee, 2018-03-04:
Radarr have updated their binary, resolves outstanding issues with latest mono library:
- 🈯️ Sonarr
- 🈯️ Radarr

Patched re installs for v6.3, ETA release March 7th. Completed.

closed 2018-03-04 by Fourdee