MichaIng/DietPi #1568: Nextcloud Error

dirtyharrywk, 2018-03-01:
## Creating a bug report/issue:
Nextcloud will not start after clean flash.

#### Required Information:
- DietPi Version: 6.2
- SBC Device: Odroid HC1
- Power supply used: 5v 4A
- SD card used: Sandisk ultra class 10
- Distro: Linux DietPi 3.10.106+ #1 SMP PREEMPT Fri Nov 24 16:26:38 CET 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?  Yes, tried it on another Sandisk Ultra

#### Expected behaviour:
Nextcloud to work.  open Chrome browser and enter http://<my_ip>/nextcloud and login page should open.
#### Actual behaviour:
HTTP ERROR 500

#### Steps to reproduce:
open Chrome browser and enter http://<my_ip>/nextcloud
HTTP ERROR 500 error appears

#### Did you submit a ```dietpi-bugreport```? Yes
8e5c3a47-14ad-41cc-ba7a-0faa27cacf39-0

#### Extra details:

[image]

MichaIng, 2018-03-01:
@dirtyharrywk 
Thanks for reporting and bugreport submission. I will investigate.
If you have any chance to copy&paste the whole installation process from terminal, that would be very helpful.

@Fourdee 
Could you give me access to the bugreport, send via email e.g.? Need to find out what went wrong with most likely MariaDB and/or PHP config/startup.

Fourdee, 2018-03-01:
@dirtyharrywk 

> 8e5c3a47-14ad-41cc-ba7a-0faa27cacf39-0

Apologies, unable to find this bug report, please can you resend?
```
dietpi-bugreport
```

MichaIng, 2018-03-17:
@dirtyharrywk 
Is your issue still present?

dirtyharrywk, 2018-03-17:
No.  I started over using something other then Diet Pi.  I may try again using a new micro as card. Thank you!

k-plan, 2018-03-17:
> No. I started over using something other then Diet Pi.

Closed.

However, please reopen if required.
Thanks.

closed 2018-03-17 by k-plan