MichaIng/DietPi #1569: TP-LINK Archer T2UH

skane165, 2018-03-01:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 6.2
- SBC Device RPi 3
- Power supply used 5v 3A RAVpower
- SD card used Sandisk ultra
- Distro (EG: Jessie) | Linux DietPi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

#### Expected behaviour:
I am trying to install TP-LINK Archer T2UH i am using install-wifi 
after installing the correct driver it do not come on i can see it in lsusb
and dmesg

#### Actual behaviour:
After reboot there where some files utilities and some other that where holding back by the os

#### Steps to reproduce:
sudo wget http://www.fars-robotics.net/install-wifi -O /usr/bin/install-wifi
sudo chmod +x /usr/bin/install-wifi
sudo install-wifi
reboot

#### Did you submit a ```dietpi-bugreport```?
40d7ac93-8ccf-49a6-a3a9-7816e6bf9157-0 

#### Extra details:

Fourdee, 2018-03-01:
@skane165 

Hi,

The driver may be available in the latest kernel. Simply run the following commands to update:
```
G_RPI_UPDATE
reboot
```

Failing the above, please provide the output of ```lsusb```

k-plan, 2018-03-01:
> Failing the above, please provide the output of` lsusb`

should be like:

`Bus 001 Device 004: ID 148f:761a Ralink Technology, Corp.`

> Product name: TP-LINK Archer T2UH 
> Chipset: MediaTek MT7610U
> Vendor ID: 0x148f
> Product ID: 0x761a

https://wikidevi.com/wiki/TP-LINK_Archer_T2UH

But with a Additional chips
`5GHz WLAN Front-End Module (FEM);Skyworks;SKY85703-11;SKY11, 85703, 520VQ;1;`

> Notes: With kernel 4.4.41 and driver mt7610u_sta only 2.4GHz works

skane165, 2018-03-02:
It wont load past hostapd now and gets stucked fresh install?

skane165, 2018-03-02:
I started it with my old wifi dongle and this is the output after i changed it

root@DietPi:~# lsusb
Bus 001 Device 008: ID 148f:761a Ralink Technology, Corp. 
Bus 001 Device 005: ID 1a2c:4c5e China Resource Semico Co., Ltd 
Bus 001 Device 006: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

skane165, 2018-03-02:
I made a fresh install and this is the lsusb
root@DietPi:~# lsusb
Bus 001 Device 007: ID 148f:761a Ralink Technology, Corp. 
Bus 001 Device 005: ID 1a2c:4c5e China Resource Semico Co., Ltd 
Bus 001 Device 006: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

install-wifi 

 *** Raspberry Pi wifi driver installer by MrEngman.
 *** Performing self-update
 *** Relaunching after update

 *** Raspberry Pi wifi driver installer by MrEngman.

Your current kernel revision = 4.14.22-v7+
Your current kernel build    = #1096

Checking for a wifi module to determine the driver to install.

Your wifi module is Bus 001 Device 007: ID 148f:761a Ralink Technology, Corp. 

And it uses the mt7610 driver.

Checking for a mt7610 wifi driver module for your current kernel.
There is a driver module available for this kernel revision.
Downloading the mt7610 driver.
Installing the mt7610 driver.

Installing driver module mt7610u.ko.
Loading and running the mt7610 driver, mt7610u.ko.

skane165, 2018-03-05:
Hello user have replied 4 days ago =P

Fourdee, 2018-03-06:
@skane165 

> Hello user have replied 4 days ago =P

Hi, we are busy with the v6.3 release for tomorrow.

I lack the device to test the driver install. 
When I'am able, i'll check the driver install code and see if anything crops up that may offer a solution.

skane165, 2018-03-08:
Oki np =)

closed 2018-04-13 by Fourdee