MichaIng/DietPi #1570: Regarding desktop view.

ronakmd, 2018-03-01:
## Creating a bug report/issue:

DietPi Version | cat /DietPi/dietpi/.version?>>DietPi_v6.0_RPi-ARMv6-Stretch

SBC Device (EG: RPi 3)?>>>>Rpi MOdel B (armv71) index=3

Power supply used (EG: 5v 1A RAVpower)?>>>From Adapter 5v 1.5A

SD card used (EG: Sandisk ultra)?>>>Samsung 16
#### Additional Information (if applicable):
Hi i want to boot my raspberry pi into desktop environment.
Exactly same as Raspbian Strech and without the compulsory login process.
Can you please suggest on which desktop environment to be used?
I want openCv libraries,python and its usual libraries.
Also i want a fast boot up say 5 to 8 seconds if possible.
Please let me know.
Thankyou.

k-plan, 2018-03-01:
> Rpi MOdel B
> raspberry pi into desktop environment.
> Also i want a fast boot up say 5 to 8 seconds if possible.

With a sd card?
Sorry, simple impossible.
If  `Raspbian Strech` can do this, stay on it.

k-plan, 2018-03-12:
No reply so far, i'll mark as closed.
However, please reopen if required.
Thanks.

closed 2018-03-12 by k-plan