MichaIng/DietPi #1574: Swap File Disabled

symbios24, 2018-03-02:
Hello, i enabled swap file from DietPi-Config/advanced options/swap file size and put it manual to 100mb and rebooted
Writing the command free -m i see that the swap file is not enabled its 0mb
How can it be enabled?

Thanks

symbios24, 2018-03-02:
i change it to 500mb and i see that is taking the disk space,so my question is
is it loading only when needed?

Fourdee, 2018-03-02:
@symbios24 

Thanks for the report, please paste results:
```
cat /etc/dphys-swapfile
systemctl status dphys-swapfile.service
free -m
which dphys-swapfile
```

Fourdee, 2018-03-02:
Reference: https://github.com/Fourdee/DietPi/issues/1572#issuecomment-369917050

symbios24, 2018-03-02:
CONF_SWAPFILESIZE=512
CONF_SWAPFILE=/var/swap
dphys-swapfile.service  - LSB Autogenerate and use a swap file
Loaded: loaded (/etc/ini
/sbin/dphys-swapfile

symbios24, 2018-03-02:
i have the picture here from the results

https://ibb.co/h7RU2n

Fourdee, 2018-03-02:
@symbios24 

Swapon failed due to IO error.

Please try with and capture output:
```
/DietPi/dietpi/func/dietpi-set_dphys-swapfile 512 /var/swap
```

@MichaIng Your idea to add global error handling is a good one.

symbios24, 2018-03-03:
Here is the capture

https://image.ibb.co/cYbmF7/2018_03_02_22_02_55.jpg

Fourdee, 2018-03-03:
@symbios24 

- Which DietPi image did you use to install this system? eg: DietPi_v6.2_RPi_xxxx.img
- Did you modify for Berryboot?
- SD card used?
- Power Supply used?

Try:
```
rm /var/swap
/DietPi/dietpi/func/dietpi-set_dphys-swapfile 512 /mnt/swap
```

Failing the above, it may be an idea to try a fresh image write.

symbios24, 2018-03-03:
i use the latest 6.2 
it is a berryboot image
8GB Toshiba class 10 UHS-1 micro sd 
2amp supply

symbios24, 2018-03-03:
the command 
rm /var/swap
/DietPi/dietpi/func/dietpi-set_dphys-swapfile 512 /mnt/swap

does not enable the swap i get the same error as in the photo: https://image.ibb.co/cYbmF7/2018_03_02_22_02_55.jpg

what is the problem the berryboot image?

i did a fresh install on this image

Fourdee, 2018-03-03:
@symbios24 

> what is the problem the berryboot image?

No sure, but searching the net shows this is an issue limited to BerryBoot.

We do not officially provide support for BerryBoot images of DietPi (we do not create them), however, our user community has been successful in creating them for testing:
https://github.com/Fourdee/DietPi/issues/1545

Please continue this discussion in the above ticket. I'll mark this as closed.

closed 2018-03-03 by Fourdee

symbios24, 2018-03-04:
sorry i did not realize about berryboot since its the same OS.
Do you know with software/s use/need swapfile??