MichaIng/DietPi #1575: Aria2 configuration cleared after reboot

roelfoubert, 2018-03-02:
I installed Aria2 but when I change the settings in the WebUI, they get back to default after reboot. 
Is there a solution for this? I can't find the Aria2.conf file.

Fourdee, 2018-03-04:
@roelfoubert 

Thanks for the report, i'll try to replicate.

Fourdee, 2018-03-04:
Test:
- ```max-connection-per-server > 3 (1)```
- Save changes
- reboot
- Reverts to 1

Confirmed. I'll take a look.

Fourdee, 2018-03-04:
Looks like a limitation with aria2-webui:
https://github.com/ziahamza/webui-aria2/issues/164#issuecomment-151669830
https://github.com/ziahamza/webui-aria2/issues/36#issuecomment-11328963

Fourdee, 2018-03-04:
@roelfoubert 

Workaround will be available once DietPi v6.3 is released, by editing the ```/var/lib/dietpi/dietpi-software/installed/aria2.conf```: https://github.com/Fourdee/DietPi/issues/1575#issue-301808769

ETA few days (mid next week) for DietPi v6.3.

Completed.

closed 2018-03-04 by Fourdee

Joulinar, 2022-06-06:
Aria2 configuration file should be located on `/mnt/dietpi_userdata/aria2/aria2.conf`

bt61, 2022-06-06:
> Aria2 configuration file should be located on `/mnt/dietpi_userdata/aria2/aria2.conf`

ah, of course, thank you