MichaIng/DietPi #1577: (rpi3) qBittorrent Bug  Gui

tnnuyri, 2018-03-02:
good afternoon ! my qBittorrent spent 4 days following work without shutting down with 37 torrents stopped working the gui part, in the torrents they seem to be sowing I simply can not get in ... raspberry pi 3

Fourdee, 2018-03-02:
@tnnuyri 

Hi,

Some issues here:
- You have disabled the swapfile. Not recommended and may cause "out of memory" errors + further issues due to that. Especially with large amounts of active torrent files.
- You have installed 2 BitTorrent clients. Ideally, you should only ever install 1 BitTorrent client (either Deluge or QBitTorrent)

Try a system reboot, see if the web interface returns.

tnnuyri, 2018-03-02:
Yes ! Whenever I reboot it is active again, I would like to know why this happens .. As for swapfile I did not disable it, I did the standard install of dietpi without changing these things I just connected an external disk by usb... tks

symbios24, 2018-03-03:
you wont make it in activating swap

Fourdee, 2018-03-03:
@tnnuyri 

> As for swapfile I did not disable it, 

Ah, this is standard as of v6.0+. Which DietPi version are you running?

You can remove unneeded software in ```dietpi-software``` > Uninstall

tnnuyri, 2018-03-05:
I'm using version v6.2 !! I think the swapfile is shut down because I use the disk in NTFS format, I'll try to put it in ext4, just wanted to know if the error was out of memory, I have other mini pcs like tinker board, odroid c2, odroid xu4

Fourdee, 2018-03-06:
@tnnuyri 

> Do you think the error is caused by the lack of memory?

I believe so.

Fourdee, 2018-03-14:
Marking as closed. Please reopen if required.

closed 2018-03-14 by Fourdee