MichaIng/DietPi #1580: NTP error

juliocoliveira, 2018-03-03:
## Creating a bug report/issue:
Using username "root".

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
[  OK  ] Root access verified.
 ───────────────────────────────────────
 DietPi     | 08:49 | Sat 03/03/18
 ───────────────────────────────────────
 v6.2       | RPi 3 Model B (armv7l)
 ───────────────────────────────────────
 IP eth0    | 192.168.0.3
 ───────────────────────────────────────
[ INFO ] DietPi-Drive_Manager | Checking RootFS R/W access. Please wait...
[  OK  ] DietPi-Drive_Manager | RootFS R/W access.

[ INFO ] DietPi-Software | Initializing database, please wait...
[ INFO ] DietPi-Software | Reading database, please wait...
 ───────────────────────────────────────
 DietPi     | 08:49 | Sat 03/03/18
 ───────────────────────────────────────
 v6.2       | RPi 3 Model B (armv7l)
 ───────────────────────────────────────
 IP eth0    | 192.168.0.3
 ───────────────────────────────────────

 Welcome to DietPi-Software

Failed to create bus connection: No such file or directory
[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of systemd-timesyncd (1/30)
Failed to create bus connection: No such file or directory
[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of systemd-timesyncd (2/30)
Failed to create bus connection: No such file or directory
[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of systemd-timesyncd (3/30)
Failed to create bus connection: No such file or directory
[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of systemd-timesyncd (4/30)
Failed to create bus connection: No such file or directory
[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of systemd-timesyncd (5/30)

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? 6.2
- SBC Device (EG: RPi 3)? RPi3
- Power supply used (EG: 5v 1A RAVpower)? 5v
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```? Stretch

ichbestimmtnicht, 2018-03-03:
I experienced the same problem and waiting for about 2-3 min solved the problem for the first boot. After that i got this error some times moore before i figured out to adjust the timeserver in dietpi-config and it worked fine.

Still fix this bug for first time installation pls.

juliocoliveira, 2018-03-03:
I ran the command but: -bash: dietpi-config: command not found

Fourdee, 2018-03-03:
@juliocoliveira 

> I ran the command but: -bash: dietpi-config: command not found

```
/DietPi/dietpi/dietpi-config
```
Networking Options: NAS/MISC > APT mirror

Once completed, to resume setup
```
/DietPi/dietpi/dietpi-software
```

---

notes:
- Change prompt on failure to include ```dietpi-config``` as an option to change NTP mirror?
- Dbus?
```
Failed to create bus connection: No such file or directory
```

juliocoliveira, 2018-03-03:
Changed the mirror but still the Dbus error

Fourdee, 2018-03-03:
@juliocoliveira 

> Changed the mirror but still the Dbus error

Thanks, we'll check that. 

NTPD Mirror now functional?

juliocoliveira, 2018-03-03:
you mean the NTP mirror selector? it removed the ntp package and emptied the /etc/ntp.conf
And tried to install and configure the NTP manually and got this:
```
root@DietPi:~# ntpd -q -g
 3 Mar 11:15:36 ntpd[6709]: ntpd 4.2.8p10@1.3728-o Sat Oct  7 14:29:08 UTC 2017 (1): Starting
 3 Mar 11:15:36 ntpd[6709]: Command line: ntpd -q -g
 3 Mar 11:15:36 ntpd[6709]: proto: precision = 1.458 usec (-19)
restrict default: KOD does nothing without LIMITED.
 3 Mar 11:15:36 ntpd[6709]: restrict default: KOD does nothing without LIMITED.
restrict ::: KOD does nothing without LIMITED.
 3 Mar 11:15:36 ntpd[6709]: restrict ::: KOD does nothing without LIMITED.
 3 Mar 11:15:36 ntpd[6709]: format error frequency file /etc/ntp.drift
 3 Mar 11:15:36 ntpd[6709]: unable to bind to wildcard address :: - another process may be running - EXITING
```

Fourdee, 2018-03-03:
@juliocoliveira 

> you mean the NTP mirror selector? 

Yep.

> it removed the ntp package and emptied the /etc/ntp.conf

Upon selecting the following option, ntp is installed:
[image]

Only when selecting the following "Advanced options" > "Time sync mode", will NTP be removed:
[image]

Please can you verify "Time sync mode" current setting.

juliocoliveira, 2018-03-03:
i can't fix it, i disabled the ntp.

Fourdee, 2018-03-03:
@juliocoliveira 

> i can't fix it, i disabled the ntp.

Ok 👍 

We'll improve the error handling for v6.3, give the user more options to resolve NTPD issues on their network when it occurs:
[image]

k-plan, 2018-03-03:
@Fourdee 

> Override  :  (NOT RECOMMENDED) ....

hmm .. what does thies mean?

`date 030317492018` ?

BTW: 
Don't believe, most user will know difference between `NTPD` and `Systemd-timesyncd`.
Could cause more confusion when gain.

MichaIng, 2018-03-03:
@Fourdee 
I just tested to switch to timesyncd as well:
- Failed as well with `Failed to create bus connection: No such file or directory` after every of the 30 attempts.
- I found the service disabled and inactive, we just use `timedatectl set-ntp true` when choosing mode 4 via dietpi-config, there also on manual execution the error occurs: https://github.com/Fourdee/DietPi/blob/testing/dietpi/func/dietpi-set_software#L238-L257
- I did:
```
root@VM-Stretch:~# systemctl enable systemd-timesyncd
Created symlink /etc/systemd/system/sysinit.target.wants/systemd-timesyncd.service → /lib/systemd/system/systemd-timesyncd.service.
root@VM-Stretch:~# systemctl start systemd-timesyncd
root@VM-Stretch:~# systemctl status systemd-timesyncd
● systemd-timesyncd.service - Network Time Synchronization
   Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; enabled; vendor preset: enabled)
  Drop-In: /lib/systemd/system/systemd-timesyncd.service.d
           └─disable-with-time-daemon.conf
   Active: inactive (dead)
Condition: start condition failed at Sat 2018-03-03 19:59:51 CET; 10s ago
           └─ ConditionFileIsExecutable=!/usr/sbin/ntpd was not met
     Docs: man:systemd-timesyncd.service(8)
```
- As you can see, the service as execution condition that ntp is not installed.
```
root@VM-Stretch:~# apt purge ntp
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libbsd0 libedit2 libopts25
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  ntp*
0 upgraded, 0 newly installed, 1 to remove and 1 not upgraded.
After this operation, 1,847 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 18264 files and directories currently installed.)
Removing ntp (1:4.2.8p10+dfsg-3+deb9u1) ...
(Reading database ... 18212 files and directories currently installed.)
Purging configuration files for ntp (1:4.2.8p10+dfsg-3+deb9u1) ...
Processing triggers for systemd (232-25+deb9u1) ...
root@VM-Stretch:~# systemctl start systemd-timesyncd
root@VM-Stretch:~# systemctl status systemd-timesyncd
● systemd-timesyncd.service - Network Time Synchronization
   Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; enabled; vendor preset: enabled)
  Drop-In: /lib/systemd/system/systemd-timesyncd.service.d
           └─disable-with-time-daemon.conf
   Active: active (running) since Sat 2018-03-03 20:01:04 CET; 2s ago
     Docs: man:systemd-timesyncd.service(8)
 Main PID: 3844 (systemd-timesyn)
   Status: "Synchronized to time server 185.207.104.70:123 (0.de.pool.ntp.org)."
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/systemd-timesyncd.service
           └─3844 /lib/systemd/systemd-timesyncd

Mar 03 20:01:04 VM-Stretch systemd[1]: Starting Network Time Synchronization...
Mar 03 20:01:04 VM-Stretch systemd[1]: Started Network Time Synchronization.
Mar 03 20:01:04 VM-Stretch systemd-timesyncd[3844]: Synchronized to time server 185.207.104.70:123 (0.de.pool.ntp.org).
```

So procedure on activating mode 4:
- `dietpi-software uninstall 106`
- add timesyncd config
- _`systemctl daemon-reload` should not be necessary, as we adjust timesyncd config and not it's service file._
- `systemctl enable systemd-timesyncd`
- `systemctl start systemd-timesyncd`

I see now uninstallation should have been done actually, but as you can see it somehow was not: https://github.com/Fourdee/DietPi/blob/testing/dietpi/func/dietpi-set_software#L222-L236

I see also we use `timedatectl status --no-pager` which creates the same "bus missing" error message. What works instead is `systemctl status systemd-timesyncd | grep 'Synchronized to time server'` e.g.

Jep confirmed `timedatectl` need `dbus`, but `systemctl status/is-active systemd-timesyncd` can be used instead to verify status.

Fourdee, 2018-03-03:
@k-plan 

> hmm .. what does thies mean? Hard reset as synced and disable forever?
This should really be the last chance ...

Yep, last resort. Basically allows the system to run with DietPi scripts, and, without updating the time.

> Can it be a option, user can set date once manually via whiptail dialog?
date 030317492018 ?

Yep, we could replace ```override``` with this option.

@MichaIng 

> So procedure on activating mode 4:
dietpi-software uninstall 106
add timesyncd config
systemctl daemon-reload should not be necessary, as we adjust timesyncd config and not it's service file.
systemctl enable systemd-timesyncd
systemctl start systemd-timesyncd
I see now uninstallation should have been done actually, but as you can see it somehow was not: https://github.com/Fourdee/DietPi/blob/testing/dietpi/func/dietpi-set_software#L222-L236

Nice 👍 I'll take a look.

Fourdee, 2018-03-05:
@MichaIng 

Had to change the detection of completion of timesync:
```
systemctl status systemd-timesyncd -l | grep -ci -m1 'Synchronized to time server'
```

The downside is, if the log entries truncate in systemd, it may fail. 

We need a better way of detecting its completion. as ```timedatectl status``` cannot be used without ```dbus```.

We may need to install ```dbus``` along side ```systemd-timesyncd```

Fourdee, 2018-03-05:
Setting high, to prevent me getting side tracked with other tickets :)

MichaIng, 2018-03-05:
@Fourdee 

> We may need to install `dbus` along side `systemd-timesyncd`

Jep thought the same. Actually everything _could_ be managed without `dbus`:
- `timedatectl set-ntp true` does nothing more than:
`systemctl enable systemd-timesyncd && systemctl start systemd-timesyncd`
- `timedatectl status` can be replaced as you mentioned.
You could use `systemctl status systemd-timesyncd | grep -q 'Status: "Synchronized to time server'` which will high likely not be truncated as it is not part of log but left aligned service info. But service must be enabled and started for this to show up.

But if already `systemd-timesyncd` is used, then users might expect `timedatectl` to work, which does not without `dbus` and would lead to bug reports. Thus jep: **We should install `dbus` along with NTP mode 4 (timesyncd).**

Fourdee, 2018-03-05:
@MichaIng 

> We should install dbus along with NTP mode 4 (timesyncd).

Agree 👍 

> To reduce mess with NTP, we could find a reliable way to use timesyncd as oneshot as well to replace NTP completely.

Yep, would be nice to remove NTP, however, I remember a while back systemd-timesync took nearly 2/3X longer to sync than NTP in my tests. May be a configuration issue (the old one), but yep, I'll look at removing NTP and switching to systemd-timesync if its viable.

Fourdee, 2018-03-06:
@MichaIng 

Ok, will leave NTP in for now (due to https://github.com/Fourdee/DietPi/issues/1580#issuecomment-370569005), i did make all the changes in a new branch to test:
https://github.com/Fourdee/DietPi/commit/29e7f53443ed0337301582cf31626dc19d7aafa9

closed 2018-03-06 by Fourdee

ShinobiWPS, 2019-01-17:
Worth mentioning softwares like OpenMediaVault depends on NTP, No NTP, no OMV

ShinobiWPS, 2019-01-17:
I fightinhg right now to keep OMV & NTP, but it fails because it waits for timesyncd completion..frustrating.

MichaIng, 2019-01-17:
@ShinobiWPS 
No big problem, just set `dietpi-config > Advanced Options > Time sync mode > 0 Custom` and DietPi will work flawlessly with NTP. It does not check successful time sync state though, but expects NTP to do it's job successfully.