MichaIng/DietPi #1582: Subsonic 6 broken link

sjdaniel10, 2018-03-04:
DietPi-Software: Connection test:
                                                                            │ https://sourceforge.net/projects/subsonic/files/subsonic/6.1.3/subsonic-6.1.3.deb
                                                                            │ /download
                                                                            │  - exit_code = 8
                                                                            │  - VERSION:v6.2 | HW_MODEL:2 | HW_ARCH:2 | DISTRO:4
                                                                            │
                                                                            │ Log file contents:
                                                                            │ Spider mode enabled. Check if remote file exists.
                                                                            │ --2018-03-04 12:42:46--
                                                                            │ https://sourceforge.net/projects/subsonic/files/subsonic/6.1.3/subsonic-6.1.3.deb
                                                                            │ /download
                                                                            │ Resolving sourceforge.net (sourceforge.net)... 216.105.38.13
                                                                            │ Connecting to sourceforge.net (sourceforge.net)|216.105.38.13|:443... connected.
                                                                            │ HTTP request sent, awaiting response... 302 Found
[image]
Location:
                                                                            │ https://downloads.sourceforge.net/project/subsonic/subsonic/6.1.3/subsonic-6.1.3.
                                                                            │ deb?r=&ts=1520167367 [following]
                                                                            │ Spider mode enabled. Check if remote file exists.
                                                                            │ --2018-03-04 12:42:47--
                                                                            │ https://downloads.sourceforge.net/project/subsonic/subsonic/6.1.3/subsonic-6.1.3.
                                                                            │ deb?r=&ts=1520167367
                                                                            │ Resolving downloads.sourceforge.net (downloads.sourceforge.net)... 216.105.38.13
                                                                            │ Connecting to downloads.sourceforge.net
                                                                            │ (downloads.sourceforge.net)|216.105.38.13|:443... connected.
                                                                            │ HTTP request sent, awaiting response... 404 Not Found
                                                                            │ Remote file does not exist -- broken link!!!

Fourdee, 2018-03-04:
@sjdaniel10 

Thanks for the report 👍 

Confirmed, link is no longer available. We'll take a look.

sjdaniel10, 2018-03-04:
cool! Thanks for the speedy response! glad i could help 👍

Fourdee, 2018-03-04:
http://www.subsonic.org/pages/download2.jsp?target=subsonic-6.1.3.deb
We'll host on dietpi.com.

MichaIng, 2018-03-04:
Jep no files of the project are available right not on sourceforge and also the download button ends with 404: https://sourceforge.net/projects/subsonic/

Maybe just temporary, but also as SubSonic is not open source anymore, I can just encourage everybody to think about switching to AirSonic, which is after LibreSonic death the current active fork: https://github.com/airsonic/airsonic

 @Fourdee 
Let's add it to DietPi-Software, at least as replacement for SubSonic5, which is btw currently of course broken as well 😉: https://github.com/Fourdee/DietPi/blob/a6c0decae0e9990fb7ee98d0903c9186ffedd03c/dietpi/dietpi-software#L4459

Fourdee, 2018-03-04:
@sjdaniel10 

Resolved link for Subsonic 6 in v6.3.

v6.3 will be released in the next few days.

---

@MichaIng 

> Let's add it to DietPi-Software, at least as replacement for SubSonic5, which is btw currently of course broken as well 

👍

sjdaniel10, 2018-03-04:
wow @MichaIng @Fourdee , i don't think i have received speedy, accurate and help support like this.... well ever! keep it up, will look out for the update 👍

closed 2018-03-04 by Fourdee